Arduino-IRremote/Arduino-IRremote #476: Problems With Codes as a Variable

mrtheduke, 2017-05-30:
**Board:** ARDUINO UNO
**Library Version:** 2.1.0
**Protocol:** Panasonic (if any)

Hi,
I'm attempting to pass an IR code via serial to my arduino and have it emit it via an IR LED.
If I try and use the code below, the serial console shows the input correctly as 0x1004C4D, but no signal is sent.  If I replace receivedChars in the sendPanasonic method to the hard coded PanasonicPower value it works fine.  This is the first time I've tried to use the library, so I'm probably just doing something wrong, but any help would be greatfully received.

**Code Block:**
```c

Serial.print("Panasonic ");
        Serial.println(receivedChars);
        #define PanasonicAddress      0x4004     // Panasonic address (Pre data) 
        #define PanasonicPower        0x1004C4D  // Panasonic Power button
        IRsend irsend;
        irsend.sendPanasonic(PanasonicAddress,receivedChars);
        newData = false;

```

Many thanks

AnalysIR, 2017-05-30:
You probably need to show more(possibly all) of your code.....to get a useful response

mrtheduke, 2017-05-31:
Thanks, the gist with all of the code is here:
[Gist](https://gist.github.com/mrtheduke/c147425ce0e491d117a8bb7ec0f524a7)

AnalysIR, 2017-05-31:
`irsend.sendPanasonic(PanasonicAddress,receivedChars);`

Your '**receivedChars**' variable is an array of Chars & the function call sendPanasonic requires an **unsigned long** type (32 bit value), which is why it is not working for you.

Essentially, you need to convert the characters you receive over serial, into a 32 bit unsigend integer.
Search for arduino examples on
"parse HEX to long" or if you want to use decimal values "parseInt"

mrtheduke, 2017-05-31:
Ahhhh, that make sense.  Thanks so much!!

mrtheduke, 2017-05-31:
All working.  Thanks again for your help.  For anyone with similar problems, this code replaced my sendPanasonic function in the gist above:

`void sendPanasonic() {
    if (newData == true) {
        Serial.print("Recived Panasonic Code: ");
        Serial.println(receivedChars);
        unsigned long int code = strtoul(receivedChars,0,16);
        #define PanasonicAddress      0x4004     // Panasonic address (Pre data) 
        IRsend irsend;
        irsend.sendPanasonic(PanasonicAddress,code);
        newData = false;
    }
}`

closed 2017-05-31 by mrtheduke