Arduino-IRremote/Arduino-IRremote #477: Obtaining sendNEC remote codes from an LIRC file

mrtheduke, 2017-05-31:
**Board:** ARDUINO UNO
**Library Version:** 2.1.0
**Protocol:** NEC (if any)

I'm trying (and failing) to send codes using irsend.sendNEC.  I'm fairly sure the issue is with the codes I'm using.  I've got the codes for my remote from this:
[LIRC FILE](gist:22841e3d2d302e837d9d9a5071b42287)

I'm attempting to send the codes using:
`irsend.sendNEC(code, 16);`

where code is one of the hex codes in the 'begin codes' section of the LIRC file.
After a bit of digging around, I found this:
[Blog Article](http://www.righto.com/2010/03/understanding-sony-ir-remote-codes-lirc.html)

Which makes me think that I need to something with the numbers in the header field, but I can't quite work out how I combine that with the hex code and convert it all into a code that sendNEC can use.

If anyone can help steer me down the right path it'd be much appreciated.

Many thanks.

AnalysIR, 2017-05-31:
`irsend.sendNEC(code, 16);`

try instead
irsend.sendNEC(code, 32);

NEC is 32 bits long (not 16)!

You may also need to add in the **pre-data** part as well
e.g.
for power signal
0x8  +  0x00FF =>0x000800FF ....or something similar

for Mute signal
0x8  +  0x18E7 => 0x000818E7

etc etc....

You may note that the last 2 bytes are the inverse of each other! _(In this particular implementation of NEC)_

mrtheduke, 2017-05-31:
Thanks!! It did need both 32 Bits (note to self: don't trust the bits field in LIRC files), and the pre-data. Working now, thanks again for your help.

closed 2017-05-31 by mrtheduke