Arduino-IRremote/Arduino-IRremote #479: Error decoding Panasonic remote code

shyam1998, 2017-06-16:
I am trying to decode Panasonic Air conditioner remote control codes. When I upload this code 

`#include <IRremote.h>
IRrecv irrecv(2);
decode_results results;
void setup()
{
  Serial.begin(9600);
  irrecv.enableIRIn();

}
void loop()
{
  if (irrecv.decode(&results))
  {
    Serial.println(results.value , HEX);
    irrecv.resume();
  }

}
`
I get the following hexadecimal value: 7200000FFFFFFFF

However when I use the following code in the transmitter , it doesn't work.
`#include <IRremote.h>
IRsend irsend;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
}

void loop() {
  irsend.sendNEC(0x7200000FFFFFFFF, 32);
}`
When I compared the signals from the transmitter and the reciever they were not the same.
What could be the problem?
Does panasonic uses diffrent protocol ?
Let me know ASAP.
Thank you.

shyam1998, 2017-06-16:
`DKWJWD`

AnalysIR, 2017-06-16:
NEC is usually 32 bits & you are trying to prove 64 bits...plus more issues with variable types etc
It would be very fortunate if teh AC timings were the same as NEC.

For longer AC signals, you are probably better using the SendRAW function.

closed 2017-08-22 by shyam1998