Arduino-IRremote/Arduino-IRremote #480: Need help decoding IR Data

shyam1998, 2017-06-17:
Hello , I am trying to capture IR Signal from my panasonic airconditioner remote , I successfully recieved the data but I can't figure out what protocol it uses . I tried decoding the data with IRremote library I got some hexadecimal value but when I send the hexadecimal value it doesn't work. so I have attached the data i got from the logic analyzer here . Could you figure out what protocol it uses ?
The software used is Sigrok PulseView.
[ffg.zip](https://github.com/z3t0/Arduino-IRremote/files/1082506/ffg.zip)

AnalysIR, 2017-06-17:
The data in the zip doesnt look very useful (to me at least).

Try recording some signals using:
[Air Conditioners: Recording long Infrared Remote control signals with Arduino](https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/)

and post the results, I will then check it out for you.

shyam1998, 2017-06-18:
@AnalysIR  This what I got.
[RAW.txt](https://github.com/z3t0/Arduino-IRremote/files/1083185/RAW.txt)

AnalysIR, 2017-06-18:
Yes it decodes Ok as a panasonic 216 bit AC protocol on AnalysIR.

Example code to send with IRremote using sendRAW is below. (I have also cleaned the timings for you)

```

/*
Automatically Generated by AnalysIR - Batch Export Utility
Registered to: xxxxxxxx
Session History
Type : Key : Value : Bits : Carrier Frequency (kHz)
0 : PANASONIC128_216_280AC :  : 400407200000006040040720009C2C01AA000070070000810000ED : 216 : 0
Note: Be sure to use the correct Carrier frequency, for each individual signal, as(or if) indicated above
*/

// NB: Not all protocols are supported by IRremote or IRLib. You may need to edit the code below manually
// Automatically Generated by AnalysIR for xxxxxxxx, visit https://www.AnalysIR.com/ or email info@....... for further details
int khz=38; //NB Change this default value as neccessary to the correct carrier frequency

unsigned int Signal_0_0[] = {3500,1750,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,1300,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,435,435,435,435,435,435,435,435,435,435,10320,3500,1750,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,1300,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,435,435,435,435,1300,435,1300,435,1300,435,435,435,435,435,435,435,435,435,1300,435,435,435,1300,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,435,435,1300,435,435,435,1300,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,1300,435,435,435,1300,435,1300,435,435,435,1300,435}; //AnalysIR Batch Export (IRremote) - RAW

irsend.sendRaw(Signal_0_0, sizeof(Signal_0_0)/sizeof(int), khz); //AnalysIR Batch Export (IRremote) - RAW
 // AnalysIR IR Protocol: PANASONIC128_216_280AC, Key:  

```

shyam1998, 2017-06-22:
@AnalysIR  I tried it , still doesn't work.

AnalysIR, 2017-06-22:
please post a photo of your emitter circuit & a description of how it is wired, along with the value/model of any components used.

make sure you are pointing the top of theIR LEd towards the AC and bring it a close a possible tothe AC.

Have you been able to get an IR signal working with your TV? Did you check the IR LED is working and wired correctly?

spaghetti-, 2017-06-22:
Commenting here because I've been having some difficulties getting my AC to recognize the IR from the emitter as well. However even though some posts mention that you can get 10ft of range using around 40mA from the PWM pin, I will report back after I use a bjt to drive the LED at around 100mA or so.

spaghetti-, 2017-06-22:
@AnalysIR How did you clean up the timings given by the .ino file linked above (to AnalysisIR)? The raw has plenty of negatives etc.

AnalysIR, 2017-06-22:
@spaghetti- Yes that is why I requested the driver circuit. A transistor IR driver, in my opinion, would be a minimum good practice for controlling AC units. (unless emmitter & AC unit are adjacent)

> How did you clean up the timings given by the .ino file linked above (to AnalysisIR)? The raw has plenty of negatives etc.

The timings are corrected using our AnalysIR application.
FYI: In the recorded data, every second duration is negative to indicate a space...+ve => mark.
However, in the C code they are both treated as +ve durations. Always starts with a mark and every other value is a space...mark,space,mark,space...etc (without  signs)

spaghetti-, 2017-06-22:
@AnalysIR I see, thanks! Would it be too much trouble to ask you for a cleanup on the above timings? I'd really get the application but doesn't seem to actually work in demo.

AnalysIR, 2017-06-22:
This seems like a new protocol and probably not relevant to IRremote.
If you want, send me a full set of signals from your remote & I will look at them. (to info@anal____.com)

The signal looks similar to an NEC format, but the timings are different.
What device did you record on? 

In the meantime, the relevant timigs look like
8432,4216....followed by marks of duration 526 and spaces of duration 526 or 1582.

You should be able to edit this manually, or if you email me a signal set, I will take a proper look.

spaghetti-, 2017-06-22:
I will email you some more details regarding this signal set then!

Thanks!

bengtmartensson, 2017-06-22:
There is also free ("open source") software that can clean and analyze your signals. [IrpTransmogrifier](https://github.com/bengtmartensson/IrpTransmogrifier)  says:

    irptransmogrifier -a 200  analyze --statistics  --maxparameterwidth 32    8468, -4180, 552, -508, 552, -1564, 552, -504, 552, -1540, 556, -500, 560, -1556, 560, -1532, 584, -476, 548, -508, 552, -1564, 552, -1564, 552, -480, 580, -1536, 556, -1560, 556, -1560, 556, -504, 556, -1536, 548, -1568, 548, -508, 552, -508, 548, -1540, 556, -1560, 556, -1564, 552, -480, 576, -480, 560, -500, 556, -504, 556, -500, 556, -504, 556, -500, 556, -476, 552, -508, 552, -504, 552, -508, 552, -508, 552, -504, 552, -480, 576, -1540, 556, -504, 552, -504, 556, -1560, 556, -1536, 548, -1568, 548, -512, 556, -500, 560, -500, 548, -1540, 552, -508, 552, -1564, 552, -508, 548, -484, 576, -480, 556, -504, 556, -500, 556, -504, 556, -504, 552, -504, 556, -476, 580, -480, 548, -508, 552, -508, 548, -508, 552, -508, 548, -484, 576, -1540, 556, -504, 552, -1564, 552, -504, 556, -1536, 556, -1560, 556, -504, 556, -500, 556, 10000
    Gaps:
    A:      533     = 1*533         48
    B:      1552    = 3*533         24
    C:      4180    = 8*533         1
    E:      10000   = 19*533        1

    Flashes:
    A:      533     = 1*533         73
    D:      8468    = 16*533        1

    Pairs:
    AA:     48
    AB:     24
    AE:     1
    DC:     1

    DC AA AB AA AB AA AB AB AA AA AB AB AA AB AB AB AA AB AB AA AA AB AB AB AA AA AA AA AA AA AA AA AA AA AA AA AA AA AB AA AA AB AB AB AA AA AA AB AA AB AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA AB AA AB AA AB AB AA AA AE
    {533,msb}<1,-1|1,-3>(16,-8,A:32,B:32,C:8,1,-19){A=0x566ece00,B=0x4e28000,C=0xac}
    weight = 17

    $ irptransmogrifier -a 200 analyze --maxparameterwidth 32  --statistic  8468, -4180, 552, -504, 556, -1560, 556, -508, 552, -1532, 552, -508, 548, -1568, 548, -1540, 576, -484, 556, -504, 552, -1564, 552, -1536, 580, -508, 520, -1568, 548, -1568, 560, -1532, 572, -484, 556, -1560, 552, -1568, 548, -508, 552, -508, 528, -1560, 556, -1560, 560, -1532, 572, -512, 524, -508, 552, -508, 548, -512, 548, -508, 548, -512, 548, -508, 528, -504, 556, -504, 552, -504, 556, -504, 552, -504, 556, -504, 552, -508, 520, -1568, 548, -512, 548, -508, 548, -1540, 576, -1540, 556, -1564, 552, -508, 548, -508, 552, -508, 572, -460, 556, -1556, 560, -1560, 556, -500, 556, -504, 524, -508, 548, -512, 548, -508, 552, -508, 548, -508, 552, -508, 548, -512, 524, -508, 552, -504, 552, -508, 552, -504, 556, -504, 552, -508, 520, -1564, 552, -508, 548, -512, 548, -1540, 576, -1540, 556, -1560, 556, -504, 552, -508, 552, 10000
    Gaps:
    A:      533     = 1*533         48
    B:      1553    = 3*533         24
    C:      4180    = 8*533         1
    E:      10000   = 19*533        1

    Flashes:
    A:      533     = 1*533         73
    D:      8468    = 16*533        1

    Pairs:
    AA:     48
    AB:     24
    AE:     1
    DC:     1

    DC AA AB AA AB AA AB AB AA AA AB AB AA AB AB AB AA AB AB AA AA AB AB AB AA AA AA AA AA AA AA AA AA AA AA AA AA AA AB AA AA AB AB AB AA AA AA AA AB AB AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA AB AA AA AB AB AB AA AA AE
    {533,msb}<1,-1|1,-3>(16,-8,A:32,B:32,C:8,1,-19){A=0x566ece00,B=0x4e18000,C=0x9c}
    weight = 17

You may also use IrScrutinizer.  (I had to add an ending silence to your sequences.)

shyam1998, 2017-06-23:
@AnalysIR This is how the circuit is wired up. And Yeah I was able to get this working with my sony tv , just doesn't work for the air conditioner.
[image]
[data.txt](https://github.com/z3t0/Arduino-IRremote/files/1097823/data.txt)
This is what I get when I upload the  Irrecvdump sketch.

shyam1998, 2017-06-23:
@AnalysIR  The airconditioner now detects the IR signal , but still it doesn't switches off the AC rather it just gives a beep sound.

AnalysIR, 2017-06-23:
@shyam1998 
Circuit looks fine, by connecting to 5V you will get approx. double the IR power output.

I wont have much time this weekend, but if you send me a set of signals, I will have a deeper look learly next week.

> If you want, send me a full set of signals from your remote & I will look at them. (to info@ana_____.com)

Some times there is a different on and off signal. Try sending the signals with the AC on and also when off (use your remote to change the state)

shyam1998, 2017-06-23:
@AnalysIR  It works! But I Can only turn it ON , how to switch it off ?

AnalysIR, 2017-06-23:
Great !
you need to record the off signal, i guess!

djpavlovic, 2017-06-29:
Hello Everyone,
If i may ask for assistance.
I have uno with [https://www.cooking-hacks.com/documentation/tutorials/control-hvac-infrared-devices-from-the-internet-with-ir-remote/](url) shield.
i am trying to decode TCL A/C unit and i got stuck at it. cant understand anything, googling trying different options nothing works, so i found this thread and i am hoping that ill get at least help for on/off decode.

here are the signals that i recorded for on/off button with @AnalysIR sketch.

`Raw: (227) 3120, -1588, 548, -1068, 524, -1064, 548, -300, 528, -292, 544, -300, 524, -1064, 548, -300, 528, -320, 528, -1060, 520, -1092, 528, -292, 524, -1092, 520, -300, 528, -320, 524, -1064, 520, -1096, 524, -296, 540, -1076, 528, -1056, 552, -296, 520, -300, 548, -1068, 524, -320, 528, -292, 524, -1092, 528, -292, 524, -324, 524, -296, 520, -328, 520, -300, 544, -300, 528, -320, 528, -292, 524, -324, 524, -296, 520, -328, 516, -304, 524, -320, 528, -292, 576, -272, 524, -296, 552, -296, 528, -320, 528, -292, 524, -320, 528, -1060, 528, -320, 528, -292, 576, -1040, 520, -1068, 544, -300, 524, -296, 552, -296, 520, -300, 548, -300, 528, -320, 524, -1060, 524, -324, 520, -1068, 524, -1092, 520, -300, 524, -320, 528, -292, 576, -272, 524, -296, 552, -1064, 524, -320, 528, -292, 524, -328, 520, -296, 520, -328, 520, -300, 524, -320, 528, -292, 576, -272, 524, -296, 572, -276, 520, -324, 520, -300, 528, -320, 528, -292, 524, -320, 528, -292, 524, -324, 520, -300, 580, -268, 528, -292, 576, -272, 520, -324, 524, -296, 520, -328, 520, -300, 528, -316, 528, -292, 524, -324, 524, -296, 572, -276, 520, -300, 576, -272, 524, -320, 548, -272, 556, -292, 556, -264, 548, -1064, 528, -1060, 520, -1068, 556, -1060, 520, -324, 552, -272, 576, -268, 560, -1028, 572, -1044, 524, 
Raw: (227) 3148, -1592, 548, -1068, 520, -1068, 544, -304, 524, -296, 552, -296, 528, -1060, 552, -296, 532, -316, 528, -1064, 516, -1096, 528, -296, 520, -1092, 528, -296, 520, -324, 524, -1068, 544, -1072, 516, -304, 544, -1072, 520, -1068, 544, -304, 520, -300, 548, -1068, 524, -324, 524, -296, 520, -1096, 524, -296, 520, -328, 520, -300, 524, -324, 524, -296, 540, -308, 520, -300, 548, -300, 524, -320, 528, -296, 520, -328, 520, -300, 524, -324, 524, -296, 572, -276, 520, -300, 548, -300, 524, -1064, 548, -300, 528, -320, 524, -1064, 528, -320, 528, -292, 524, -1092, 528, -1064, 516, -328, 532, -292, 544, -304, 524, -296, 548, -300, 528, -316, 532, -1060, 520, -324, 520, -1072, 520, -1096, 528, -292, 524, -324, 520, -300, 580, -268, 528, -292, 544, -1072, 528, -292, 544, -304, 524, -324, 524, -296, 520, -328, 516, -304, 524, -324, 524, -296, 572, -276, 528, -292, 576, -272, 524, -300, 548, -296, 532, -316, 528, -292, 524, -324, 524, -296, 528, -320, 528, -296, 572, -276, 520, -300, 576, -272, 524, -320, 528, -296, 520, -328, 520, -300, 524, -324, 524, -296, 520, -328, 520, -300, 580, -268, 524, -296, 552, -304, 512, -328, 528, -292, 528, -320, 524, -296, 520, -1096, 528, -1060, 528, -1088, 524, -300, 528, -1084, 528, -296, 568, -280, 528, -1060, 572, -1044, 524, 
Raw: (227) 3120, -1588, 548, -1068, 524, -1064, 548, -300, 528, -292, 544, -300, 524, -1064, 548, -300, 528, -320, 528, -1060, 520, -1092, 528, -292, 524, -1092, 520, -300, 528, -320, 524, -1064, 520, -1096, 524, -296, 540, -1076, 528, -1056, 552, -296, 520, -300, 548, -1068, 524, -320, 528, -292, 524, -1092, 528, -292, 524, -324, 524, -296, 520, -328, 520, -300, 544, -300, 528, -320, 528, -292, 524, -324, 524, -296, 520, -328, 516, -304, 524, -320, 528, -292, 576, -272, 524, -296, 552, -296, 528, -320, 528, -292, 524, -320, 528, -1060, 528, -320, 528, -292, 576, -1040, 520, -1068, 544, -300, 524, -296, 552, -296, 520, -300, 548, -300, 528, -320, 524, -1060, 524, -324, 520, -1068, 524, -1092, 520, -300, 524, -320, 528, -292, 576, -272, 524, -296, 552, -1064, 524, -320, 528, -292, 524, -328, 520, -296, 520, -328, 520, -300, 524, -320, 528, -292, 576, -272, 524, -296, 572, -276, 520, -324, 520, -300, 528, -320, 528, -292, 524, -320, 528, -292, 524, -324, 520, -300, 580, -268, 528, -292, 576, -272, 520, -324, 524, -296, 520, -328, 520, -300, 528, -316, 528, -292, 524, -324, 524, -296, 572, -276, 520, -300, 576, -272, 524, -320, 548, -272, 556, -292, 556, -264, 548, -1064, 528, -1060, 520, -1068, 556, -1060, 520, -324, 552, -272, 576, -268, 560, -1028, 572, -1044, 524, 
Raw: (227) 3148, -1592, 548, -1068, 520, -1068, 544, -304, 524, -296, 552, -296, 528, -1060, 552, -296, 532, -316, 528, -1064, 516, -1096, 528, -296, 520, -1092, 528, -296, 520, -324, 524, -1068, 544, -1072, 516, -304, 544, -1072, 520, -1068, 544, -304, 520, -300, 548, -1068, 524, -324, 524, -296, 520, -1096, 524, -296, 520, -328, 520, -300, 524, -324, 524, -296, 540, -308, 520, -300, 548, -300, 524, -320, 528, -296, 520, -328, 520, -300, 524, -324, 524, -296, 572, -276, 520, -300, 548, -300, 524, -1064, 548, -300, 528, -320, 524, -1064, 528, -320, 528, -292, 524, -1092, 528, -1064, 516, -328, 532, -292, 544, -304, 524, -296, 548, -300, 528, -316, 532, -1060, 520, -324, 520, -1072, 520, -1096, 528, -292, 524, -324, 520, -300, 580, -268, 528, -292, 544, -1072, 528, -292, 544, -304, 524, -324, 524, -296, 520, -328, 516, -304, 524, -324, 524, -296, 572, -276, 528, -292, 576, -272, 524, -300, 548, -296, 532, -316, 528, -292, 524, -324, 524, -296, 528, -320, 528, -296, 572, -276, 520, -300, 576, -272, 524, -320, 528, -296, 520, -328, 520, -300, 524, -324, 524, -296, 520, -328, 520, -300, 580, -268, 524, -296, 552, -304, 512, -328, 5 [... cut ...] 28, 520, -300, 524, -324, 524, -296, 520, -328, 520, -300, 576, -272, 524, -296, 572, -276, 520, -300, 548, -300, 524, -320, 528, -292, 524, -324, 524, -296, 528, -320, 528, -292, 576, -272, 524, -296, 572, -276, 520, -300, 544, -304, 524, -320, 528, -292, 524, -324, 524, -296, 528, -1088, 524, -1064, 528, -1060, 552, -1064, 524, -324, 524, -296, 572, -276, 520, -1068, 572, -1040, 520, 

`
ac unit is marked as : TCL TAC-12CHSA/HCI
Hoping that one of your guys could help me out, thanks in advance

AnalysIR, 2017-06-29:
Yes I imported your first signals into AnalysIR & it decodes as TCL112AC  (112 bits).
Below is the automatically generated code to send this using IRremote. You need to add in some extra stuff in your sketch. (Please refer to the sendRAW demo in IRremote examples.)
I have also cleaned the timings for this signal, which should improve performance/range etc.
You can adjust timings for any other signal by editing or using a spreadsheet.

```
/*
Automatically Generated by AnalysIR - Batch Export Utility
Registered to: xxxxxxxx
Session History
Type : Key : Value : Bits : Carrier Frequency (kHz)
0 : TCL112AC :  : C4D364800004C0B04000000001E3 : 112 : 0
Note: Be sure to use the correct Carrier frequency, for each individual signal, as(or if) indicated above
*/

// NB: Not all protocols are supported by IRremote or IRLib. You may need to edit the code below manually
// Automatically Generated by AnalysIR for xxxxxxxx, visit https://www.AnalysIR.com/ or email info@....... for further details
int khz=38; //NB Change this default value as neccessary to the correct carrier frequency

unsigned int Signal_0_0[] = {2900,1800,405,1125,405,1125,405,405,405,405,405,405,405,1125,405,405,405,405,405,1125,405,1125,405,405,405,1125,405,405,405,405,405,1125,405,1125,405,405,405,1125,405,1125,405,405,405,405,405,1125,405,405,405,405,405,1125,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,1125,405,405,405,405,405,1125,405,1125,405,405,405,405,405,405,405,405,405,405,405,405,405,1125,405,405,405,1125,405,1125,405,405,405,405,405,405,405,405,405,405,405,1125,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,405,1125,405,1125,405,1125,405,1125,405,405,405,405,405,405,405,1125,405,1125,405}; //AnalysIR Batch Export (IRremote) - RAW

irsend.sendRaw(Signal_0_0, sizeof(Signal_0_0)/sizeof(int), khz); //AnalysIR Batch Export (IRremote) - RAW
 // AnalysIR IR Protocol: TCL112AC, Key:  
```
In addition to the examples available with the IRremote library, you will find more relevant examples and posts on our blog here: 
[IR related blog posts for Air Conditioners](https://www.analysir.com/blog/tag/air-conditioner/)

djpavlovic, 2017-07-01:
Hi,
Thanks @AnalysIR i appreciate the help and time spent, but unfortunately just beeps and does nothing with this signal.
I still cant understand why and how u got those modified timings, or how should i modify my future recordings.

I made few more investigations and i found out that  in this library it is included decoding of TCL112AC, but somehow it cant decode because it is in conflict with this: #https://github.com/z3t0/Arduino-IRremote/pull/259
And here is what i got after several attempts.

```
//recordings using AnalysIR sketch
//off signal
Raw: (227) 3136, -1592, 544, -1064, 580, -1004, 544, -304, 572, -244, 552, -292, 576, -1008, 548, -296, 572, -276, 572, -1012, 576, -1008, 604, -240, 576, -1036, 576, -240, 572, -276, 572, -1012, 548, -1060, 580, -240, 544, -1068, 576, -1008, 548, -296, 572, -244, 552, -1060, 580, -264, 572, -248, 528, -1084, 576, -240, 576, -272, 524, -292, 524, -324, 520, -296, 540, -308, 520, -324, 524, -292, 520, -328, 520, -296, 520, -324, 524, -296, 520, -324, 520, -300, 516, -328, 520, -296, 548, -296, 520, -324, 524, -296, 520, -324, 520, -1064, 528, -316, 520, -300, 524, -1084, 580, -1004, 524, -324, 572, -244, 540, -304, 524, -296, 552, -292, 520, -324, 524, -296, 520, -324, 524, -292, 524, -1088, 520, -296, 520, -324, 524, -296, 548, -296, 520, -300, 548, -296, 520, -324, 520, -1064, 516, -1068, 596, -1012, 576, -268, 528, -292, 524, -320, 524, -296, 520, -324, 524, -292, 544, -300, 524, -296, 552, -292, 524, -320, 528, -292, 520, -324, 524, -292, 524, -324, 524, -292, 524, -320, 524, -296, 540, -304, 524, -320, 528, -292, 520, -324, 524, -292, 524, -324, 524, -292, 520, -324, 524, -296, 540, -304, 524, -296, 548, -296, 520, -324, 524, -296, 520, -324, 520, -300, 516, -1092, 520, -300, 524, -320, 516, -300, 528, -1084, 524, -1060, 540, -1072, 572, -1008, 548, -1064, 580, 

//on signal - cool 23c
Raw: (227) 3140, -1584, 544, -1064, 576, -1008, 552, -296, 520, -296, 548, -300, 516, -1064, 548, -300, 524, -320, 528, -1056, 572, -1012, 600, -244, 520, -1092, 568, -252, 524, -320, 528, -1056, 544, -1068, 572, -244, 552, -1084, 544, -1012, 548, -300, 528, -316, 520, -1064, 576, -268, 528, -292, 524, -1084, 576, -244, 520, -324, 524, -296, 520, -324, 520, -300, 548, -296, 520, -324, 524, -296, 528, -316, 520, -300, 524, -320, 528, -292, 524, -320, 524, -296, 520, -324, 524, -296, 540, -304, 520, -1064, 548, -296, 520, -324, 520, -1064, 580, -264, 520, -300, 524, -1088, 576, -1008, 520, -324, 524, -296, 548, -296, 520, -296, 552, -296, 520, -324, 520, -296, 520, -328, 520, -296, 528, -1056, 608, -240, 524, -320, 524, -292, 544, -300, 528, -320, 516, -300, 524, -320, 528, -1056, 576, -1008, 600, -1012, 580, -264, 520, -300, 524, -320, 528, -292, 524, -320, 524, -296, 540, -304, 524, -320, 528, -292, 520, -324, 524, -292, 524, -324, 524, -292, 520, -328, 520, -296, 572, -272, 524, -296, 548, -296, 520, -328, 520, -296, 520, -324, 520, -300, 528, -316, 520, -300, 524, -320, 528, -292, 544, -304, 520, -296, 540, -304, 524, -320, 528, -292, 520, -324, 524, -296, 520, -1064, 548, -296, 528, -316, 520, -1064, 548, -1064, 524, -1060, 552, -1060, 520, -1064, 544, -1068, 520, 

//off
Raw: (227) 3108, -1584, 544, -1064, 576, -1004, 544, -300, 528, -316, 520, -296, 528, -1056, 544, -300, 528, -316, 520, -1060, 580, -1032, 580, -236, 524, -1088, 576, -240, 524, -320, 528, -1056, 544, -1064, 576, -244, 552, -1084, 548, -1008, 548, -296, 520, -324, 524, -1060, 572, -272, 520, -296, 520, -1064, 600, -248, 516, -324, 520, -300, 548, -296, 520, -296, 548, -300, 516, -324, 524, -296, 520, -324, 520, -300, 516, -328, 520, -296, 520, -324, 520, -300, 548, -296, 520, -296, 552, -296, 516, -328, 520, -296, 520, -324, 520, -1064, 580, -264, 520, -296, 520, -1092, 568, -1012, 548, -300, 524, -292, 544, -300, 528, -316, 520, -300, 524, -320, 516, -300, 528, -316, 520, -300, 524, -1056, 544, -304, 524, -320, 516, -300, 548, -296, 516, -328, 520, -296, 520, -324, 524, -1060, 516, -1068, 596, -1012, 580, -264, 520, -300, 524, -320, 516, -300, 548, -296, 520, -300, 544, -300, 516, -328, 520, -296, 520, -324, 520, -300, 528, -316, 520, -296, 528, -320, 516, -300, 548, -296, 520, -300, 544, -300, 516, -328, 520, -296, 520, -324, 520, -300, 516, -328, 520, -296, 520, -324, 520, -300, 548, -296, 520, -324, 524, -296, 516, -328, 520, -296, 520, -324, 524, -296, 516, -1068, 544, -300, 528, -316, 520, -296, 548, -1064, 528, -1056, 544, -1064, 524, -1060, 540, -1068, 524, 

//on 19
Raw: (227) 3104, -1592, 544, -1064, 576, -1032, 516, -304, 524, -296, 548, -296, 520, -1064, 548, -296, 516, -328, 520, -1060, 580, -1032, 572, -244, 528, -1056, 608, -236, 524, -320, 528, -1056, 544, -1092, 548, -244, 544, -1064, 576, -1036, 524, -292, 520, -328, 520, -1060, 572, -272, 524, -296, 520, -1064, 596, -248, 528, -316, 520, -300, 544, -300, 516, -300, 548, -300, 524, -320, 516, -300, 528, -316, 520, -300, 524, -320, 516, -300, 528, -316, 520, -300, 544, -300, 528, -316, 520, -300, 524, -1056, 544, -300, 532, -316, 516, -1064, 576, -268, 528, -292, 524, -1084, 576, -1036, 516, -300, 524, -296, 548, -296, 520, -300, 548, -296, 520, -324, 524, -292, 520, -324, 524, -1060, 572, -1036, 572, -248, 528, -316, 520, -300, 544, -300, 516, -328, 520, -296, 520, -324, 524, -1060, 580, -1028, 580, -1032, 548, -268, 528, -292, 524, -320, 524, -292, 544, -300, 528, -292, 544, -300, 524, -320, 516, -300, 528, -320, 516, -300, 524, -320, 516, -304, 524, -320, 524, -292, 548, -296, 528, -292, 544, -300, 524, -320, 516, -300, 528, - [... cut ...]  500,1050, 550,1050, 500,300, 550,300, 500,350, 500,300, 500,350, 500,300, 500,300, 550,300, 550,250, 550,300, 550,250, 550,300, 500,300, 550,300, 500,300, 500,350, 500,300, 550,300, 500,350, 500,300, 500,300, 550,300, 500,300, 550,300, 500,300, 550,250, 550,300, 550,250, 550,300, 500,350, 500,300, 500,350, 500,300, 500,1050, 550,300, 500,350, 500,300, 500,350, 500,300, 550,300, 500,300, 550,300, 500};  // PANASONIC C4D3:64800024
unsigned int  addr = 0xC4D3;
unsigned int  data = 0x64800024;

```

AnalysIR, 2017-07-01:
> I still cant understand why and how u got those modified timings, or how should i modify my future recordings.

It is a feture of our AnalysIr application. You can do this manually, by changing the times manually (e.g. if its 500 chnge it to 405...if 300 change it to 405 etc.

it could be that you are sendin the OFF command and the AC is already OFF...so nothing changes (so test with the AC on and then with it OFF)

djpavlovic, 2017-07-01:
Ok, thanks.
I am posting for people who did find this post for this A/C unit.
```

#include <IRremoteInt.h>
#include <IRremote.h>

int recvPin = 2; //receive pin on pin 2as coocking hacks, not used in this
int IRledPin =  3;      // IR emitter LED connected to digital pin 3
int buttonPin1 = 4;     // "Send" push-button connected to digital pin 4
int buttonPin2 = 5;     // "Receive" push-button connected to digital pin 5
int blueLedPin =  6;    // Blue LED connected to digital pin 6
int orangeLedPin =  7;  // Orange LED connected to digital pin 7
int buttonState1 = 0;  //storing button state

int khz = 38; // 38kHz carrier frequency for the NEC protocol

//IRrecv irrecv(recvPin);//not used
IRsend irsend;

void setup()   {
  // Initialize the Orange LED pin as an output
  pinMode(orangeLedPin, OUTPUT);
  // Initialize the Orange LED pin as an output
  pinMode(blueLedPin, OUTPUT);
  // Initialize the "Send" push-button pin as an input
  pinMode(buttonPin1, INPUT);
  // Initialize the "Receive" push-button pin as an input
  pinMode(buttonPin2, INPUT);

  // Set uart baudrate
  Serial.begin(9600);
}

void loop()   {
  // Read the state of the "Send" push-button value
  buttonState1 = digitalRead(buttonPin1);

  if (buttonState1 == HIGH) {
    Serial.println("Sending IR signal");
    digitalWrite(orangeLedPin, HIGH);
    delay(200);
    //sending code
    SendIRCode();
    digitalWrite(orangeLedPin, LOW);

    delay(5000);  // wait 5 seconds to resend the code
    Serial.println("5 secs passed");
    digitalWrite(blueLedPin, HIGH);
    delay(200);
    //sending code
    SendOffCode();
    digitalWrite(blueLedPin, LOW);
    delay(5000);  // wait 5 seconds to resend the code
    Serial.println("5 secs passed");

  }
}

void SendIRCode() {
  //sending 19c cool
  unsigned int  irSignal[] = {3100, 1550, 600, 1000, 550, 1050, 550, 250, 550, 300, 550, 250, 550, 1050, 550, 300, 500, 300, 550, 1050, 550, 1000, 600, 250, 550, 1050, 550, 250, 600, 250, 500, 1100, 550, 1050, 500, 300, 550, 1050, 500, 1050, 600, 250, 550, 250, 550, 1050, 550, 300, 550, 250, 550, 1050, 550, 300, 500, 300, 550, 300, 550, 250, 550, 300, 550, 250, 550, 300, 550, 300, 500, 300, 550, 300, 500, 300, 550, 300, 500, 300, 550, 300, 550, 250, 550, 300, 550, 250, 550, 1050, 550, 300, 500, 300, 550, 1050, 500, 300, 550, 300, 550, 1050, 500, 1050, 600, 250, 500, 300, 600, 250, 550, 250, 550, 300, 500, 300, 550, 300, 550, 250, 550, 1050, 550, 1050, 550, 300, 500, 300, 550, 300, 550, 250, 550, 300, 550, 1050, 500, 300, 550, 300, 500, 300, 550, 300, 500, 300, 550, 300, 550, 250, 550, 300, 550, 250, 550, 300, 550, 250, 550, 300, 550, 300, 500, 300, 550, 300, 500, 300, 550, 300, 550, 250, 550, 300, 500, 300, 550, 300, 500, 300, 550, 300, 500, 300, 550, 300, 500, 300, 550, 300, 550, 250, 600, 250, 500, 300, 550, 300, 550, 250, 550, 300, 500, 300, 550, 300, 550, 300, 500, 300, 550, 1050, 500, 300, 600, 1000, 550, 300, 550, 1050, 500, 300, 600, 250, 500, 1050, 550, 1050, 550}; // TCL112AC FFFFFEAC
  irsend.sendRaw(irSignal, sizeof(irSignal) / sizeof(irSignal[0]), khz);

}

void SendOffCode() {
  //off signal
  unsigned int Signal_0_0[] = {2900, 1800, 405, 1125, 405, 1125, 405, 405, 405, 405, 405, 405, 405, 1125, 405, 405, 405, 405, 405, 1125, 405, 1125, 405, 405, 405, 1125, 405, 405, 405, 405, 405, 1125, 405, 1125, 405, 405, 405, 1125, 405, 1125, 405, 405, 405, 405, 405, 1125, 405, 405, 405, 405, 405, 1125, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 1125, 405, 405, 405, 405, 405, 1125, 405, 1125, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 1125, 405, 405, 405, 1125, 405, 1125, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 1125, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 405, 1125, 405, 1125, 405, 1125, 405, 1125, 405, 405, 405, 405, 405, 405, 405, 1125, 405, 1125, 405}; //AnalysIR Batch Export (IRremote) - RAW
  irsend.sendRaw(Signal_0_0, sizeof(Signal_0_0) / sizeof(Signal_0_0[0]), khz);

}
```

AnalysIR, 2017-07-01:
Great...glad it worked out.

If you run out of SRAM, you might find the following post helpful....where you can store many more signals in FLASH.
[Sending long AC Signals from Flash with IRremote](https://www.analysir.com/blog/2016/04/11/sending-long-ac-signals-flash-irremote/)

shyam1998, 2017-07-16:
@AnalysIR Any idea on how to record the 'off' signal?

AnalysIR, 2017-07-16:
@shyam1998 

It is possible that there are 2 differnet signals, one for off and one for on.

So by recording multiple sequential presses of the power button, you may find 2 different IR signals, that you can use? You will have to test it out.

**_TIP:_** _Before capturing the signals, set all parameters on your AC to the default you want (temp, fan etc) as  the ON signal **may** then contain these parameters for you._

shyam1998, 2017-08-20:
@AnalysIR  I have recorded the on and off siganl from my remote. Can you clean up the signals for me?
[signals.txt](https://github.com/z3t0/Arduino-IRremote/files/1236965/signals.txt)

shyam1998, 2017-08-20:
@AnalysIR  I have recorded my on and off signals. Can you clean up the signals?
[signals (1).txt](https://github.com/z3t0/Arduino-IRremote/files/1236990/signals.1.txt)

AnalysIR, 2017-08-20:
@shyam1998 
I have responded also in your other opened issue on this this same topic #508, which doubles the effort in responding. Please post any follow ups in the other issue...so that this one can be closed....or close this one yourself.

@z3t0 please close this issue....as it seems to have drifted a lot away from the original & too long to read ever again :)

closed 2017-08-20 by shyam1998

missquesera, 2019-09-11:
hi. can anyone help me regarding controlling the aircond using smartphone? Im facing problem when i decode the AC remote for on and off button , the hex code is keeping on changing. I try again on/off button after 10 second , the value does not fix. can anyone help me anyway?how am i going to decode the signal if the hex value does not fix?

AnalysIR, 2019-09-11:
@missquesera 

....not related to this issue
... not related to IRremote
... little relevant/useful information provided

...sorry :(

jaldomir, 2019-09-11:
Where is relation between this question and Arduino irRemote library? None?

Good. No donut for you.

___
Jaldomir da Silva Filho, Sc.M.
*enviado de dispositivo móvel

missquesera, 2019-09-17:
hi , sorry for not explaining properly. Now im doing a project for controlling the aircond , so i already decode the IR remote from airconditoner remote control.  So i go through any forum related to this which i had exactly same problem as  " I tried decoding the data with IRremote library I got some hexadecimal value but when I send the hexadecimal value it doesn't work."from https://github.com/shyam1998. I succesfully decode and encode using tv remote control and it works. For tv , i can control the tv using IRremoteESP8266 library but could not do the same for aircond.

Here i attached you the data i received uisng IRremote library : 
Enabling IRin
Enabled IRin
8BB515AC
8BB515AC
8BB515AC
8BB515AC
504343D2
8BB515AC
8BB515AC
8BB515AC
8BB515AC
8BB515AC
8BB515AC

above is the hex coode when i'm pressing on and off at 1.31PM. Below is the hexcode im pressing on and off at 1.32PM. i can see the value is changing due to time. do u have any idea how can i eliminate the the time that contain in hex code , i only want on and off code only. hope this will help. Please asking if i can provide more info. Thank you so much

17FD798
17FD798
BDE0A42F
17FD798
17FD798
ECA0A6A8

missquesera, 2019-09-17:
here i attached you the raw code for on and off at time 1.57PM. how do i process this code to get info?

IR code too long. Edit IRremoteInt.h and increase RAWBUF
Timing[100]: 
     +4700, -2450     + 500, - 250     + 450, - 900     + 450, - 850
     + 450, - 300     + 450, - 850     + 450, - 300     + 450, - 300
     + 450, - 300     + 450, - 300     + 450, - 850     + 450, - 300
     + 450, - 300     + 500, - 250     + 450, - 850     + 450, - 300
     + 450, - 300     + 500, - 850     + 450, - 850     + 450, - 850
     + 450, - 300     + 500, - 850     + 450, - 300     + 400, - 900
     + 450, - 300     + 450, - 850     + 450, - 300     + 450, - 300
     + 450, - 300     + 450, - 300     + 450, - 300     + 450, - 300
     + 450, - 300     + 450, - 300     + 400, - 300     + 450, - 300
     + 500, - 250     + 450, - 900     + 450, - 300     + 450, - 250
     + 500, - 250     + 450, - 300     + 450, - 300     + 450, - 300
     + 450, - 300     + 450, - 850     + 450, - 300     + 500, - 250
     + 450, - 300     + 500, - 850
unsigned int  rawData[100] = {4700,2450, 500,250, 450,900, 450,850, 450,300, 450,850, 450,300, 450,300, 450,300, 450,300, 450,850, 450,300, 450,300, 500,250, 450,850, 450,300, 450,300, 500,850, 450,850, 450,850, 450,300, 500,850, 450,300, 400,900, 450,300, 450,850, 450,300, 450,300, 450,300, 450,300, 450,300, 450,300, 450,300, 450,300, 400,300, 450,300, 500,250, 450,900, 450,300, 450,250, 500,250, 450,300, 450,300, 450,300, 450,300, 450,850, 450,300, 500,250, 450,300, 500,850 };  // UNKNOWN 3380D686

IR code too long. Edit IRremoteInt.h and increase RAWBUF
Timing[100]: 
     +4700, -2450     + 500, - 250     + 500, - 850     + 450, - 850
     + 450, - 300     + 450, - 850     + 500, - 250     + 450, - 300
     + 450, - 300     + 450, - 300     + 450, - 850     + 500, - 250
     + 500, - 250     + 450, - 300     + 450, - 850     + 500, - 250
     + 450, - 300     + 500, - 850     + 450, - 850     + 450, - 850
     + 500, - 250     + 500, - 850     + 450, - 300     + 450, - 850
     + 450, - 300     + 450, - 850     + 500, - 250     + 450, - 300
     + 450, - 300     + 450, - 300     + 450, - 300     + 450, - 300
     + 450, - 300     + 450, - 300     + 400, - 300     + 500, - 250
     + 500, - 250     + 500, - 850     + 450, - 300     + 400, - 300
     + 500, - 250     + 500, - 250     + 500, - 250     + 450, - 300
     + 500, - 250     + 450, - 850     + 500, - 250     + 450, - 300
     + 450, - 300     + 500, - 850
unsigned int  rawData[100] = {4700,2450, 500,250, 500,850, 450,850, 450,300, 450,850, 500,250, 450,300, 450,300, 450,300, 450,850, 500,250, 500,250, 450,300, 450,850, 500,250, 450,300, 500,850, 450,850, 450,850, 500,250, 500,850, 450,300, 450,850, 450,300, 450,850, 500,250, 450,300, 450,300, 450,300, 450,300, 450,300, 450,300, 450,300, 400,300, 500,250, 500,250, 500,850, 450,300, 400,300, 500,250, 500,250, 500,250, 450,300, 500,250, 450,850, 500,250, 450,300, 450,300, 500,850 };  // UNKNOWN 3380D686

IR code too long. Edit IRremoteInt.h and increase RAWBUF
Timing[100]: 
     +4700, -2500     + 400, - 300     + 450, - 900     + 450, - 850
     + 450, - 300     + 450, - 900     + 400, - 300     + 450, - 300
     + 450, - 300     + 450, - 300     + 450, - 900     + 400, - 300
     + 450, - 300     + 450, - 300     + 450, - 900     + 450, - 300
     + 400, - 300     + 450, - 900     + 450, - 850     + 450, - 900
     + 400, - 300     + 450, - 900     + 450, - 300     + 450, - 850
     + 450, - 300     + 450, - 850     + 450, - 300     + 450, - 300
     + 450, - 300     + 450, - 300     + 450, - 300     + 450, - 300
     + 450, - 300     + 450, - 300     + 450, - 300     + 400, - 350
     + 400, - 350     + 400, - 900     + 450, - 300     + 450, - 300
     + 450, - 300     + 400, - 350     + 400, - 300     + 450, - 300
     + 450, - 300     + 450, - 900     + 450, - 300     + 400, - 300
     + 450, - 300     + 450, - 900
unsigned int  rawData[100] = {4700,2500, 400,300, 450,900, 450,850, 450,300, 450,900, 400,300, 450,300, 450,300, 450,300, 450,900, 400,300, 450,300, 450,300, 450,900, 450,300, 400,300, 450,900, 450,850, 450,900, 400,300, 450,900, 450,300, 450,850, 450,300, 450,850, 450,300, 450,300, 450,300, 450,300, 450,300, 450,300, 450,300, 450,300, 450,300, 400,350, 400,350, 400,900, 450,300, 450,300, 450,300, 400,350, 400,300, 450,300, 450,300, 450,900, 450,300, 400,300, 450,300, 450,900 };  // UNKNOWN 3380D686

IR code too long. Edit IRremoteInt.h and increase RAWBUF
Timing[100]: 
     +4700, -2450     + 500, - 250     + 450, - 900     + 450, - 850
     + 450, - 300     + 450, - 850     + 450, - 300     + 450, - 300
     + 500, - 250     + 450, - 300     + 450, - 850     + 450, - 300
     + 500, - 250     + 500, - 250     + 500, - 800     + 500, - 250
     + 500, - 250     + 500, - 850     + 450, - 850     + 450, - 850
     + 450, - 300     + 450, - 900     + 450, - 300     + 450, - 850
     + 450, - 300     + 450, - 850     + 450, - 300     + 450, - 300
     + 450, - 300     + 450, - 300     + 450, - 300     + 450, - 300
     + 450, - 300     + 450, - 300     + 450, - 300     + 450, - 250
     + 500, - 250     + 500, - 850     + 450, - 300     + 450, - 300
     + 450, - 250     + 500, - 250     + 500, - 250     + 500, - 250
     + 450, - 300     + 500, - 800     + 450, - 300     + 500, - 250
     + 500, - 250     + 500, - 850
unsigned int  rawData[100] = {4700,2450, 500,250, 450,900, 450,850, 450,300, 450,850, 450,300, 450,300, 500,250, 450,300, 450,850, 450,300, 500,250, 500,250, 500,800, 500,250, 500,250, 500,850, 450,850, 450,850, 450,300, 450,900, 450,300, 450,850, 450,300, 450,850, 450,300, 450,300, 450,300, 450,300, 450,300, 450,300, 450,300, 450,300, 450,300, 450,250, 500,250, 500,850, 450,300, 450,300, 450,250, 500,250, 500,250, 500,250, 450,300, 500,800, 450,300, 500,250, 500,250, 500,850 };  // UNKNOWN 3380D686

AnalysIR, 2019-09-17:
it seems you have not even read through this thread. If you, do you might find a solution.

Otherwise please open a new issue to address your questions.

antonmayor, 2020-02-12:
Hi, I'm trying to clone the control signal of my a / c just like our friend. I managed to get the signal, but now there is no way to send the signal, I have used the IRremote library but I cannot turn off / on my air. Can you help me please

AnalysIR, 2020-02-12:
@antonmayor A good starting point would be to open your own(new) issue and follow the instructions there. _(Posting at the end of a 2.5 year old thread without providing any useful information or explaining what you have tried yourself, only serves to confuse or complicate matters and more importantly waste time)_

Before opening your own issue- trying out the examples provided with the library on your TV may shine some light on common beginner issues for you.

antonmayor, 2020-02-13:
Good day. I am new to this, I have a project of my internships and I have
to automate the emergency air in the server room. I tried to clone the
remote control signal with the IRremote library, in principle I had some
problems but in the end I managed to get the code, which is this: Encoding:
PANASONIC Code: C4D3: 64800004 (48 bits) Timing [100]:      +3850, -1350 +
650, -1100 + 650, -1050 + 650, - 400      + 650, - 350 + 700, - 350 + 600,
-1100 + 650, - 400      + 650, - 350 + 650, -1100 + 650, -1100 + 600, - 400
     + 650, -1100 + 600, - 400 + 650, - 350 + 650, -1100      + 650, -1050
+ 700, - 350 + 650, -1100 + 650, -1100      + 600, - 400 + 650, - 350 +
650, -1100 + 650, - 350      + 650, - 350 + 700, -1100 + 600, - 400 + 650,
- 350      + 650, - 400 + 650, - 350 + 700, - 350 + 600, - 400      + 650,
- 350 + 650, - 400 + 650, - 350 + 650, - 400      + 650, - 350 + 650, - 400
+ 600, - 400 + 650, - 350      + 650, - 350 + 700, - 350 + 650, - 400 +
650, - 350      + 650, - 400 + 600, - 400 + 650, -1100 + 600, - 400      +
650, - 350 + 700, -1050 unsigned int rawData [100] = {3850,1350, 650,1100,
650,1050, 650,400, 650,350, 700,350, 600,1100, 650,400, 650,350, 650,1100,
650,1100, 600,400, 650,1100, 600,400, 650,350, 650,1100, 650,1050, 700,350,
650,1100, 650,1100, 600,400, 650,350, 650,1100, 650,350, 650,350, 700,1100,
600,400, 650,350, 650,400, 650,350, 700,350, 600,400, 650,350, 650,400,
650,350, 650,400, 650,350, 650,400, 600,400, 650,350, 650,350, 700,350,
650,400, 650,350, 650,400, 600,400, 650,1100, 600,400, 650,350, 700,1050};
// PANASONIC C4D3: 64800004 unsigned int addr = 0xC4D3; unsigned int data =
0x64800004; I have tried to send it through the IRremote library that I
don't get an answer, I can't turn on my A / C. Any help could help me and
if I am making a mistake and you appreciate it, please tell me. I am new at
this.

AnalysIR, 2020-02-13:
2 things...
1. The solution to your issue has been covered many times here & a good search would help you find the solution.
2. Did you you read my previous comment from yesterday or did you just decide to ignore it completely?

antonmayor, 2020-02-13:
No, do not ignore your comment.
And I have searched enough but I have not been able to solve my
problem with any of the things I find.