Arduino-IRremote/Arduino-IRremote #482: Forum idea

z3t0, 2017-06-22:
I think we discussed this earlier but never really decided anything. 

I think we would benefit from having a forum that is separate from the issues. The Issue tracker can focus on bugs, additions and requests while the forum can focus on support. What do you think?

AnalysIR, 2017-06-22:
@z3t0  This is one of those questions, where the only real answer is YES....but I guess there will always be one who will say no, but not me :)

I agree, the issues have been getting clogged by discussion topics.

bengtmartensson, 2017-06-22:
I would love to have a forum for remote control issues, like the ones addressed by the current project. I think be better to try to be rather broad, being able to discuss also other project with similar scope, like my projects. I am potentially ready to invest some resources too. 

Just having a discussion forum for "z3t0/Arduino-IRremote" does not sound that attractive to me, Frankly speaking, the discussions and the support questions do not seem to be a killer argument.

z3t0, 2017-06-22:
Okay I think then a general forum for discussion about IR remote will suit our needs. In terms of resources, I have a free digital ocean droplet that isn't doing anything. Though I don't have a domain.

We need to decide on what we want to call this forum and what software we want to use.

z3t0, 2017-06-22:
http://flarum.org/  Seems easy to setup but from what I can tell it does not have categories which may be a deal breaker.

Or it might not be an issue at all since we are focusing on a specific topic ie remote control instead of just more general microcontrollers.

z3t0, 2017-06-22:
Though phpbb seems to be a more solid platform.

z3t0, 2017-06-30:
Here's something I went ahead and created http://104.131.122.186:8000

It's only a proof of concept more than anything and the main requirements for whatever software package used is that the forum has to be low maintenance, at least technically.

Demo account:
user: admin
pass: testingpassword

Or you can try making your own account.

bengtmartensson, 2017-06-30:
Looked quite neat when I looked at it. Is that on a private server? Anyhow, at the time of this writing, it is "pining for the fjords" (Pythonian reference :-)) however, it answers ping.

z3t0, 2017-07-01:
Thanks, yes it is on a private server.

> "pining for the fjords" (Pythonian reference :-)) however, it answers ping.

Sorry I'm not too well versed in Pythonian! But I'm sure it was a good joke...

z3t0, 2017-07-01:
If we like the forum used in that I will go ahead and spend some time setting it up. Otherwise I am still open to finding other options if thats what we want. The main motivation for using that in particular is that I am familiar with js and python, not so much php which is used by most other frameworks.

bengtmartensson, 2017-07-01:
> "pining for the fjords" 

[The dead parror sketch](https://www.youtube.com/watch?v=4vuW6tQ0218) 1:39 

> The main motivation for using that in particular is that I am familiar with js and python, not so much php which is used by most other frameworks.

The implementation language is probably secondary unless you plan on "hacking" it. Having said that, I also know/like Python better than PHP, which has the reputation of "Open source version of Visual Basic". Not sure if that should be considered a valid argument...

z3t0, 2017-07-01:
Well I don't want to be modifying any source code at all for the forum but I can certainly imagine it growing to the point where we may need to make our own customisations. 

The issue I do have with these forum systems however is that they are newer and far less used/tested and not nearly as stable. Ideally I should be able to set it up once and not worry about it again.

z3t0, 2017-07-01:
Okay so I decided i would rather invest the time in something stable. http://104.131.122.186/viewforum.php

probonopd, 2017-07-04:
Hi, I would start to consider using it if I could log in using GitHub credentials...

z3t0, 2017-07-05:
I will look into that @probonopd thanks for the suggestion!

z3t0, 2017-07-05:
@probonopd Done!

z3t0, 2017-07-06:
The forum seems to be crashing from MySQL randomly... will investigate

bengtmartensson, 2017-07-09:
> The forum seems to be crashing from MySQL randomly... will investigate

Yess,,, Any progress?

z3t0, 2017-07-12:
I seem to have made it worse... Now it's a 403 forbidden

z3t0, 2017-07-12:
Oh no.. got it working again!

probonopd, 2017-07-12:
Not for me...

> SQL ERROR [ mysqli ]
> 
> No such file or directory [2002]

Seriously, better don't try to host a forum yourself. It's no fun...

closed 2020-10-17 by ArminJo

bengtmartensson, 2020-10-17:
Somewhere I read that Github will add a "discussion forum feature" next year. Would be great; right now I really miss a possibility to separate support and beginner's questions from development-

probonopd, 2020-10-17:
Yes, I think the feature is already in Beta.

ArminJo, 2020-10-17:
I am looking forward too and can adapt the issue template if it is available..