Arduino-IRremote/Arduino-IRremote #483: Code is blocking my outputs?

chevalierc, 2017-06-28:
**Board:** ARDUINO UNO
**Library Version:** 2.1.0
```

#include <IRremote.h> // Include the IRremote library

const uint16_t BUTTON_A = 0xF807;
const uint16_t BUTTON_B = 0x7887;
const int RECV_PIN = 11;
const int OUT_A = 3;
const int OUT_B = 5;
const int FEEDBACK_INTENSITY = 60;
const int FEEDBACK_REPEAT = 2;
const int FEEDBACK_LENGTH_ON = 500;
const int FEEDBACK_LENGTH_OFF = 500;

/* Connect the output of the IR receiver diode to pin 11. */

/* Initialize the irrecv part of the IRremote  library */
IRrecv irrecv(RECV_PIN);
decode_results results;
bool is_on = false;

void setup(){
  Serial.begin(9600); // Use serial to debug. 
  irrecv.enableIRIn(); // Start the receiver
  analogWrite(OUT_A, 0);
  analogWrite(OUT_B, 0);
  turn_on_feedback(OUT_A);
}

void turn_on_feedback(int output_location){
  Serial.println(output_location);
  if(!is_on){
    is_on = true;
    for(int i = 0; i < FEEDBACK_REPEAT; i++){
      Serial.println("IS ON");
      analogWrite(output_location, FEEDBACK_INTENSITY);
      delay(FEEDBACK_LENGTH_ON);
      analogWrite(output_location,0);
      Serial.println("IS OFF");
      delay(FEEDBACK_LENGTH_OFF);
    }
    delay(FEEDBACK_LENGTH_OFF);
    is_on = false;
  }
}

void loop() {
  if (irrecv.decode(&results)){
      uint16_t resultCode = (results.value & 0xFFFF);
      switch (resultCode){
        case BUTTON_A:
          Serial.println("A");
          turn_on_feedback(OUT_A);
          break;
        case BUTTON_B:
          Serial.println("B");
          turn_on_feedback(OUT_A);
          break;
      }
      irrecv.resume(); 
  }
}

AnalysIR, 2017-06-28:
..might be a good idea to provide more detailed info!

...a quick look doesnt show any use of pinMode to set pins as output etc. (OutA&OutB)

chevalierc, 2017-06-28:
Sorry my bad. I did actually edit my `setup` to call `pinMode(OUT_A, OUTPUT);
  pinMode(OUT_B, OUTPUT);` after calling `irrecv.enableIRIn(); `. Im trying to run some 3.3V feedback vibrators through the outputs. They will buzz if I comment out `irrecv.enableIRIn(); ` but do not do so, if I have that method called. Is there any reason they would block output? All the expected serial outs are printed.

chevalierc, 2017-06-28:
Hi, I'm not sure why but switching from port 3 to port 6 worked for me. Is there a documented reason for this?

closed 2017-06-28 by chevalierc

AnalysIR, 2017-06-28:
there is no port 3...I presume you mean pin 3 or D3.

Yes it could be related to the timers being used, which impact only certain pins