Arduino-IRremote/Arduino-IRremote #485: Demo code won't run on ATTiny85

citobori, 2017-07-02:
**Board:** ATTiny85 Digistump clone
**Library Version:** 2.1.0, 2.0.1 and latest
**ATTinyCore Version:** latest
**Protocol:** JVCPanasonicSendDemo
**Arduino IDE:** 1.6.11 on Win7
**IR output analyzed on:** IrScrutinizer

**Code Block:**
```c

#include <IRremote.h>
 
#define PanasonicAddress      0x4004     // Panasonic address (Pre data) 
#define PanasonicPower        0x100BCBD  // Panasonic Power button

IRsend irsend;

void setup()
{
}

void loop() {
  irsend.sendPanasonic(PanasonicAddress,PanasonicPower); // This should turn your TV on and off
  while(1) { } // halt
}
```
Firstly, thanks for this great library! It works brilliantly on my Arduino Micro.

I've been doing some testing of various combinations of versions of this library and can't get code to run on the Tiny85.
I started off with larger project code (sequence of transmitted IR codes) that compiles and runs fine on a genuine Arduino Micro (ATMega32u4). This longer code compiles but hangs on the Tiny85 after a single nonsense transmit.

The Arduino 'Blink' demo compiles and runs fine on the Tiny85. I'm using a USBASP to program.
 
When running the above, stripped down version of the JVC/Panasonic sending demo transmit code on the Tiny85 the captured output logged by IrScrutinizer is just an intro of +5560 -65535 at 21kHz (should be 37 so I'm guessing this is a nonsense code)

Has anyone got a Tiny85 to transmit **any IR signal at all**?
How can I get this working?
Is there anything that needs editing in .H or .CPP files? I'm using them as downloaded. 

Any help would be much appreciated!
cito

z3t0, 2017-07-03:
This is library does support the platform. It's hard to say exactly what might be going on but the first thing to trouble shoot would be the frequency. 

What is the CPU frequency and what voltage is the micro controller running on?

citobori, 2017-07-03:
Thanks z3to. The clock is set to 8MHz internal in the Arduino IDE and bootloader is burned to set the fuses.

The 'Blink' example provided in the IDE (led on, wait 1sec, led off, wait 1 sec, repeat) compiles and runs. It blinks at the correct rate (well, after a minute of blinking compared to a clock it's fast by about 1 second).

The Tiny85 is a 20SU model running at 5V from USB.

z3t0, 2017-07-03:
Okay thanks for the information.

Most likely the issue is with the code [here](https://github.com/z3t0/Arduino-IRremote/blob/master/boarddefs.h#L522). 

I'll look through that.

Which pin are you connecting the transmitter to?

z3t0, 2017-07-03:
@citobori After spending some time cross referencing with the data sheet I think the code, or at least the part that sets the frequency, is correct.

Can you confirm that you are using the pin indicated in the diagram below.

[image]

citobori, 2017-07-04:
Yes I can confirm that the LED is on PB1.
Schematic for the Digispark board from Digistump is here:
[https://s3.amazonaws.com/digistump-resources/files/97a1bb28_DigisparkSchematic.pdf](url)

There is a red LED on the board and the IR LED is connected in parallel as required, so I can visibly see that it is transmitting (or not).

AnalysIR, 2017-07-04:
> There is a red LED on the board and the IR LED is connected in parallel as required, so I can visibly see that it is transmitting (or not).

It is likely, that you may have an issue there, It is likely that either one of the LEDs will dominate to the detriment of the other. My guess is that the IR LED would dominate...but then the (Red LED) resistor value would be way too high to get any useful IR.

Try connecting the IR LED to ground via a 100R resistor (or 68R) and see does it improve things. You may get the available current shared between both LEDs. (= shorter range)

NB: It would **NOT** be a normal approach to wire up in parallel like this.

You could try to wire in an transistor driver circuit, as a more permanent solution.

citobori, 2017-07-05:
Yes I had considered that having 2 LED's on one pin could be causing overload and brownout the chip. I believe the max current per I/O pin is 40mA.

The onboard surface-mount red LED has a 1k resistor in series with it (R5 on schematic). The IR LED has a 270-ohm resistor soldered in series with it and this combo is connected between ground and PB1 when required. I doubt these draw more than 20mA combined.

The Blink sketch runs fine with both LEDs connected.

With or without the IR LED connected, my project sketch (a sequence of transmit codes with 1 second delay between) still hangs on the first transmit. I can see this on the red LED.

With the IR LED connected the demo sketch above gives the nonsense output as received by IrScrutinizer. IrScrutinizer receives fine from standard IR remote controls or when I run sketches on the Arduino Micro.

The IR LED is pointing away from the PCB at rightangles so I doubt there is any IR interference coming from the red LED.

AnalysIR, 2017-07-05:
'**If**' your circuit is working 100%, you would be getting 14mA thru the IR LED & 3mA thru the Red Led.

14mA is way too Low.

My previous comments are still valid. 

Try to eliminate the attiny/IRremote, by testing on the Arduino Micro with 2 LEDs in parallel (=identical circuit). Other than trying to eliminate issues, it will be difficult to provide meaningful assistance, based on your setup, unless someone wants to or can easily replicate it.

bengtmartensson, 2017-07-05:
@citobori ; As @z3t0 said

> This [is] library does support the platform.

What you are doing should be termed "porting the library to a presently unsupported platform". It is (at least a priori) not at all a simple task, and it requires (again, at least a priori) considerable experience and knowledge.

> the captured output logged by IrScrutinizer is just an intro of +5560 -65535 

This means that a pulse of 5560 micros was detected, and then nothing more. "-65535" is the ending time out (assuming you are using GirsLite or similar on the capturer). Most likely, the ATtiny just bails out (running out or stack or something) after the first flash. But this shows that it _is_ capable to transmit detectable IR.

> 14mA is way too Low.

While it may be to low for living room use, it is not to low to detected, or use with a stick-on emitter.

z3t0, 2017-07-06:
@bengtmartensson I thought we did have support for ATTiny85?

bengtmartensson, 2017-07-06:
@z3t0 : Ooops. I stand corrected. Sorry for the confusion.

AnalysIR, 2017-07-06:
Some follow ups...
Although the library supports ATtiny85....I believe that is with the standard Arduino core for that platform. Digistump may be using a different or custom core...which could introduce some complications that have not been tested with the implementation in the library.
For example, does digistump core use the same timers? (probably) The OP said he uses USBASP, so then he is most likely using the standard Arduino core for 85 & not the one from Digistump.

> But this shows that it is capable to transmit detectable IR.

Not neccessarily....noise, glitches, poor quality receiver, badly formed carrier  or inaccurate carrier frequency are capable of generating one mark in a receiver. For example, some receivers are designed to turn off if they detect signals outside their design specs. _In theory, a red led could generate such a mark close up. (I think I tried that before and couldnt get it to work with a quality receiver...I must try it again sometime with a poor quality receiver)_

>  or use with a stick-on emitter.

..which is not being used here AFAIK....and 'living room use' is what I assume based on the info posted.

citobori, 2017-07-06:
Thanks for the responses everyone.
To answer:
@bengtmartensson As a receiver I use a genuine Arduino Uno R3 running the awesome Girslite :)

@AnalysIR 

> 14mA is way too Low.

It's enough to send a recognisable signal from the Arduino Micro a couple of metres to my CD player

> Try to eliminate the attiny/IRremote, by testing on the Arduino Micro with 2 LEDs in parallel (=identical circuit). Other than trying to eliminate issues, it will be difficult to provide meaningful assistance, based on your setup, unless someone wants to or can easily replicate it.

Yes the Micro also has on onboard LED on the default pin (pin 13) which flickers when transmitting sequences. With the IR LED and 270R connected to pin 13 the signal goes at least 2 metres to the CD player.

> ...most likely using the standard Arduino core for 85 & not the one from Digistump

There is no Tiny85 support in the Arduino IDE so ATTinyCore has been added to the Arduino IDE, along with z3t0's library. **No changes have been made to any config files.** This all works fine with the Arduino Micro (32u4). Do I need any config changes in this IR library for the Tiny85 to work?

There is no Digistump core, micronucleus bootloader or other party's code being used apart from the ATTinyCore. The board is a clone and the Atmel was seemingly blank. I can compile and burn sketches to the Tiny board using USBASP programmer wired to the SPI pins, such as the IDE's built-in Blink and Fade examples, which illuminate the onboard LED as expected.

So, moving forward - does anyone have a known working IR sketch that I can try on the Tiny85?
If so can you post it, along with what version of IDE, ATTinyCore and IRremote you are using?

schizobovine, 2018-01-01:
I have also been unable to get IRRemote to work with ATTiny85's. I've tried two different programmers, all three internal clock frequencies available in the Arduino 1.8 UI (1MHz, 8MHz, and 16MHz), and had zero luck getting anything to pick up the signals. Same code works just fine on an ATMega328 at 8MHz/3.3V.

(Aside: I cannot use an ISP to program the Tiny85 while the IR LED is in circuit connected to PB1; this is likely due to my laziness and not using a resistor in series with it (I was trying to figure out if it was just too dim). But if you get errors trying to upload, try removing the LED, uploading, and then putting it back in. And/or see if the 270Ohm resistor @citobori is using works better for you.)

The LED is firing, as jamming a visible one in I can see it flash when the IR code is supposed to be sent (I put it in setup so I could just keep hitting reset to test it with a constant message). My guess is that something with the timing is borked; maybe the internal RC oscillator isn't close enough to what frequency we're expecting?

ArminJo, 2020-10-11:
At least [IRMP ](https://github.com/ukw100/IRMP)is running fine for ATtinies and Digispark.

closed 2020-10-11 by ArminJo