Arduino-IRremote/Arduino-IRremote #487: Sending on 2 LEDs Ardunio Nano

bubblesses, 2017-07-06:
**Board:** ARDUINO Nano ATMEGA328
**Library Version:** 2.1.0
**Protocol:** Sony (if any)

Hello,

I am attempting to use 2 separate infrared LED's with one Arduino Nano. Each one will be sending separate signals at different times. They will not be transmitting at the same time, but one right after another. I see in readme.md that pins 3 & 9 can be used for transmitting, but pin 3 is used by default when using for example, `irsend.sendSony(data, size)`. How would one go about sending on pins 3 & 9 consecutively? Is there some easy way to specify which pin to use when sending? I feel like I'm missing something. 

Any guidance at all would be greatly appreciated.

Thanks, 
Bublesses

bubblesses, 2017-07-06:
Found a solution here: https://github.com/z3t0/Arduino-IRremote/issues/218

z3t0, 2017-07-06:
It should be fairly easy to modify and add support for extra receivers. Essentially you will just be duplicating parts of the code as was done [here](https://github.com/z3t0/Arduino-IRremote/issues/128)

AnalysIR, 2017-07-06:
@bubblesses 
I am curious to understand why you need to send on 2 different IR LEDs/Pins?

Maybe there is an easier way to solve this.

bubblesses, 2017-07-06:
It's a little complicated to explain, but I'll take a stab at explaining it.  Essentially I am creating a 'people counter' that will track the number of people that enter/exit a room. It uses 2 infrared beams, each sending its own code across to the other side, and by checking which beam was broken first, it can be determined whether the person was entering or exiting. They must be physically separated so that the beams are broken at different times, and must send different codes so that the receiving side knows which transmitter sent the code it is receiving. 

Does that make sense?

bubblesses, 2017-07-06:
@z3t0 I actually need to [i]transmit[/i] using 2 separate LEDs, the receiving side of the project is fine. The solution I am considering now is to just use 2 separate Arduino Nanos, one for each beam. I have poked around the code, and it seems the library is not really designed for something like what I described above.

bubblesses, 2017-07-06:
Thanks for the thorough reply! I'll address each of the points you made:

1. & 2. I'll check this out, thanks! 

4. & 6. It is actually not a continuous stream, but data pulses sent rapidly, alternating back and forth between the 2 beams. There are 2 receivers which each receive both data pulses, but the program only searches for one specific pulse on one receiver. This also means that the beams do not need to be focused, as the data pulses are filtered on the receiving side. And I don't have to worry about any other remote control interference. No data pulses for a given period of time = beam is blocked = somebody is enetering/exiting.  

5. Interesting, I hadn't considered this! Thanks for the suggestion. 

I am currently using 2x TSOP53338 receivers at 38khz. Would the one you mentioned have advantages over this one?

AnalysIR, 2017-07-06:
TSSP4038 (or the 56kHZ variant) is a superior solution for this application, but check the data sheet as there are different distance ranges. If your existing setup is working OK, then use it.

You could also just drive 2 IR LEDs in series from the same pin and alternate the data value, which can be decoded on the far side. _Even better use an incrementing counter, so you can easily see the number of missed packets._
For a DUAL IR emitter circuit see: [Constant current infrared LED emitter circuit](https://www.analysir.com/blog/2013/11/22/constant-current-infrared-led-circuit/)

You could also use logic gates or transistors to enable /disable alternating beams if you wished.(Driven from another pin).

bubblesses, 2017-07-06:
Great, thanks for all the info. It's greatly appreciated! Closing this issue now.

closed 2017-07-06 by bubblesses