Arduino-IRremote/Arduino-IRremote #489: Is it possible to capture codes for Bluetooth devices?

wmweis62, 2017-07-08:
**Board:** ARDUINO UNO
**Library Version:** 2.1.0
**Protocol:** Sony (if any)

**Code Block:**
```c

#include <IRremote.h> 

.....

```

Use [a gist](gist.github.com) if the code exceeds 30 lines

**checklist:**
- [] I have **read** the README.md file thoroughly
- [] I have searched existing issues to see if there is anything I have missed.
- [] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [] The title of the issue is helpful and relevant 

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

The above is a short template allowing you to make detailed issues!

AnalysIR, 2017-07-08:
IRremote does not support bluetooth.

However, if the bluetooth device ends up generating IR somewhere, then it should be possible.

I would be surprised if the manufacturer had an 'open' interface on their BT solution. Often, they do not.

Just get someone to look at the remote thru a phone camera and if it is IR, they will see flashes of bright light when looking thru the camera at the remote (close up). If you can get a model number, you may be able to research before visiting.

I guess the other thing to note....is whether the target device can accept IR (regardless of the remote). It could be dual with both IR & BT.

The target device might also have a spare port/connector for control, which you may be able to hack into.

However, for this application I suspect safety, robustness & reliability is paramount. Often the best hack is the simplest. For example, by getting a spare BT remote, it should be possible to open it up and connect directly to the button pads on the inside of the remote(for control by the arduino) & incorporate this 2nd remote into your own project box. This way most complexities are removed.

If the solution is to be deployed in a hospital or similar location, you may also run into other issues with approvals, insurance etc.

wmweis62, 2017-07-08:
Thanks for the reply. I learned about that trick to point a camera at the LED on a remote after I had the opportunity to see the one for the bed. I will be at their home tomorrow and will know the answer to all this in 24 hours, but as I started thinking about the limitations of someone laying in bed using a remote vs a Bluetooth device, I started thinking I may be facing a Bluetooth device and not a IR device. I was not expecting the IRremote library to help me, but was looking for an equivalent. 

I have considered connecting my ESP8266 into the hand held remote, and perhaps that will end up being my best option, at least for this family. With safety considerations in mind, I modified my program to only execute a bed movement command for 3 seconds. In other words, a Head Up command would run 3 seconds and stop. Due to latency associated with each Google Home command, it made sense to use timed vs separate commands for start and stop. It also reduces the likelihood that I would be driving the same part of the bed in two directions at once. (If I really have to use the approach of a wired connection to the remote, I will add another safety feature which stops all motion before initiating another).

The bigger picture I am trying to help is to define a solution that many families would benefit from. Although it is fun and rewarding to hack something together for a one-off situation, I would find it much more impactful to move this to an out-of-the box experience so many non technical families with a similar need can go buy a Echo or Google Home and a Harmony hub and have a working solution.

closed 2020-10-12 by ArminJo