Arduino-IRremote/Arduino-IRremote #490: Arduino Leonardo Read / Send in one sketch

hernae, 2017-07-19:
**Board:** ARDUINO LEONARDO
**Library Version:** 2.1.0
**Protocol:** Panasonic, Nec

Hey,

in my setup my Arduino reads IR-Signals (Panasonic-Code) and sends IR-Signals (NEC-Code). If the Panasonic value for VolumeUp is read, the NEC value for VolumeUp is send etc.

My code is working with my **Arduino Uno perfectly**. Now I am trying to get it done with an Arduino  Leonardo and nothing seems to work. If I isolate reading and sending in two sketches it seems to work again. Is this a known issue for the Leonardo?
 
To get the sending feature to work I had to change the boarddefs.h file. See 
https://github.com/z3t0/Arduino-IRremote/issues/12

Sketch: https://gist.github.com/hernae/6f4c2b7a897cbff67f916680507bb968

hernae, 2017-07-19:
Arduino Pro Mini works as well. Something is fishy with this [Leonardo](https://i.ytimg.com/vi/5aty1FauBAo/hqdefault.jpg)

AnalysIR, 2017-07-19:
Please check the README, the pins for sending IR are different on the Leonardo vs UNO...

...although based on your post that may not be the issue?

hernae, 2017-07-20:
I just found out that the sending behavior of my leonardo changes after flashing. 

First I flash the leonardo. The ir codes that are transmitted are correct. After **unplugging** it from usb and **replugging** it again, the ir codes are suddenly entirely different.

AnalysIR, 2017-07-20:
Sounds like either a hardware issue or more likely a (Windows?) driver related issue.

Please do some searching/troubleshooting wrt the serialUSB driver issues for your leonardo.
_(I seem to recall similar issues in the distant past, when the wrong driver was being associated with the Leonardo on Windows)._

Based on info in your last post, it doesn't appear to be related to IRremote & I assume the original issues you posted are now resolved.

closed 2020-10-17 by ArminJo