Arduino-IRremote/Arduino-IRremote #492: Same button different rawdata

Gherma00, 2017-07-24:
**Board:** ARDUINO NANO
**Library Version:** 2.1.0

Hi everybody, I have a problem to capture a Mitsubishi air conditioner remote controller codes. 
I get same Hex code and different rawdata every time I press power button(but also for any button). 
It is encoded as SANYO, and I don't know how use irsend.sendRaw method when RawDate appear to change at random.
Naturally it doesn't work with ac.
The output of irRecvDumpV2 sample:

POWER ON (24°C cold):
```
Encoding  : SANYO
Code      : FFFFFFFF (0 bits)
Timing[129]: 
     + 500, -1450     + 550, -3400     + 500, -1450     + 550, -3400
     + 500, -1500     + 500, -1450     + 500, -1450     + 500, -1500
     + 500, -1450     + 500, -1450     + 500, -1500     + 500, -1450
     + 500, -1450     + 500, -1500     + 500, -1450     + 500, -1450
     + 500, -1500     + 500, -1450     + 500, -1500     + 500, -3400
     + 500, -1500     + 450, -1500     + 500, -1450     + 500, -3450
     + 500, -1450     + 500, -1500     + 450, -1500     + 500, -1450
     + 500, -1500     + 500, -1450     + 500, -3400     + 550, -1450
     + 500, -3400     + 500, -1500     + 500, -3400     + 550, -1450
     + 500, -3400     + 550, -3400     + 500, -3450     + 500, -3400
     + 550, -3400     + 500, -3450     + 500, -3400     + 550, -3400
     + 500, -3450     + 500, -3400     + 550, -3400     + 500, -3450
     + 500, -3400     + 550, -3400     + 500, -3450     + 500, -1450
     + 500, -3450     + 500, -3450     + 500, -3400     + 500, -1500
     + 500, -3400     + 500, -3450     + 500, -3450     + 500, -3400
     + 500, -3450     + 500, -3450     + 500, -1450     + 500, -3450
     + 500
unsigned int  rawData[129] = {500,1450, 550,3400, 500,1450, 550,3400, 500,1500, 500,1450, 500,1450, 500,1500, 500,1450, 500,1450, 500,1500, 500,1450, 500,1450, 500,1500, 500,1450, 500,1450, 500,1500, 500,1450, 500,1500, 500,3400, 500,1500, 450,1500, 500,1450, 500,3450, 500,1450, 500,1500, 450,1500, 500,1450, 500,1500, 500,1450, 500,3400, 550,1450, 500,3400, 500,1500, 500,3400, 550,1450, 500,3400, 550,3400, 500,3450, 500,3400, 550,3400, 500,3450, 500,3400, 550,3400, 500,3450, 500,3400, 550,3400, 500,3450, 500,3400, 550,3400, 500,3450, 500,1450, 500,3450, 500,3450, 500,3400, 500,1500, 500,3400, 500,3450, 500,3450, 500,3400, 500,3450, 500,3450, 500,1450, 500,3450, 500};  // SANYO FFFFFFFF
unsigned int  data = 0xFFFFFFFF;
```
POWER OFF
```
Encoding  : SANYO
Code      : FFFFFFFF (0 bits)
Timing[129]: 
     + 500, -1450     + 550, -3400     + 500, -1450     + 550, -3400
     + 500, -1450     + 550, -1450     + 500, -1450     + 500, -1500
     + 500, -1450     + 500, -1450     + 500, -1500     + 500, -1450
     + 500, -1450     + 500, -1500     + 500, -1450     + 500, -1500
     + 450, -1500     + 500, -1450     + 500, -1500     + 450, -3450
     + 500, -1500     + 500, -1450     + 500, -1450     + 500, -1500
     + 450, -1500     + 500, -1450     + 500, -1500     + 500, -1450
     + 500, -1450     + 500, -1500     + 450, -3450     + 500, -1450
     + 500, -3450     + 500, -1450     + 550, -3400     + 500, -1450
     + 500, -3450     + 500, -3450     + 500, -3400     + 550, -3400
     + 500, -3450     + 500, -3400     + 550, -3400     + 500, -3450
     + 500, -3400     + 550, -3400     + 500, -3450     + 500, -3400
     + 550, -3400     + 500, -3450     + 500, -3400     + 550, -1450
     + 500, -3400     + 550, -3400     + 500, -3450     + 500, -3450
     + 500, -3400     + 500, -3450     + 500, -3450     + 500, -3400
     + 550, -3400     + 500, -3450     + 500, -1450     + 500, -3450
     + 500
unsigned int  rawData[129] = {500,1450, 550,3400, 500,1450, 550,3400, 500,1450, 550,1450, 500,1450, 500,1500, 500,1450, 500,1450, 500,1500, 500,1450, 500,1450, 500,1500, 500,1450, 500,1500, 450,1500, 500,1450, 500,1500, 450,3450, 500,1500, 500,1450, 500,1450, 500,1500, 450,1500, 500,1450, 500,1500, 500,1450, 500,1450, 500,1500, 450,3450, 500,1450, 500,3450, 500,1450, 550,3400, 500,1450, 500,3450, 500,3450, 500,3400, 550,3400, 500,3450, 500,3400, 550,3400, 500,3450, 500,3400, 550,3400, 500,3450, 500,3400, 550,3400, 500,3450, 500,3400, 550,1450, 500,3400, 550,3400, 500,3450, 500,3450, 500,3400, 500,3450, 500,3450, 500,3400, 550,3400, 500,3450, 500,1450, 500,3450, 500};  // SANYO FFFFFFFF
unsigned int  data = 0xFFFFFFFF;
```

POWER ON (24°C cold):
```
Encoding  : SANYO
Code      : FFFFFFFF (0 bits)
Timing[129]: 
     + 500, -1450     + 500, -3450     + 500, -1450     + 550, -3400
     + 500, -1450     + 550, -1450     + 500, -1450     + 500, -1450
     + 550, -1450     + 500, -1450     + 500, -1450     + 550, -1450
     + 500, -1450     + 500, -1450     + 550, -1450     + 500, -1450
     + 500, -1500     + 500, -1450     + 500, -1450     + 500, -3450
     + 500, -1450     + 500, -1450     + 550, -1450     + 500, -3400
     + 550, -1450     + 500, -1500     + 450, -1500     + 500, -1450
     + 500, -1500     + 450, -1500     + 500, -3400     + 500, -1450
     + 550, -3400     + 500, -1500     + 500, -3400     + 500, -1500
     + 500, -3400     + 500, -3450     + 500, -3450     + 500, -3400
     + 500, -3450     + 500, -3450     + 500, -3400     + 500, -3450
     + 500, -3450     + 500, -3400     + 550, -3400     + 500, -3450
     + 500, -3400     + 550, -3400     + 500, -3450     + 500, -1500
     + 450, -3450     + 500, -3400     + 550, -3400     + 500, -1450
     + 550, -3400     + 500, -3450     + 500, -3400     + 550, -3400
     + 500, -3450     + 500, -3400     + 550, -1450     + 500, -3400
     + 550
unsigned int  rawData[129] = {500,1450, 500,3450, 500,1450, 550,3400, 500,1450, 550,1450, 500,1450, 500,1450, 550,1450, 500,1450, 500,1450, 550,1450, 500,1450, 500,1450, 550,1450, 500,1450, 500,1500, 500,1450, 500,1450, 500,3450, 500,1450, 500,1450, 550,1450, 500,3400, 550,1450, 500,1500, 450,1500, 500,1450, 500,1500, 450,1500, 500,3400, 500,1450, 550,3400, 500,1500, 500,3400, 500,1500, 500,3400, 500,3450, 500,3450, 500,3400, 500,3450, 500,3450, 500,3400, 500,3450, 500,3450, 500,3400, 550,3400, 500,3450, 500,3400, 550,3400, 500,3450, 500,1500, 450,3450, 500,3400, 550,3400, 500,1450, 550,3400, 500,3450, 500,3400, 550,3400, 500,3450, 500,3400, 550,1450, 500,3400, 550};  // SANYO FFFFFFFF
unsigned int  data = 0xFFFFFFFF;
```

All power-on commands are differents, which "rawdata" i should consider....?

Thank you again.

z3t0, 2017-07-24:
Try searching existing and closed issues for "Ac" or "air conditioner " where there are suggestions that might help.

bengtmartensson, 2017-07-25:
A few comments:

> I have a problem to capture a Mitsubishi air conditioner remote controller codes.

As Rafi said, there are a few issues on AC themes, just search.

> I get same Hex code and different rawdata every time I press power button(but also for any button).

Since the "rawdata" are phyical measurements, this is normal, even if the sent signal is "the same".

> It is encoded as SANYO, 

How did you come up with that conclusion? If some SW spat out `sanyo` as decode, it is severely broken... The protocol in the file `ir_Sanyo,cpp` allegedly decodes 12 bits, so your signals are much to long to fit.

I ran your data through [IrpTransmogrifier](https://github.com/bengtmartensson/IrpTransmogrifier) (CI-build) and got
```

Gaps:
B:      1467    = 3*507         96
C:      3421    = 7*507         96
D:      10000   = 20*507        3

Flashes:
A:      507     = 1*507         195

Pairs:
AB:     96
AC:     96
AD:     3

#0:     AB AC AB AC AB AB AB AB AB AB AB AB AB AB AB AB AB AB AB AC AB AB AB AC AB AB AB AB AB AB AC AB AC AB AC AB AC AC AC AC AC AC AC AC AC AC AC AC AC AC AC AB AC AC AC AB AC AC AC AC AC AC AB AC AD
{507,msb}<1,-3|1,-7>(A:32,B:32,1,-20){A=0x50001102,B=0xafffeefd}
weight = 12
#1:     AB AC AB AC AB AB AB AB AB AB AB AB AB AB AB AB AB AB AB AC AB AB AB AB AB AB AB AB AB AB AC AB AC AB AC AB AC AC AC AC AC AC AC AC AC AC AC AC AC AC AC AB AC AC AC AC AC AC AC AC AC AC AB AC AD
{507,msb}<1,-3|1,-7>(A:32,B:32,1,-20){A=0x50001002,B=0xafffeffd}
weight = 12
#2:     AB AC AB AC AB AB AB AB AB AB AB AB AB AB AB AB AB AB AB AC AB AB AB AC AB AB AB AB AB AB AC AB AC AB AC AB AC AC AC AC AC AC AC AC AC AC AC AC AC AC AC AB AC AC AC AB AC AC AC AC AC AC AB AC AD
{507,msb}<1,-3|1,-7>(A:32,B:32,1,-20){A=0x50001102,B=0xafffeefd}
weight = 12
```
so the captures look quite plausible to me. (The first and the third are the same, the second different.)

AnalysIR, 2017-07-26:
> I try to relay rawdata, but nothings works.

Did you follow the suggestions from either of the previous 2 posters? I suspect you will find an answer with one or both of them. If you did, you should post your results.

closed 2020-10-17 by ArminJo