Arduino-IRremote/Arduino-IRremote #493: Every Sec i got Signal

DaveOff, 2017-07-30:
**Board:** ARDUINO Mega
**Library Version:** 2.1.0
**Protocol:** Sony (if any)

hi, when i connect IR Rec to board and run program, every sec i got raw :
where is mistake ? is that noise ?

	Encoding  : SANYO
	Code      : FFFFFFFF (0 bits)
	Timing[25]: 
		 + 300, - 100     + 200, -  50     + 450, - 750     + 100, -  50
		 + 150, - 250     + 900, -  50     +6300, -  50     + 700, -  50
		 +  50, - 200     +  50, -1200     + 350, - 450     + 150, - 150
		 + 150
	unsigned int  rawData[25] = {300,100, 200,50, 450,750, 100,50, 150,250, 900,50, 6300,50, 700,50, 50,200, 50,1200, 350,450, 150,150, 150};  // SANYO FFFFFFFF
	unsigned int  data = 0xFFFFFFFF;

	Encoding  : UNKNOWN
	Code      : 3B50106D (32 bits)
	Timing[17]: 
		 + 950, - 400     + 150, - 150     + 100, -  50     + 500, -  50
		 + 200, -  50     +7800, - 750     + 150, - 350     + 700, - 600
		 + 100
	unsigned int  rawData[17] = {950,400, 150,150, 100,50, 500,50, 200,50, 7800,750, 150,350, 700,600, 100};  // UNKNOWN 3B50106D

bengtmartensson, 2017-07-30:
Yes, that is likely "noise". First thing that I would try (after killing ambient light of course) would be to solder 100 nF (for example) between plus and ground on the receiver.

AnalysIR, 2017-07-30:
You should also post some more info....e.g.

- The pin number you have configured in the sketch for IR Rx
- if you made any changes to the defaults in the library
- the model#/link of the IR receiver being used.
- anything else unusual about your set-up

DaveOff, 2017-07-30:
hi AnalysIR, tnks for reply

My PIN (13 PWM):
	int recvPin = 13;
	IRrecv irrecv(recvPin);
	
and my code is IRrecvDump V2 (from examples)

my IR rec module : KST-RX706 
my Board : Arduino Mega 2650
Picture of Module : https://www.microele.com/491-thickbox_default/-kst-rx706-315mhz.jpg
Picture of Arduino and Module : http://uplod.ir/yskzv8cp1x14/photo_2017-07-30_17-06-32.jpg.htm
Pciture of my Remote for send : http://uplod.ir/wgc06jfwc385/photo_2017-07-30_17-07-18.jpg.htm

AnalysIR, 2017-07-30:
It looks like your IR receiver is in fact an **RF receiver**.

So the problem is that you are using the wrong library & it is not designed for your receiver or ever RF.

I suggest you search for an Arduino remote control RF library...there should be a few of them available.

z3t0, 2017-08-09:
Closing as the issue seems to have been recognised and is not related to this library.

closed 2017-08-09 by z3t0