Arduino-IRremote/Arduino-IRremote #496: Occasional random values when sending data over serial at low baud rates

markasoftware, 2017-08-06:
**Board:** ARDUINO UNO
**Library Version:** 2.0.1
**Protocol:** All

When the serial baud rate is set low (for example, 9600, although I experienced issues until i set it all the way up to 115200), incorrect and seemingly random values will be reported frequently. Even at 115200, this occured occasionally (although rarely enough that it's ok for my uses). This is a problem because the examples use 9600 as the baud rate, triggering this issue.

z3t0, 2017-08-09:
Could you provide a sample sketch and the corresponding output vs what we should be seeing (using higher baud rates)

AnalysIR, 2017-08-09:
It could also be:
- bad wiring /  loose connection
- IR receiver(what model are you using?)link?
- interference from sunlight, wifi, other light sources etc

...as well. Although **if** it works perfectly at 1 or more BAUD rates, the list above is less likely.

bengtmartensson, 2017-08-09:
Let me add to the list:

* Bad power
* Overloading output pins

closed 2020-08-17 by ArminJo