Arduino-IRremote/Arduino-IRremote #498: Arduino 101 Receive support

z3t0, 2017-08-10:
#481 Added receiving support but also implemented some changes that are not compatible with the design choices made in #437 

This issue records the need to refactor the receiving implementation from #481 

@jeffjvick Assuming you still need this, this issue might be relevant.

jeffjvick, 2017-08-11:
Thanks @z3t0.  What I really needed was send support so I had to abandoned the use of IR in the hackathon I was participating in and went a different direction.  (I was required to use the 101)

All of this is moot now though because Intel has dumped making Curie and the 101 recently: http://www.zdnet.com/article/intel-dumps-maker-board-arduino-101-and-curie-chip/

If they wanted more people to use the thing they should have been working better with library maintainers like yourself to provide code and sample hardware support...

bengtmartensson, 2017-08-11:
As a general remark, supporting a certain piece of HW in a *tested* way is a good idea, even if the hardware has been discontinued. 

If the implementation of #481 can be used, and its working and "saneness" validated with a moderate amount of work, I would vote for incorporating it.

z3t0, 2017-08-11:
> If the implementation of #481 can be used, and its working and "saneness" validated with a moderate amount of work, I would vote for incorporating it.

Sounds good, let's go forward with that then.

jeffjvick, 2017-08-12:
Yes, please don't let my comment prevent you from moving forward with incorporating it.  I'm sure there are still plenty of people with these boards.  

I was just venting my frustration of working on a project for an Intel sponsored hackathon where Intel ironically pulled the plug on the very product the hackathon was meant to promote in the middle of the hackathon.  🤣

z3t0, 2017-08-12:
I completely share your sentiment, it really is quite frustrating.

AnalysIR, 2017-08-12:
Probably a good idea to close this, pending a volunteer....unlikely!

:)

z3t0, 2017-08-12:
That's true! There isn't much point trying to support a platform abandoned by its manufacturers.

closed 2017-08-12 by z3t0

bengtmartensson, 2017-08-12:
I got the impression that #481 contained an implementation of the receiving part. Therefore I wrote 

> **If** the implementation of #481 can be used, **and its working and "saneness" validated** with a **moderate amount of work**, I would vote for incorporating it.

(emphasis added), but strictly speaking I neither volunteer nor am I a stakeholder

z3t0, 2017-08-12:
For the time being my plan was to refactor the changes over the beta branch and produce a PR which can remain open until any interested party ie. a community member wants to test it.

Thanks,
Rafi Khan