Arduino-IRremote/Arduino-IRremote #501: Documenting Protocol Implementations

z3t0, 2017-08-13:
This issue documents the need to create a wiki/document explaining the technical specifications of all support. There are a few options for this, we can write the documentation in the existing wiki provided by GitHub for this repository, we could write it using a service such as readthedocs, or we could simply write it in a folder in the existing repository.

z3t0, 2017-08-13:
One thing to note is that I would like the documentation to be written in markdown simply because GitHub has very good support for the format. So for example we could easily add the documentation to the existing Github Pages website for this project.

z3t0, 2017-08-13:
Also sent an email to the arduino developer mailing list hoping to add documentation to https://www.arduino.cc/en/Tutorial/LibraryExamples

bengtmartensson, 2017-08-14:
When documenting a protocols, there are really three parts:

1. The exact definition. If using a "domain specific language"/formalism (like IRP), this is in principle self-documenting. Otherwise a page like [this](http://www.sbprojects.com/knowledge/ir/nec.php) can be written for "every" protocol...
2. Documentation for syntax and semantics of the input parameters of the protocol
3. Other stuff; "comments", links to official documentation.

The framework of IrpTransmogrifier handles this. The xml data base can be transformed to different human readable formats, like (X)HTML. XSLT is a good tool for this. (Actually, double clicking the xml file in a suitable browser invokes an XML transformation internally in the browser, so you really see the HTML version.) It is also possible to write all kind of other filters transforming into other formats, like markdown. 

BTW, markdown is IMHO not a very good choice; it contains very possibilities, and is not really [standardized](https://en.wikipedia.org/wiki/Markdown). It was designed for simple tasks, to be written by (untrained) humans. It the fast food of text formatting... But if you have the technology for "transmogrifying anything to anything"  you are not really tied up :-).

z3t0, 2017-08-17:
My plan was to follow the example of http://www.sbprojects.com/knowledge/ir/nec.php as you linked above. It is a very easy format to read and understand. 

Markdown is probably not ideal but I do not really know much about any other options such as XML / HTML. If you can give an example of something we could use I can read up on it

bengtmartensson, 2017-08-18:
My point in the previous post was to advocate automatic generation of documentation. Manual documentation, with few exceptions, either is not written, or -- if written -- outdated. (see  [IRLib](https://github.com/cyborg5/IRLib) for an example...)

commit a8c24ee4 (2020-12-08): Documented - Adding new protocols closes #501
commit e4517d69 (2020-12-08): Documented - Adding new protocols closes #501
closed 2020-12-08 by ArminJo