Arduino-IRremote/Arduino-IRremote #502: What is the purpose of the Pronto protocol?

z3t0, 2017-08-13:
I am trying to determine what Pronto is and whether it should be removed, or better documented

bengtmartensson, 2017-08-13:
The best explanation of Pronto Hex (sometimes called "CCF") is [the one I have written](http://harctoolbox.org/Glossary.html#ProntoSemantics) :-). You can also see how I have implemented it in Infrared4Arduono: [Pronto.h](https://github.com/bengtmartensson/Infrared4Arduino/blob/master/src/Pronto.h) and [Pronto.cpp](https://github.com/bengtmartensson/Infrared4Arduino/blob/master/src/Pronto.cpp). (just static functions).

The reason I recommended the current implementation for deletion is that the design is unclear and the implementation bad. Another reason for not keeping it is that IRremote, as opposed to the Pronto Hex format, presently do not have a concept of a signal consisting of intro- and repeat signal.

Not that I am "against" Pronto Hex, but I think in the present situation it is cleaner to junk it and to re-implement it when "the smoke has settled". It is really not that hard.

z3t0, 2017-08-14:
It seems a good idea to remove the current implementation then, unless
there are any objections I will get on that once I figure out what I had
messed up in the Travis CI build log.

mataide, 2017-08-16:
Please NO! I work with residential automation and every product in the industry works(**) with pronto hex, the most know site used to get this is http://www.remotecentral.com/index.html, this is the biggest IR Database available. At moment I need to read the every IR code and get the new value, but if the pronto as properly working my application has easily a huge database from day one.

(**) = The software setup only accept Pronto hex.

I can even donate to get Pronto hex improved, this is still heavy used in industry and cant be removed.
I was really happy to see pronto in the commits, but when I saw "delete" omg! Almost a heart attack..

bengtmartensson, 2017-08-16:
No-one questions the long-time value of supporting Pronto Hex. Read my above posting carefully.

If I understand @z3t0  correctly, he is doing these changes in a branch ("beta"), probably to be merged into master when it is "finished". No need to panic.

Question: are you really using the present implementation? *Now?* (If so, can you give an example sketch.) Or you think that it will come in handy in some future?

Also, note that Pronto Hex can easily be converted to "raw format" using an interactive  program like the open-source IrScrutinizer. 

Remote Central is not a data base. It is a large _collection of different files in different file formats_, some of these in proprietary formats.

Possibly we can even help you if you use a more constructive language than "omg", "heart attack", and thumbs-down.

mataide, 2017-08-16:
> Question: are you really using the present implementation? Now? (If so, can you give an example sketch.) Or you think that it will come in handy in some future?

I am not using Pronto Hex at moment because I coudnt find any library that works well with it.

> Also, note that Pronto Hex can easily be converted to "raw format" using an interactive program like the open-source IrScrutinizer.

I already know about IrScrutinizer and others programs that can do it, but doesnt solve my need.
I sell a software + hardware package to the industry and I cant attach a Open Source program to convert Pronto Hex code, because I wont give support for them.

**Alternative Solution:**
I couldnt find any code that translate Pronto Hex to Raw, if Its possible, then I could use it in a built-in function.

> Remote Central is not a data base. It is a large collection of different files in different file formats, some of these in proprietary formats. 

Whatever it is, thats where everyone look to find an IR Code and its all Pronto Hex.

z3t0, 2017-08-17:
@mataide I have noted your concern and have no intention of crippling users out there that need Pronto Hex.

In fact the purpose of this issue is to recognise that the current Pronto Hex implementation is either flawed/untested or just unknown to the current contributors. We intend to implement Pronto Hex properly, not to remove it.

Furthermore, as @bengtmartensson has noted. All changes are being added to the beta branch and you can continue to use the master / releases. This means none of these *development* changes will be exposed to the end user until they have been polished and any glaring issues, such as Pronto Hex, fixed.

seltix5, 2019-12-17:
> Please NO! I work with residential automation and every product in the industry works(**) with pronto hex, the most know site used to get this is http://www.remotecentral.com/index.html, this is the [...]

hello, how do you send the pronto codes? it is not working for me :S thanks!

ArminJo, 2020-07-04:
Lets wait for someone who is interested to put the effort for testing and validating pronto. Currently it is disabled.

closed 2020-07-04 by ArminJo

ArminJo, 2020-07-04:
Whats about using the pronto implementation of https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Pronto.cpp

bengtmartensson, 2020-07-04:
In the interest of code hygiene, I suggest deleting it instead of just "disabling". The junked code is available in Git anyhow.

Possibly adding a few line of documentation why "sending Pronto" is a really bad idea on a board with limited memory, and what to do instead.

ArminJo, 2020-07-04:
I have no experience with Pronto all I know is that it is implemented in the ESP8266 version and in your Infrared4Arduino, so it must not a bad idee to have it available. But as I said, I have no experience.

bengtmartensson, 2020-07-11:
PR #702 implements the requested functionality.  @mataide, I think it is fair to expect a test and a review from you...