Arduino-IRremote/Arduino-IRremote #503: Delay and other time functions not working

MadTooler, 2017-08-15:
I am running the current build of IRremote, 2.0.1, Uno compatible Redboard from Sparkfun, the Sparkfun Ardumoto motor driver, and trying to control my robot as it receives IR signals on pin A1.

I am having trouble with the arduino hanging where a delay() is called after a motor command.  I am using pwm on pins 9 and 10 currently, although I am cheating and have pins 9 jumped to 3 and pins 11 jumped to 10 since the ardumoto wants to use 3 and 11.  Before I played around with jumping the pins around, I did change the Timer in the IRremote library with the identical results of bad delays.

It is assumed to be some kind of timer issue where the delay is started, but never completed, so the robot remains in the state of motors on and never seems to leave the delay.

This does not seem to be the expected result with the library, nor what I recall to be an issue from prior usage.  However, testing with the older 1.0.0 has the same performance.  Is this a known limit of using delays with the library, or have I some other issue? 

Thanks to all that may be able to shed visible light on my issue.

AnalysIR, 2017-08-15:
If I recall correctly, Arduino delay is derived from Timer0. So if you have messed with that it is possible you may have disabled T0, which means delay would never return.

Similarly, your robot library may be messing with T0 or the one used by IRremote.

You need to figure out exactly which timers are being used by the various libraries and ensure there are no conflicts.

I presume you have set 3 & 10 a high impedance inputs or properly isolated such that the 2 shields are messing with each other's pins.

...as a 'hack?' you could try using the in-built AVR ms delay functions, which do not use timers(code loops) instead of the Arduino timer0 based delay()
[AVR delay function](http://www.atmel.com/webdoc/avrlibcreferencemanual/group__util__delay_1gad22e7a36b80e2f917324dc43a425e9d3.html)
 if you need a longer delay, just use a for loop or similar

MadTooler, 2017-08-15:
Thanks for the quick reply.  

I do not believe I have anything messing with T0.  

I only have the one shield.  I simply jumped the pins together, meaning I am not actually using the 3 or 11 arduino pins and have not set them as input or output.  I know this is a hack.  Like I said before, it functions the same as if I were to change the timer in the IR library and keep the original ardumoto pin configuration.  Additionally, the delay() works properly when I do not enableIRIn.

I should also mention I have the pololu QTR-8RC sensor running with the QTRSensors library.  I took a quick look and did not see any mention of timers.

My custom robot control library, GobbitLineCommand (does not directly change any timers), does force me to load the Adafruit_MotorShield library, although I am not using that shield.  It runs on I2c.  I looked around, but did not find any known issues between it and the IR library.  Is there any issue with I2c?

Still trying to strip and debug some of these excess features to determine where the culprit conflict lies.  My great hope was there was a known issue with delay() to explain this, but it does not sound like the case.

MadTooler, 2017-08-16:
Regarding the QTRSensor Library from Pololu, 

> "Internally, this library uses all standard Arduino functions such as micros() for timing and analogRead() or digitalRead() for getting the sensor values, so it should work on all Arduinos without conflicting with other libraries."

MadTooler, 2017-08-16:
Just to make sure, I eliminated the mess of the cross wiring of the motor pwm pins.  Now I am using pins 9(pwm), 10(pwm), 12, and 13.

Still no difference.

I tried _delay_ms(50); with the same results as with standard delay.  I did notice with either delay type, the next IR command was receivable and did change the motor state even when it appeared the loop was stuck in the last delay. 

I am really not sure where else to look for the conflict.  Any other suggestions?

Here is my code being tested.  Without looking inside my library, which is available through the manager, there may be a lot unknown.

```
#define RECV_PIN A0

#include <IRremote.h>
IRrecv irrecv(RECV_PIN);
decode_results results;

#include <GobbitLineCommand.h>
GobbitLineCommand MyBot;

int start = -1;

void setup() {

  MyBot.setQTRpins(2, 4, 5, 6, 7, 8, A2, A3);
  MyBot.setLeftMotorPinsDirPWM(12, 9);
  MyBot.setRightMotorPinsDirPWM(13, 10);

  MyBot.setBatteryVolts(9.6);

  MyBot.beginGobbit();

  MyBot.calibrateLineSensor(0);

  irrecv.enableIRIn(); // Start the receiver

}

void loop() {

  if (irrecv.decode(&results)) {
     
     switch (results.value){
     
      case 0x1F:  // start ir command
        start = 1;
        break;

      case 0x07: // stop ir command
        if (start == -1) MyBot.setMinMaxSpeeds(-100, 100);
        start = 0;
        break;
     }
       
    irrecv.resume(); // Receive the next value
  }

  switch (start) {
    
    case -1:  // initial slower speed as bot approaches start line
      MyBot.setMinMaxSpeeds(-75, 75);
      MyBot.followLine(1);
      break;

    case 1:  // race begins
      MyBot.followLine(1);
      break; 

    case 0:  // stop with quick motor reversal for reduced overrun
      //MyBot.brakeMotors(100);
      MyBot.move(-100, 0);
      delay(10); // **** this is where it sticks forever unless it gets another IR command
      MyBot.move(0, 0);
      break;
  }

}
```

MadTooler, 2017-08-16:
A little further dissection... as soon as I call "irrecv.decode(&results)" the trouble begins.  Before that, I am able to use all of my other library functions and delays without issue.

AnalysIR, 2017-08-16:
@MadTooler nothing obvious jumps out.

Any low SRAM warnings from compiler?

Did you disable all unused IR protocols in IRremote.h...worth a try.

If it happens after/on moving a motor....consider noise getting back into MCU or weak power supply/battery.

MadTooler, 2017-08-16:
No warnings.  Here is compiler response from the code noted earlier:

>Sketch uses 20032 bytes (62%) of program storage space. Maximum is 32256 bytes.
Global variables use 881 bytes (43%) of dynamic memory, leaving 1167 bytes for local variables. Maximum is 2048 bytes.

I think I stepped into what is the cause.  This robot is receiving from another arduino that is also running the IRremote library.  I had some difficulty in determining what the appropriate choice of bit should be for my choice of NEC code (it could be any format, I don't care, only trying to communicate between arduinos).  Here is the sending arduino's code:
```
#define LDR_L A0
#define LDR_R A1 
#define BEEP_PIN 2
#define LEFT_SERVO_PIN 9
#define GREEN_SERVO_PIN 10
#define RIGHT_SERVO_PIN 11

#define FLAG_UP_ANGLE 8
#define FLAG_SWING_ANGLE 40
#define FLAG_DOWN_ANGLE 120

#include <Servo.h>

Servo leftServo; 
Servo greenServo; 
Servo rightServo; 

int ldr_valueL = 0;        //variable to store LDR values

int ldr_valueR = 0;        //variable to store LDR values

byte start =0;

#include <IRremote.h>

IRsend irsend;

void setup()
{

  leftServo.attach(LEFT_SERVO_PIN);  
  greenServo.attach(GREEN_SERVO_PIN); 
  rightServo.attach(RIGHT_SERVO_PIN); 

  leftServo.write(FLAG_UP_ANGLE);
  greenServo.write(FLAG_UP_ANGLE);
  rightServo.write(FLAG_UP_ANGLE);

  delay(2000);

  leftServo.write(FLAG_DOWN_ANGLE);
  greenServo.write(FLAG_DOWN_ANGLE);
  rightServo.write(FLAG_DOWN_ANGLE);

  pinMode(BEEP_PIN, OUTPUT);
}

void loop()
{

  ldr_valueL = analogRead(LDR_L); //reads the Left lane LDR values
  ldr_valueR = analogRead(LDR_R);  //reads the right lane LDR values

  if(ldr_valueL > 400 || ldr_valueR > 400){
    // beep buzzer
    digitalWrite(BEEP_PIN, HIGH);
  
    // send stop signal to robot
    irsend.sendNEC(0x07, 32); // the code value can be anything, not sure what the bit should be

    // set start flag
    start=1;
    
    // wait a little then turn off buzzer
    delay(200);
    digitalWrite(BEEP_PIN, LOW);

    // wait a little while robot sits on start line
    delay(2000);
  }
  
  else digitalWrite(BEEP_PIN, LOW);
 
  if(start){
    // send start signal to robot
    irsend.sendNEC(0x1F, 32);
    
    // wave the start flag
    for(int i=0;i<8;i++){
      greenServo.write(FLAG_SWING_ANGLE);
      delay(200);
      greenServo.write(FLAG_UP_ANGLE);
      delay(200);
    }
    greenServo.write(FLAG_DOWN_ANGLE);
    start=0;
  }
}
```

I just tried changing my code on the receiving end to decode, but not switch/case based upon the value, only perform an act on the motors if a code was received.  I used both a standard cable box remote and the sending arduino, and there were no hangups with either.  This leads me to believe I am maybe sending wrong, or using a finicky IR code, and somehow confusing the receiver that way.  Does my code format for the sending appear wrong, and is this a possibility?

Thanks again.

MadTooler, 2017-08-16:
As a follow up to what I am trying to do, in case there is a clearer method, I am trying to directly communicate between two arduinos.  Is there an IR code protocol or method that is best suited?

AnalysIR, 2017-08-16:
comments 
- please put code inside code tags so it is more readable ( i didn't spend time on it  because of that)
- Get your IR working without any  motor/robot stuff
- NEC is more than adequate, but I would use more imagination with the NEC code try something like 0xA55A5507 instead

MadTooler, 2017-08-16:
> please put code inside code tags so it is more readable ( i didn't spend time on it because of that)

Sorry, I thought using the "Insert Code" was placing it inside code tags.  I will look into what you may mean and how to better use the forum.

I think I figured out what was wrong.  Testing using the longer format of IR code and starting from scratch with only IR stuff forced me to pay a bit closer attention to what I was doing, and I realized I am an idiot... as you probably were guessing.  I was forcing my robot to appear to stay inside a delay loop since it was repeating what was suppose to be a short reversal of the motors, but did not condition it to only do that once.  It would then quickly repeat and run too far away from the IR transmitter to be in range to receive the next command.  I should have caught that far sooner, but got myself turned around since other commands were having similar issues, no doubt for the same related error on my part.  Sorry to be such a knucklehead.

However, if I could find some clear documentation as to the sending format, that would be very helpful.  It is not clear, from what I have found, what NEC codes are usuable and what "bit" size should be declared (16 and 32 appeared to work but I did find a note by Ken that it should always be 32).  

If you have any direction to documentation on the full length IR code format you suggested, please pass them along.

I see some of you are planning a new revision where documentation appears to be a priority.   That is very good to hear.

Thanks again.

z3t0, 2017-08-16:
> However, if I could find some clear documentation as to the sending format, that would be very helpful. It is not clear, from what I have found, what NEC codes are usable and what "bit" size should be declared (16 and 32 appeared to work but I did find a note by Ken that it should always be 32).

Sorry for the poor state of documentation! As a community run project we don't have a lot of time polishing and what little time we do have is spent on bug fixing and adding features. However, we are currently working on improving the available user documentation

The current NEC implementation can be found [here](https://github.com/z3t0/Arduino-IRremote/blob/master/ir_NEC.cpp) and a cursory look shows that it supports 32 bits. Perhaps if you can give us a list of the kinds of commands you would like to send eg. Power on / off then we can suggest what "codes" you can use.

AnalysIR, 2017-08-17:
yes, 32 bits is standard. As before varying th edits a bit may be useful or less error prone.(vs 0x00000007).

Good to hear you made some progress. 

It is usually a good idea to work thru all of the examples provided with the library as means of learning how to use it. So playing with a TV and its remote is pretty straightforeward as a means of getting started with IRremote.

MadTooler, 2017-08-17:
@z3t0 ,
Thanks.  Please do not take my comments about documentation as a slam on any of the great work all of you have given the community.  I was only trying to bring up my difficulty in finding what I needed for both help in finding it, and so others may as well.  Sometimes it is right in front of me and I just need a good poke in the eye to see it clearly.

I had already looked over the NEC file you referred.  My codes can be whatever I want since I am only trying to speak directly from arduino to arduino, no existing remote controls or appliances will be used.  I am trying to determine if I can simply use any code I can create, such as "12345" or "abc" or if it is a specific Hex or other value range.  I realize this is now off topic.

AnalysIR, 2017-08-17:
You can use **any** 32 bit value as long as both Arduinos are 'on the same page' and use matching values.

If you want to be 100% compliant( and there  is no need to be) the 2nd byte should be the inverse of the 1st byte (A5/5A) & the 4th byte should be the inverse of the 3rd byte. This provides some basic error checking....but not really more than matching a known 32 bit value. As a minimum I would avoid multiple 0 bits or multiple 1 bits, if possible.

Obviously avoid the same values used by any other devices in the same room ( e.g. TV etc)

MadTooler, 2017-08-17:
Excellent!!  Thanks again for all the guidance.

closed 2017-08-17 by MadTooler