Arduino-IRremote/Arduino-IRremote #506: Removing ASCII "art"?

z3t0, 2017-08-17:
I am not really a fan of art in plain text, especially not in source code. I would like to remove the large headings in files (see screenshot) if no one is opposed.

[image]

Suggested alternative....

[image]

jassing, 2017-08-17:
Change it. I am old, but for standout notations, I like
/***********
  ** Sony **
***********/
But the asci art should go.

z3t0, 2017-08-18:
That's good enough for me, glad to know I'm not the only one bothered by it :smile:

bengtmartensson, 2017-08-18:
Ascii art banners is definitely not a technology for 2017. 

Instead, I suggest file comments of Doxygen, see #461.

Aside: in some situations, "ascii art" (not banners) can be useful, see [these nice examples](http://busyducks.com/wp_4_1/2015/11/16/ascii-art-arduino-pinouts/).

ArminJo, 2020-07-04:
It looks a bit retro, so why not keep it.

closed 2020-07-04 by ArminJo