Arduino-IRremote/Arduino-IRremote #508: Need help cleaning up IR codes

shyam1998, 2017-08-20:
I recieved these codes from @AnalysIR 's sketch. I recorded both the on and off signals. I need help in cleaning up these codes.
[signals (1).txt](https://github.com/z3t0/Arduino-IRremote/files/1236990/signals.1.txt)

AnalysIR, 2017-08-20:
See below & post back if they work....

```
/*
Automatically Generated by AnalysIR - Batch Export Utility
Registered to: xxxxxxxx
Session History
Type : Key : Value : Bits : Carrier Frequency (kHz)
0 : PANASONIC128_216_280AC : ON : 400407200000006040040720001C5401AA00007007000081000035 : 216 : 38
1 : PANASONIC128_216_280AC : OFF : 400407200000006040040720009C5401AA000070070000810000B5 : 216 : 38
Note: Be sure to use the correct Carrier frequency, for each individual signal, as(or if) indicated above
*/

// NB: Not all protocols are supported by IRremote or IRLib. You may need to edit the code below manually
// Automatically Generated by AnalysIR for xxxxxxxx, visit https://www.AnalysIR.com/ or email info@....... for further details
int khz=38; //NB Change this default value as neccessary to the correct carrier frequency

unsigned int Signal_ON_0[] = {3500,1750,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,1300,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,435,435,435,435,435,435,435,435,435,435,10336,3500,1750,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,1300,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,1300,435,435,435,435,435,435,435,1300,435,435,435,1300,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,435,435,1300,435,435,435,1300,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,435,435,1300,435,435,435,1300,435}; //AnalysIR Batch Export (IRremote) - RAW

irsend.sendRaw(Signal_ON_0, sizeof(Signal_ON_0)/sizeof(int), khz); //AnalysIR Batch Export (IRremote) - RAW
 // AnalysIR IR Protocol: PANASONIC128_216_280AC, Key:  ON

unsigned int Signal_OFF_1[] = {3500,1750,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,1300,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,435,435,435,435,435,435,435,435,435,435,10320,3500,1750,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,1300,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,435,435,435,435,1300,435,1300,435,1300,435,435,435,435,435,435,435,1300,435,435,435,1300,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,435,435,1300,435,435,435,1300,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,435,435,1300,435,1300,435,435,435,1300,435,435,435,1300,435}; //AnalysIR Batch Export (IRremote) - RAW

irsend.sendRaw(Signal_OFF_1, sizeof(Signal_OFF_1)/sizeof(int), khz); //AnalysIR Batch Export (IRremote) - RAW
 // AnalysIR IR Protocol: PANASONIC128_216_280AC, Key:  OFF
```

AnalysIR, 2017-08-20:
double post from #480 ..... linked for reference

shyam1998, 2017-08-20:
@AnalysIR Insufficient Memory :(

shyam1998, 2017-08-20:
Having Memory issues. Any Tips for reducing memory usage?This is my code
`#include "DHT.h"
#include<IRremote.h>
#define TRIG_PIN 8
#define ECHO_PIN 6
IRsend irsend;
DHT dht;
int khz=38;
int distance;
long duration = 0;
bool TurnedOn = false;
void setup() {
Serial.begin(9600);
dht.setup(2);
pinMode(TRIG_PIN , OUTPUT);
pinMode(ECHO_PIN , INPUT);
}

void loop() {
  
  delay(dht.getMinimumSamplingPeriod());
  float humidity = dht.getHumidity();
  float temperature = dht.getTemperature();
  Serial.println(temperature);
  int obj_distance = calculateDistance();
  if (temperature > 28.0) {
    if (TurnedOn == false) {
      turnOn();
      Serial.println("TRIGGERED DUE TO TEMPERATURE");
     }
  }
  if (obj_distance < 21) {
    if (TurnedOn == false) {
       turnOn();
       Serial.println("TRIGGERED DUE TO ULTRASONIC");
    }
  }
  delay(1000);
  if (TurnedOn) {
     for(int i = 1 ; i <= 3600 ; i++) {
        obj_distance = calculateDistance();
        if(temperature < 22 || obj_distance < 21) {
          if (TurnedOn) {
            turnOff();
            Serial.println("TURNED OFF DUE TO ULTRASONIC1");
    } 
  } 
        delay(1000);
    }
     if (TurnedOn) {
     turnOff();
     Serial.println("TURNED OFF AFTER 3 SEC");
   }
  }
  obj_distance = calculateDistance();
  if(temperature < 22 || obj_distance < 21) {
    if (TurnedOn) {
      turnOff();
      Serial.println("TURNED OFF DUE TO ULTRASONIC2");
    } 
  }  
}
void turnOn() {
  unsigned int OnSignal[] = {3500,1750,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,1300,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,435,435,435,435,435,435,435,435,435,435,10336,3500,1750,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,1300,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,1300,435,435,435,435,435,435,435,1300,435,435,435,1300,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,435,435,1300,435,435,435,1300,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,435,435,1300,435,435,435,1300,435};
  irsend.sendRaw(OnSignal, sizeof(OnSignal)/sizeof(int), khz);
  TurnedOn = true;
}
void turnOff() {
  unsigned int OffSignal[] = {3500,1750,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,1300,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,435,435,435,435,435,435,435,435,435,435,10320,3500,1750,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,1300,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,435,435,435,435,1300,435,1300,435,1300,435,435,435,435,435,435,435,1300,435,435,435,1300,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,435,435,1300,435,435,435,1300,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,1300,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,435,1300,435,435,435,1300,435,1300,435,435,435,1300,435,435,435,1300,435};
  irsend.sendRaw(OffSignal , sizeof(OffSignal)/sizeof(int) , khz);
  TurnedOn = false;
}
int calculateDistance() {
    digitalWrite(8 , LOW);
    delayMicroseconds(2);
    digitalWrite(8 , HIGH);
    delayMicroseconds(5);
    digitalWrite(8 , LOW);
    duration = pulseIn(6 , HIGH);
    distance = duration/58.2;
    return distance;
}`

AnalysIR, 2017-08-20:
**First, get one signal working** with a basic IRemte sendRAW sketch, before moving to the next step.

Once you have the signals working (individually), you may run out of SRAM on standard Arduinos using multiple Long AC signals.

Fortunately, we have several examples on our blog showing how to overcome this:
[List of blog Posts](https://www.analysir.com/blog/sitemap_analysir_posts/)

Look for examples of sending IR signals from Flash and other ones showing how to send long AC signals from their HEX values. Both approaches are enough to save your SRAM.

shyam1998, 2017-08-20:
@AnalysIR  Everything Works perfectly. Thank you so much!

AnalysIR, 2017-08-20:
Great!.....don't forget to close your issue & post how you resolved it in the end (to benefit others).

closed 2020-08-17 by ArminJo