Arduino-IRremote/Arduino-IRremote #509: Denon IR

mataide, 2017-08-20:
**Board:** ARDUINO MEGA
**Library Version:** 2.4.0
**Protocol:** Denon

**Code Block:**

```
#include <IRremote.h>

IRsend irsend;

void setup()
{
  Serial.begin(9600);
}

void loop() {
	for (int i = 0; i < 3; i++) {
		irsend.sendDenon(0x824, 14);
		delay(40);
	}
 Serial.println("Sent DENON");
	delay(5000); //5 second delay between each signal burst
}
```

I have a Denon DVD that IR Code doesnt works.

This is what I get from decode Open/Close button:

```
Encoding  : Denon
Code      : 824 (14 bits)
Timing[31]: 
     + 350, - 700     + 350, - 650     + 350, - 700     + 350, -1750
     + 350, - 700     + 350, - 650     + 350, - 700     + 350, - 700
     + 350, - 700     + 350, -1700     + 350, - 700     + 350, - 750
     + 300, -1750     + 300, - 750     + 350, - 650     + 400
unsigned int  rawData[31] = {350,700, 350,650, 350,700, 350,1750, 350,700, 350,650, 350,700, 350,700, 350,700, 350,1700, 350,700, 350,750, 300,1750, 300,750, 350,650, 400};  // Denon 824
unsigned int  data = 0x824;
``` 
None button works.
The system is working with others IR Codes

AnalysIR, 2017-08-20:
You are likely missing the 2nd part of the signal.....(assuming you have tried sendRAW already)

...try the following sketch, which should capture it.
[sketch via blog](https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/)

mataide, 2017-08-20:
@AnalysIR Thanks for the Sketch, I tried to use it but nothing shows in Monitor.

I am using Arduino MEGA and my IR Receiver is at PIN 11.

mataide, 2017-08-20:
SendRaw also doenst work

AnalysIR, 2017-08-21:
yes the sketch is for UNOs, so you will have to make some adjustments for the pins on the Mega

try changing this line

`    attachInterrupt(0, rxIR_Interrupt_Handler, CHANGE);//re-enable ISR for receiving IR signal
`
to

`    attachInterrupt(4, rxIR_Interrupt_Handler, CHANGE);//re-enable ISR for receiving IR signal
`

pin **2**  is **INT4** ON THE arduino MEGA2560 AND **INT0** ON THE UNO

sendRAW does work...but you need to get the full signal first.

mataide, 2017-08-21:
@AnalysIR 
IR Receiver Signal is at PIN 2.
Code changed to: `attachInterrupt(4, rxIR_Interrupt_Handler, CHANGE);//re-enable ISR for receiving IR signal`

But still nothing is showed :(

AnalysIR, 2017-08-21:
did you change the 2 attachInterrupt statements ?  and also change the detachInterrupt(0) to detachInterrupt(4)

If it still does not work, post a photo of your setup.

mataide, 2017-08-21:
Yes, changed all 2 interrupts and nothing.

Here is my setup: 
I use PIN 11 to work with IRrecvDumpV2 and works good.
I changed PIN 11 to PIN 2 to work with your sketch but doesnt work for me.

[image]

AnalysIR, 2017-08-21:
Ok now it seems that you are using an **Arduino UNO** and not an **Arduino Mega** as per your earlier post!

:(

So, now you should read up the Arduino reference documentation for attachInterrupt & detachInterrupt and configure those statements(3) in the sketch to work with your platform and the pin you wish to use. For this sketch on a UNO you can only use pin 2 or pin 3 ....so use pin 2. Also make sure other statements that use the pin are correct.

However, if you use Pin 2 on the UNO, the  original sketch should work, OK.

Did you confirm the BAUD rate is correct & matches the sketch?

marckaraujo, 2017-08-21:
@AnalysIR Sorry, for the image from UNO, but I use Mega, it was just an illustration.

Yes, BAUD is correct.

AnalysIR, 2017-08-21:
OK...not great, but I am unsure of where you are at now. 

On the Arduino Mega2560, use pin 2 for IR in & set all the attach/detach interrupt statements correclty for that pin. (I believe it is INT4.)

Note: Pin 11 **cannot** be used for this on the Mega2560.

If that doesn't work, photo please.....(also asked for it earlier!)

mataide, 2017-08-21:
Follow the photo.

[image]

mataide, 2017-08-21:
@AnalysIR  Got it working on Mega:

`const byte interruptPin = 2;`

  `attachInterrupt(digitalPinToInterrupt(interruptPin), rxIR_Interrupt_Handler, CHANGE);//set up ISR for receiving IR signal`

Reference: https://www.arduino.cc/en/Reference/AttachInterrupt

AnalysIR, 2017-08-21:
Great!

That IR Receiver isn't great, but if it works for you....

Hopefully you will get  a more complete signals now....lets see.

mataide, 2017-08-21:
@AnalysIR You are right, was missing part of the code.

`Raw: (95) 332, -720, 328, -724, 328, -720, 332, -1772, 304, -748, 328, -724, 328, -724, 324, -724, 328, -724, 328, -1796, 304, -724, 328, -720, 332, -1772, 328, -724, 328, -720, 332, -47968, 332, -720, 328, -724, 328, -724, 328, -1772, 328, -724, 328, -1776, 324, -1776, 328, -1776, 324, -1800, 280, -772, 304, -1796, 280, -1800, 324, -724, 328, -1776, 328, -1796, 304, -41568, 332, -720, 328, -724, 328, -724, 328, -1772, 328, -724, 328, -724, 328, -720, 332, -720, 328, -724, 328, -1800, 276, -772, 304, -748, 304, -1772, 328, -724, 328, -724, 328`

Couldnt it be an improve for this library? Have you discussed about that?

PS: You should update AnalysIR Firmware for Mega too, that is the same issue with Interrupt.

AnalysIR, 2017-08-21:
Yes, it is a 3 part Denon protocol/signal. 1st & 3rd parts are identical.

> PS: You should update AnalysIR Firmware for Mega too, that is the same issue with Interrupt.

There is no issue with the sketch on our blog post. The original sketch is intended for 'standard' Arduinos with ATmega328s only. I would expect people progressing to the Mega2560 to be able to handle  the trivial code changes without prompting. However, I have already added a note to the blog post to cover the Mega2560, for future reference, along with the similar note that was already there for Yun/Leonardo.

Regarding the IRremote library, feel free to open an new issue documenting the problem, ideally in a new issue, linked back to this one....or if you are interested contribute the code fix. I think **one** of the issues you encountered may be fixed by adjusting a customisable parameter (not 100% sure).

mataide, 2017-08-22:
Ok, take a look later. thx

closed 2017-08-22 by mataide