Arduino-IRremote/Arduino-IRremote #510: beta branch: add (sensible) paragraph to library.properties

bengtmartensson, 2017-08-21:
Continuation of #464, #507.

The [Arduino library specification](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification) is unclear upon if `paragraph` is required in `library.properties`. It was recently removed in the beta branch. With Arduino IDE 1.8.3 and Linux (Fedora 26), this leads to the warning messages (for example when opening the library manager)
```
Invalid library found in /home/bengt/Arduino/libraries/IRremote: Missing 'paragraph' from library
Invalid library found in /home/bengt/Arduino/libraries/IRremote: Missing 'paragraph' from library
Invalid library found in /home/bengt/Arduino/libraries/IRremote: Missing 'paragraph' from library
Invalid library found in /home/bengt/Arduino/libraries/IRremote: Missing 'paragraph' from library
Invalid library found in /home/bengt/Arduino/libraries/IRremote: Missing 'paragraph' from library
Invalid library found in /home/bengt/Arduino/libraries/IRremote: Missing 'paragraph' from library
Invalid library found in /home/bengt/Arduino/libraries/IRremote: Missing 'paragraph' from library
Invalid library found in /home/bengt/Arduino/libraries/IRremote: Missing 'paragraph' from library
```

So I suggest that is filled with a sensible content. Note that is should only make sense when preceded by the sentence of `sentence`.

z3t0, 2017-08-22:
I also found this message as an error (sometimes) when uploading/compiling.

hballinas, 2017-08-22:
How do I unsubscribe to this email chain?

z3t0, 2017-08-22:
There is an unsubscribe button on the right side of this issue (on GitHub). Or you can unwatch at the top of the repository.

commit 5bbd55c6 (2017-08-22): Fix #510
closed 2020-06-17 by ArminJo