Arduino-IRremote/Arduino-IRremote #513: Add Support Vivo/GVT IR

mataide, 2017-08-22:
I am trying to implement support for Vivo IR.

What I got from IRrecvDumpV2, from pressing the same button "1".

```
Encoding  : UNKNOWN
Code      : 95ABCFC5 (32 bits)
Timing[35]: 
     + 500, - 250     + 200, - 200     + 200, - 550     + 250, - 400
     + 200, - 200     + 250, - 500     + 250, - 200     + 250, - 200
     + 250, - 200     + 200, - 550     + 250, - 550     + 200, - 400
     + 200, - 550     + 200, - 250     + 200, - 250     + 200, - 250
     + 200, - 350     + 250
unsigned int  rawData[35] = {500,250, 200,200, 200,550, 250,400, 200,200, 250,500, 250,200, 250,200, 250,200, 200,550, 250,550, 200,400, 200,550, 200,250, 200,250, 200,250, 200,350, 250};  // UNKNOWN 95ABCFC5

Encoding  : UNKNOWN
Code      : D7135515 (32 bits)
Timing[35]: 
     + 500, - 200     + 250, - 200     + 200, - 600     + 200, - 400
     + 200, - 200     + 250, - 500     + 250, - 250     + 200, - 200
     + 200, - 250     + 200, - 250     + 200, - 550     + 200, - 400
     + 200, - 550     + 250, - 200     + 250, - 200     + 200, - 250
     + 200, - 400     + 250
unsigned int  rawData[35] = {500,200, 250,200, 200,600, 200,400, 200,200, 250,500, 250,250, 200,200, 200,250, 200,250, 200,550, 200,400, 200,550, 250,200, 250,200, 200,250, 200,400, 250};  // UNKNOWN D7135515
```

What I get from AnalysIR:

Code for the button "1", same code if you press twice.

`Raw: (107) 428, 268, 172, 268, 176, 624, 152, 436, 176, 264, 180, 596, 176, 268, 176, 288, 152, 268, 176, 624, 152, 600, 176, 436, 176, 600, 176, 268, 172, 268, 176, 268, 176, 432, 176, 24480, 428, 264, 180, 264, 180, 616, 156, 432, 176, 268, 176, 600, 176, 288, 156, 264, 180, 264, 176, 600, 176, 600, 176, 432, 180, 596, 180, 264, 176, 264, 180, 264, 180, 428, 180, 24472, 428, 288, 156, 288, 156, 620, 156, 432, 176, 288, 156, 596, 180, 284, 156, 288, 156, 288, 156, 596, 180, 596, 180, 452, 156, 600, 176, 264, 180, 284, 156, 288, 156, 452, 156`

It doesnt work if you sendRaw with those codes.
The code seems to be RC6 but with a properly header.

[image]

A forum thread used this code to work with ESP8266:

```
void ICACHE_FLASH_ATTR IRsend::sendRC6gvt(unsigned long data, int nbits) {
  // Set IR carrier frequency
  enableIROut(36);
  // Header
  mark(RC6_HDR_MARK); //header
  space(RC6_HDR_SPACE);
  mark(RC6_T1);  
  space(RC6_T1);
  mark(RC6_T1);  
  space(RC6_T1);
  mark(RC6_T1);  
  space(RC6_T1);  
  space(RC6_T1);      
  mark(RC6_T1);  
  space(2*RC6_T1); //trailer bits     
  mark(2*RC6_T1);   
 
  int t;
  // Data
  for (unsigned long i = 0, mask = 1UL << (nbits - 1); mask; i++, mask >>= 1) {
    // The fourth bit we send is a "double width trailer bit".
    t = RC6_T1;
   
    if (data & mask) {  // 1
      mark(t);
      space(t);
    } else {  // 0
      space(t);
      mark(t);
    }
  }
  // Footer
  ledOff();
}
```

AnalysIR, 2017-08-22:
FYI: It decodes as RCMM for me on AnalysIR. This can be a difficult protocol to work with as the timings can be critical.

Try this with a 'cleaned' version of your signal:

```
/*
Automatically Generated by AnalysIR - Batch Export Utility
Registered to: xxxxxxxx
Session History: RCMM
Type : Key : Value : Bits : Carrier Frequency (kHz)
0 : RCMM :  : 2480A601 : 32 : 0
Note: Be sure to use the correct Carrier frequency, for each individual signal, as(or if) indicated above
*/

// NB: Not all protocols are supported by IRremote or IRLib. You may need to edit the code below manually
// Automatically Generated by AnalysIR for xxxxxxxx, visit https://www.AnalysIR.com/ or email info@....... for further details
int khz=38; //NB Change this default value as neccessary to the correct carrier frequency

unsigned int Signal_0_0[] = {417,278,167,278,167,611,167,444,167,278,167,611,167,278,167,278,167,278,167,611,167,611,167,444,167,611,167,278,167,278,167,278,167,444,167,24480,417,278,167,278,167,611,167,444,167,278,167,611,167,278,167,278,167,278,167,611,167,611,167,444,167,611,167,278,167,278,167,278,167,444,167,24472,417,278,167,278,167,611,167,444,167,278,167,611,167,278,167,278,167,278,167,611,167,611,167,444,167,611,167,278,167,278,167,278,167,444,167}; //AnalysIR Batch Export (IRremote) - RAW

irsend.sendRaw(Signal_0_0, sizeof(Signal_0_0)/sizeof(int), khz); //AnalysIR Batch Export (IRremote) - RAW
 // AnalysIR IR Protocol: RCMM, Key:  
```
PS: There may be a mistake in your image above (_or at least the way you are interpreting it, you seem to confusing the marks with the spaces_), as the receiver output is inverted logic, See trace image from AnalysIR below. (Click to expand)
[image]

mataide, 2017-08-22:
@AnalysIR 

Your code works for the first time. But nothing happens for the next actions.
I think that the second time you press the same button, it send a different code.

It because If I press the button from Remote Controller, then the arduino start to work again and stop right after.

PS: I am working at your firmware with Mega2560, I did some changes, accept me on Forum so we can discuss about.

AnalysIR, 2017-08-23:
Yes the 32-bit version of this protocol, may use a toggle bit...if I remember correctly.

I suggest you try out the IRLIB2 library, which seems to/may have better support for this particular protocol.

Alternatively use our sketch to capture multiple/subsequent versions of the same signal & you will find where the toggle bit is by comparison. You can then use these to achieve a working solution.

> accept me on Forum so we can discuss about.

I have just approved all outstanding registration requests on our [IRForum](https://irforum.analysir.com/), so you should be clear to post now.

bengtmartensson, 2017-08-23:
@mataide : I would suggest that you use IrScrutinizer to capture all your signals of your remote. You can then automatically from the program generate code for Arduino using IRremote (alternatively Infrared4Arduino or IRLib) using raw codes, residing in flash memory. If you capture in parametric mode, the signals will be automatically "cleaned". You will then see if the toggle (parameter T) varies between your captures.

IrScrutinizer (DecodeIR really) identifies the signal as a Nokia32 protocol. (This protocol is a special case of RCMM, so my statement does not contradict AnalysIR's.) Nokia32 has a toggle bit, but still it is possible that devices using the protocol ignores it.

mataide, 2017-10-13:
Would someone develop this for me if I make a donate? Its been hard to develop this and I am full with other jobs.

mrscarvalho, 2020-09-21:
Hello Mataide, did you have some progress controlling Vivo devices?

closed 2020-10-17 by ArminJo