Arduino-IRremote/Arduino-IRremote #514: Close Issues

mataide, 2017-08-22:
There is a lot of open issues that can be closed, so we can focus on what really matters.

Here is a list:

#478 - When to support Arduino101
#470 - I'm having trouble sending codes
#469 - Pin used for sending on a Arduino MEGA
#452  - Better testing
#431 - Fix RobotIRremote conflict
#426 - Cleaning up Issues
#395 - Restructure Files / Rename things too
#359 - Difference of pressing IRremote button for a long time or a shor time
#358 - Consult to how to decode sharp TV IRremote signals and send
#355 - Intel based Arduino support
#353 - Trinket 5V Support (internal oscilator)
#339 - Maintainer away
#294 - First interval
#290 - sending IR commands from a long distance
#280 - Freezing device receiver while sending NEC code
#275 - irrecv.resume(); outside if statement of decode makes lot of difference ?
#261 - Save program memory space by storing strings in once
#221 - Cross project portability
#147 - Missing DECODE functions
#146 - Missing SEND functions
#145 - #ifdef AIWA_RC_T501
#143 - irrecv.resume not working after irsend.sendNEC

There are some topics that needs to be marked with some labels

z3t0, 2017-08-22:
Thanks I will go ahead and close them now.

z3t0, 2017-08-22:
Done

closed 2017-08-22 by z3t0