Arduino-IRremote/Arduino-IRremote #518: WiFi broken on ESP32 if IRremote used

narbuzov, 2017-09-05:
**Board:** ESP32 
**Library Version:** 2.3.3
**Protocol:** Sony (if any)

**Code Block:**
```c
#include <WiFi.h>
#include <IRremote.h>

IRrecv irrecv(16);

const char * ssid = "network";
const char * password = "password";

void setup(){
   Serial.begin(115200);
   Serial.println("----------Program Start ------------");
   
   delay(10);
   irrecv.enableIRIn();

   Serial.print("Connecting to ");
   Serial.println(ssid);

   WiFi.begin(ssid, password);

   while (WiFi.status() != WL_CONNECTED) {
       delay(500);
       Serial.print(".");
   }
   Serial.println("Success");
 }
 
 void loop() {}

```

Produces this output: 

> Executing task: platformio device monitor <

--- Miniterm on /dev/cu.SLAB_USBtoUART  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0008,len:8
load:0x3fff0010,len:160
load:0x40078000,len:10632
load:0x40080000,len:252
entry 0x40080034
----------Program Start ------------
Connecting to network
Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Register dump:
PC      : 0x400dc5dc  PS      : 0x00060034  A0      : 0x40081c69  A1      : 0x3ffc0be0
A2      : 0x00000001  A3      : 0x00000002  A4      : 0x000000ff  A5      : 0x40085f48
A6      : 0x00000001  A7      : 0x3ffcccb0  A8      : 0x800832ac  A9      : 0xe00a1400
A10     : 0x00000000  A11     : 0x3ffc1480  A12     : 0x20000000  A13     : 0x00000001
A14     : 0x00000400  A15     : 0x00060523  SAR     : 0x00000017  EXCCAUSE: 0x00000007
EXCVADDR: 0x00000000  LBEG    : 0x40001609  LEND    : 0x4000160d  LCOUNT  : 0x00000000

Backtrace: 0x400dc5dc:0x3ffc0be0 0x40081c69:0x3ffc0c00

CPU halted.

z3t0, 2017-09-19:
Perhaps the timers used for WIFI are used by this library? I have no experience with ESP32 so perhaps others might have better advice.

MarkyAD, 2017-09-19:
Try enabling the IR receiver after wifi has been established. Works fine here.

mitchsf, 2017-09-21:
Same issue here. Starting IR after WiFi does work, but restarting WiFi when a connection is lost will crash. I don't see a way to stop and restart IR in the library. I tried switching to timers 0 and 2, same error.

mitchsf, 2017-09-23:
Now Preferences crashes at x.begin("data", false), after decoding IR. Any suggestions for at least a temporary workaround?

Guru Meditation Error: Core  1 panic'ed (Cache disabled but cached memory region accessed)
Register dump:
PC      : 0x40080dfb  PS      : 0x00060034  A0      : 0x800815ec  A1      : 0x3ffc0bc0  
A2      : 0xbad00bad  A3      : 0x3ffc1170  A4      : 0x20000000  A5      : 0x00000001  
A6      : 0x00000400  A7      : 0x00060423  A8      : 0x3ffc79e0  A9      : 0x3f400038  
A10     : 0x00000001  A11     : 0x3ffd9520  A12     : 0x800845b9  A13     : 0x3ffda1b0  
A14     : 0x00000003  A15     : 0x00060423  SAR     : 0x00000002  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x40080dfb:0x3ffc0bc0 0x400815ec:0x3ffc0be0 0x40082369:0x3ffc0c00

CPU halted.

Decoding 6 results
0x40080dfb: IRTimer() at C:\Users\Mitchell\SkyDrive\Arduino\libraries\Arduino-IRremote/IRremote.cpp line 142
0x40080dfb: IRTimer() at C:\Users\Mitchell\SkyDrive\Arduino\libraries\Arduino-IRremote/IRremote.cpp line 142
0x400815ec: __timerISR at C:\Program Files (x86)\Arduino\hardware\espressif\esp32\cores\esp32/esp32-hal-timer.c line 179
0x40082369: _xt_lowint1 at xtensa_vectors.o line ?

jt040-jetnet, 2017-12-07:
same issue is with me 
i am using esp32 in arduino core,  is this issue fixed ?
kindly suggest how to solve the issue

mitchsf, 2017-12-07:
Not that I know of. I just use an ATTiny85 to process IR, and a simple serial connection to the ESP32.

ArminJo, 2020-08-17:
Fixed in the current version

closed 2020-08-17 by ArminJo