Arduino-IRremote/Arduino-IRremote #521: sendSanyo problem

jodhaakbar, 2017-09-14:
**Board:** ARDUINO UNO
**Library Version:** 2.2.3
**Protocol:** Sanyo

i cannot using irsend.SendSanyo. any idea guys ?

z3t0, 2017-09-19:
Need more information

AnalysIR, 2017-09-20:
Yes, it looks like there is no 'send' implemented in the code for Sanyo...yet!

...which would of course explain your issues.

@z3t0 Essentially this is less of an 'issue' and more of a feature request.

Usually, additional features like this are added when contributed.

anduscheung, 2017-09-20:
There is a pull request here with Sanyo Send
[https://github.com/z3t0/Arduino-IRremote/pull/370]
I think I am gonna give it a try.

bengtmartensson, 2017-09-21:
In my "carrier",  I have never seen a protocol named "Sanyo" outside of this community, To my knowledge, most Sanyo products use some variant of the NEC protocol. Looking at the code in ir_Sanyo reveals that it is at least very close to NEC. 

To clarify the situation, please try to, for example, provide a high-quality capture of "Sanyo" signals, and/or mention the exact device (or remote) that is to be controlled.

The alternate suggestion is to remove ir_Sanyo.cpp.

anduscheung, 2017-09-21:
I am using a fan remote of model Panasonic ELECTRIC FAN F-307KH, the signal is detected as Sanyo and if I remove ir_Sanyo I am receiving an unknown signal instead of NEC

AnalysIR, 2017-09-21:
FYI:.._..in case you didnt already know?_
You can always record the raw signal(s) using IRremote & then use sendRAW to send the 'RAW'  signal.
_(see the examples)_

That way you dont need to know anything about the Sanyo protocol.

anduscheung, 2017-09-21:
Well I am very new to Arduino infrared and I couldn't find any example of sending Raw signals, could you kindly provide a link if there's any? Btw almost every online tutorial are pointing to this page [Arduino-IRremote](http://z3t0.github.io/Arduino-IRremote/) and I believe this means this library is great!

AnalysIR, 2017-09-21:
You need to look in the examples for IRremote within the Arduino IDE...
[or in the examples folder here](https://github.com/z3t0/Arduino-IRremote/tree/master/examples)

The 2 of interest are....
> 	IRrecvDumpV2 	
> 	IRsendRawDemo

anduscheung, 2017-09-21:
Thanks, now I checked again and I am actually receiving this,

Rx:
Unknown encoding: 43430E7F (32 bits)
 52 Pulses: s3010600    m3600    s3300    m1000    s2500     m950    s2500     m950     s750     m900    s2550     m900    s2550     m900    s2550     m900     s850     m900     s800     m900     s850     m850     s850     m900     s850     m900    s2550     m950     s750     m900     s850     m850    s2600     m900     s800     m900     s850    m1000     s700     m900    s2550    m1000    s2450     m900    s2600     m950    s2500     m850    s2600     m850     s850     m850

How could I turn this to raw?

anduscheung, 2017-09-21:
FYI this is received after I made some changes following [SanyoSend](https://github.com/z3t0/Arduino-IRremote/pull/370), originally the signal is said to be Sanyo xxxxxx(I forgot the code) so probably it is a detectable Sanyo signal detected by ir_Sanyo

bengtmartensson, 2017-09-21:
I hand-edited the code above and put it through IrScrutinizer (which is using DecodeIR). The signal was identified as [Emerson](http://www.hifi-remote.com/wiki/index.php?title=DecodeIR#Emerson), D=59, F=32. So, now we know the truth behind "Sanyo".

anduscheung, 2017-09-21:
@bengtmartensson It's so surprising that Panasonic fan is using a Sanyo protocol which is actually Emerson lol

bengtmartensson, 2017-09-21:
Names of protocols are actually quite arbitrary. Funniest (IMHO) is that the protocol most people call "Denon" is by Denon (the firm) called "Sharp"!

AnalysIR, 2017-09-24:
> How could I turn this to raw?

@anduscheung You should be able to generate the RAW signal from the demos linked above...the output should start like:
Raw (nn): .....

On AnalysIR your signal is identified as Konka or Matsushita _(as mentioned above there are no real naming standards and are thus mostly arbitrary)_

The following may be of help to you - just include it in a standard IRremote sketch:

```
/*
Automatically Generated by AnalysIR - Batch Export Utility
Registered to: xxxxxxxx
Session History
Type : Key : Value : Bits : Carrier Frequency (kHz)
0 : KONKA :  : DC123E : 24 : 0
Note: Be sure to use the correct Carrier frequency, for each individual signal, as(or if) indicated above
*/

// NB: Not all protocols are supported by IRremote or IRLib. You may need to edit the code below manually
// Automatically Generated by AnalysIR for xxxxxxxx, visit https://www.AnalysIR.com/ or email info@....... for further details
int khz=38; //NB Change this default value as neccessary to the correct carrier frequency

unsigned int Signal_0_0[] = {3350,3350,850,2500,850,2500,850,850,850,2500,850,2500,850,2500,850,850,850,850,850,850,850,850,850,850,850,2500,850,850,850,850,850,2500,850,850,850,850,850,850,850,2500,850,2500,850,2500,850,2500,850,2500,850,850,850}; //AnalysIR Batch Export (IRremote) - RAW

irsend.sendRaw(Signal_0_0, sizeof(Signal_0_0)/sizeof(int), khz); //AnalysIR Batch Export (IRremote) - RAW
 // AnalysIR IR Protocol: KONKA, Key:  
```

closed 2020-10-12 by ArminJo