Arduino-IRremote/Arduino-IRremote #524: query

dunk8888, 2017-09-16:
Hi can you help me understand why the bellow code differs from the same key press.
Ime using trying understand a sketch some one wrote,the key codes bellow are from the same buttons on the same remote,but the two sketches differ like bellow,i just cant understand whats going on.
I CANT UNDERSTAND WHY THEY ARE DIFFERANT.
#define IRKEY_ONE    0x010  
#define IRKEY_TWO    0x810
#define IRKEY_THREE  0x410
#define IRKEY_FOUR   0xc10

#define IRKEY_ONE    0x80  
#define IRKEY_TWO    0x81  
#define IRKEY_THREE  0x82  
#define IRKEY_FOUR   0x83

AnalysIR, 2017-09-16:
the bit order is just reversed :)

convert to binary and you will see the pattern

z3t0, 2017-09-19:
I think this has been solved :smile:

closed 2017-09-19 by z3t0

dunk8888, 2017-09-19:
Oh right,so i need a hex to binary converter? ile have a look and see if i can figure it out lol ime new to binary hex etc will keep me busy.thanks

dunk8888, 2017-09-19:
Why has this been done?

dunk8888, 2017-09-19:
Thanks i can see now,i can add my own remote codes to an existing project,brill thanks again.

dunk8888, 2017-09-19:
Is it somthing to do the way its read from memory? One sketch uses pgmspace
and the other doesnt so ime thinking its going into memory or somthing then
getting switched round or read backwards msb lsb etc so the code has to be
switched to start with,am i on the wright track or no? lol ime just trying
to figure it out why has it been done that way?