Arduino-IRremote/Arduino-IRremote #526: send of NEC 64bits

AsTiger117, 2017-09-21:
**Board:** ARDUINO UNO
**Library Version:** 2.1.0
**Protocol:** Sony (if any)

**Code Block:**
```c

#include <IRremote.h> 

.....

```

Use [a gist](gist.github.com) if the code exceeds 30 lines

**checklist:**
- [] I have **read** the README.md file thoroughly
- [] I have searched existing issues to see if there is anything I have missed.
- [] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [] The title of the issue is helpful and relevant 

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

The above is a short template allowing you to make detailed issues!

bengtmartensson, 2017-09-21:
Short answer: no, there is no easy and natural solution.

The first argument sendNEC is an unsigned long, which is 32 bits on "most" (all?) boards.  I suspect the compiler issues a warning when you try to assign 0x6057000000D1020D to that parameter.

AsTiger117, 2017-09-21:
Yes. This is the message: warning: large integer implicitly truncated to unsigned type [-Woverflow]

kyet, 2018-08-30:
@AsTiger117 
sendNEC(..) consists of 
- sendNEC_header()
- sendNEC_data(..)
- sendNEC_footer()
These member functions are private.
If you change this to public, we could write like this,
```
irsend.sendNEC_header();
irsend.sendNEC_data(0x60570000, 32);
irsend.sendNEC_data(0x00D1020D, 32);
irsend.sendNEC_footer();
```

AnalysIR, 2018-08-30:
or just write your own custom funtion using iresend.mark & irsend.space
...as we show on our blog [](https://www.analysir.com/blog/2015/09/01/simple-infrared-pwm-on-arduino-part-3-hex-ir-signals/)

_{The example shows the use of softPWM,which can be easily swapped with the built-in IRremote mark & space functions)_

ArminJo, 2020-09-16:
You can copy the function sendNEC and use the function sendPulseDistanceWidthData(NEC_BIT_MARK, NEC_ONE_SPACE, NEC_BIT_MARK, NEC_ZERO_SPACE, data, nbits); multiple times.

closed 2020-09-16 by ArminJo