Arduino-IRremote/Arduino-IRremote #528: How to find protocol

SelaO, 2017-10-01:
**Board:** ARDUINO UNO
**Library Version:** 2.4.0
**Protocol:** don't know

I run the IRtest example and I get:

```
NEC tests
Testing 0: Error
0 (0 bits)
Raw (0): 
Testing FFFFFFFF: Error
0 (0 bits)
Raw (0): 
Testing AAAAAAAA: Error
0 (0 bits)
Raw (0): 
Testing 55555555: Error
0 (0 bits)
Raw (0): 
Testing 12345678: Error
0 (0 bits)
Raw (0): 
Sony tests
Testing FFF: Error
0 (0 bits)
Raw (0): 
Testing 0: Error
0 (0 bits)
Raw (0): 
Testing AAA: Error
0 (0 bits)
Raw (0): 
Testing 555: Error
0 (0 bits)
Raw (0): 
Testing 123: Error
0 (0 bits)
Raw (0): 
RC5 tests
Testing FFF: Error
0 (0 bits)
Raw (0): 
Testing 0: Error
0 (0 bits)
Raw (0): 
Testing AAA: Error
0 (0 bits)
Raw (0): 
Testing 555: Error
0 (0 bits)
Raw (0): 
Testing 123: Error
0 (0 bits)
Raw (0): 
RC6 tests
Testing FFFFF: Error
0 (0 bits)
Raw (0): 
Testing 0: Error
0 (0 bits)
Raw (0): 
Testing AAAAA: Error
0 (0 bits)
Raw (0): 
Testing 55555: Error
0 (0 bits)
Raw (0): 
Testing 12345: Error
0 (0 bits)
Raw (0): 
```
I have a TSOP38238 connected and this is the output even if don't do anything. I added the `#define TEST` to the h file, but there's no o file anywhere, so I just restarted the Arduino IDE.

How am I supposed to find the protocol of the remote then?

Use [a gist](gist.github.com) if the code exceeds 30 lines

**checklist:**
- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [X] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [X] The title of the issue is helpful and relevant 

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

The above is a short template allowing you to make detailed issues!

AnalysIR, 2017-10-01:
try using the  **IRrecvDumpV2** example instead. without any changes to the library.

SelaO, 2017-10-01:
Cool that one works:
```
Encoding  : NEC
Code      : 96C0070A (32 bits)
Timing[73]: 
     +8850, -4400     + 700, -1550     + 700, - 500     + 650, - 550
     + 650, -1600     + 650, - 500     + 650, -1650     + 650, -1600
     + 650, - 550     + 650, -1600     + 650, -1600     + 650, - 550
     + 600, - 550     + 650, - 550     + 650, - 500     + 650, - 550
     + 650, - 550     + 650, - 500     + 650, - 550     + 650, - 550
     + 650, - 500     + 650, - 550     + 650, -1600     + 650, -1600
     + 650, -1650     + 650, - 500     + 650, - 550     + 650, - 550
     + 650, - 500     + 650, -1600     + 650, - 550     + 650, -1600
     + 650, - 550     + 650, - 550     + 650, -1600     + 650, - 500
     + 650
unsigned int  rawData[73] = {8850,4400, 700,1550, 700,500, 650,550, 650,1600, 650,500, 650,1650, 650,1600, 650,550, 650,1600, 650,1600, 650,550, 600,550, 650,550, 650,500, 650,550, 650,550, 650,500, 650,550, 650,550, 650,500, 650,550, 650,1600, 650,1600, 650,1650, 650,500, 650,550, 650,550, 650,500, 650,1600, 650,550, 650,1600, 650,550, 650,550, 650,1600, 650,500, 650};  // NEC 96C0070A
unsigned int  data = 0x96C0070A;
```

SelaO, 2017-10-01:
How can I use this if the protocol is unknown like this?

```
Encoding  : UNKNOWN
Code      : E0984BB6 (32 bits)
Timing[23]: 
     +1300, - 400     +1250, - 400     + 450, -1200     +1300, - 350
     +1300, - 400     + 450, -1200     + 450, -1250     + 450, -1250
     +1250, - 400     + 450, -1250     + 400, -1250     + 450
unsigned int  rawData[23] = {1300,400, 1250,400, 450,1200, 1300,350, 1300,400, 450,1200, 450,1250, 450,1250, 1250,400, 450,1250, 400,1250, 450};  // UNKNOWN E0984BB6

```

closed 2017-10-01 by SelaO

AnalysIR, 2017-10-01:
Good....using sendRAW :)

FYI: Your last signal above is decoded as a LASKO 12 Bit AC signal, with AnalysIR _(but could also be any other as IR protocol names are not really standardised)_

SelaO, 2017-10-01:
How did you find that it's LASKO? 

Also, I don't see LASKO send function, so it doesn't really help right?

AnalysIR, 2017-10-01:
Using AnalysIR, which is a different application.

You are fine using IRremote & sendRAW.

bengtmartensson, 2017-10-03:
Although closed, I would like to make some "philosophical" comments on the subject in the title.

There are hundreds of IR protocols, and since they are (with some exception) not defined by official documents, their name are not universally agreed-upon either. Not only that, but new ones are "invented" regularly, in particular for other equipment than traditional audio- and video devices. To make things even more "challenging", the present project uses a different parametrization, even for protocols that _are_ universally accepted. Live with it...

The present project is just a small project (2k lines) to be run on small Arduinos. It is not realistic to expect such a small project, running on such a small platform, to be able to detect more that a "few" different protocols.

Harder decodings tasks are better performed on another platform, like a PC. There exists at least two programs for the purpose: the non-free AnalysIR (Windows only), and the free IrScrutinizer (Java platforms). There are two ways to use these: 

1. cut-n-paste from the Arduino console, using a sketch like  IRrecvDumpV2, or,
2. Flash appropriate firmware on the Arduino and let the PC program take command.

There is also [DecodeIR](https://github.com/bengtmartensson/Arduino-DecodeIR), which is available in the Arduino library manager, and cooperates with IRremote. For example, its example sketch `IRremoteDecode.ino` allows using it with the serial Arduino console.  However, it is quite large (5k lines) and will not run on an Arduino considerably smaller than a Mega2560 (which  can be had as China-clone for around 10€, btw).  It is an old decoder with some problems (in particular, it is not extensible). (It is presently used in IrScrutinizer.)

ArminJo, 2020-06-09:
For all the one who read this, try [this](https://github.com/ukw100/IRMP#allprotocol-example) example first, it covers 40 protocols.