Arduino-IRremote/Arduino-IRremote #529: Debugging with one Arduino?

SelaO, 2017-10-03:
**Board:** ARDUINO UNO
**Library Version:** 2.4.0
**Protocol:** NEC

I'm sending the hex codes I decoded from my remote with `irsend.sendNEC(0x86C00B0A, 128);` (is 128 correct? since 8*16=128, the code has 8 hex characters), but nothing happens.

I see with my phone's camera that the led does turn on (although its a little weak compared to my remote even without any resistors), I even placed 3 leds, two 950nm and one 850nm to boost the signal but still nothing happens. 

Can I check if what I send is correct when connecting both the receiver and leds to the same board? I tried the following but I don't think it works since by the time we get to `irrecv.decode` the transmission already ended.

**Code Block:**
```
void loop() {
  for (int i = 0; i < 3; i++) {
    irsend.sendNEC(0x86C00B0A, 128);
    if (irrecv.decode(&results)) {
      Serial.println(results.value, HEX);
      irrecv.resume(); // Receive the next value
    }
    delay(40);
  }
  delay(1000);
}
```

**checklist:**
- [x] I have **read** the README.md file thoroughly
- [x] I have searched existing issues to see if there is anything I have missed.
- [x] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [x] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [x] The title of the issue is helpful and relevant 

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

The above is a short template allowing you to make detailed issues!

AnalysIR, 2017-10-03:
From memory the library turns off the receiver when sending. The receiver must be re-enabled after sending.

I suspect it has to do with using the same timer peripheral for both sending & receiving. During sending with IR remote the only way to receive simultaneously, would be to use interrupts for Rx in your custom code. The library tries to minimise the use of resources/peripherals where possible.

I usually use 2 or more devices when testing, so that I know at least one device is working properly. If you use just one device it can be difficult to isolate issues that arise or to know whether an issue is with the Tx or Rx function. Also note that there can be delays of up to 200 uSecs as a signal passes thru a receiver.

Using your TV or similar to test is a simple way of getting a second device, when trying things out.

As a 'general guideline', I always suggest the emitter should be 1m+ away from the receiver, unless you use some way of attenuating the signal being received/transmitted.

AnalysIR, 2017-10-03:
PS: for IR remote control drop the 850nm led & only use the 940-950nm one(s).

If you check the data sheet of your receiver, you should see that most, if not all , of the signal is rejected at 850nm.

850nm emitters are targeted at other applications (.i.e not IRremote control)

yesyesuk, 2017-10-03:
The 128 seems wrong. Each hexadecimal digit uses 4 bits. There are 8 digits, so it should be 8 * 4bit = 32bit.

SelaO, 2017-10-03:
@yesyesuk yes you're right, I also saw that in the raw output. 

@AnalysIR you were right again, I tried to use `sendRaw` with another device that I can bring the arduino close to and it worked when it was about 10cm away, with two leds and no resistors. 

How can I boost the signal more to make it work from afar like the remotes? They're 5mm leds.

I have 1000uF capacitors lying around, can I use them to boost the signal?

AnalysIR, 2017-10-03:
Unfortunately capacitors arent much use for this.....you will need a IR emitter transistor driver circuit.....via web search.

or check out the folowing blog posts...

https://www.analysir.com/blog/2013/11/22/constant-current-infrared-led-circuit/
https://www.analysir.com/blog/2014/10/03/driving-infrared-led-directly-arduino-pin/

SelaO, 2017-10-03:
I see in https://www.analysir.com/blog/2014/10/03/driving-infrared-led-directly-arduino-pin/ that they were able to get a range of 4m with connecting the led directly, so I'll need to get a better led like the Vishay TSAL6100 they used there?

bengtmartensson, 2017-10-03:
Using a setup like [this](http://www.harctoolbox.org/arduino_nano.html) (two IR diodes SFH 4546 + SFH 4544 + 68 ohm, _directly driven_ from an AT328mega) I get a performance (subjectively) as a good commercial "clicker". Just tried with 7 meters, no problems, (however, with a narrow beam).  For a driver, my favorite is [2N7000 (equivalent BS170)](https://en.wikipedia.org/wiki/2N7000). It saves you the base (gate) resistor, bringing the BOM down to 1 resistor, 1-2 LEDs, 1 transistor. 

> https://www.analysir.com/blog/2013/11/22/constant-current-infrared-led-circuit/

The universal advantage of the constant-current circuit is slowly turning into one of Internets most widely spread myths... (Not only AnalysIR, see also [IrToy](http://dangerousprototypes.com/docs/USB_Infrared_Toy).) Its advantage is that it produces constant current through the LED(s), even if the supply voltage is varying. True, but _how often is this a needed feature_? Both Arduinos and the IrToy drives the LED(s) with stabilized voltage... And is "constant current" really what we want, not just "sufficiently much"?  The drawback is the high component count, and, possibly, that (most of) the power dissipation takes place in the transistor, not in a resistor.

> with connecting the led directly

You are not considering taking out the resistor, right? _Don't_.

AnalysIR, 2017-10-03:
@SelaO I suspect for you, it will be more to do with the current going thru the IR LED. Your existing 950nm emitter should be fine.

So I would first suggest using a transistor driver of some sort, to get reliable performance over range.

The TSAL6100 has a narrower angle of transmission, so the same IR power is concentrated(focused) on a smaller area (vs a larger area for a wider angle IR LED).  The TSAL6100 comes into play when you want to use higher power and longer distances etc.

PS: Never a good idea to connect the IR led to the Arduino without a current limiting resistor.

SelaO, 2017-10-03:
I just read those leds are rated for max 50mA and burst max of 80mA, and the pwm pin supplies 40mA so I'm good without a resistor no?

AnalysIR, 2017-10-03:
You should **always** use a resistor if driving directly from an Arduino, to avoid damaging both. 
_A transistor driver circuit is better. A constant current driver is even better again._

Using a 74 Ohms resistor or greater with one led in series, should improve things for you (Assuming Arduino UNO or 5V AVR). After that progress on to a transistor driver circuit, if needed. (many basic examples are available via web search)

bengtmartensson, 2017-10-03:
> I just read the leds I use are rated for max 50mA and burst max of 80mA, and the pwm pin supplies 40mA so I'm good without a resistor since we use very short bursts no?

It appears as if you lack a background in electronics/electrical engineering. A resistor is not an "attenuator", it is a thing that converts voltage to current (or vice versa) by Ohms law. If you look at the equations, you will see that if you set R = 0, you will divide by 0. Differently put, the circuits will not be well defined (using the common models for the components). Practically, you will end up with a short circuit of some sort, likely damaging the components, unless there is some limiting resistor somewhere.

> _A constant current driver is even better again._

No, it is not, as I explained previously. Feel free to address my arguments, though.

AnalysIR, 2017-10-03:
> Feel free to address my arguments, though.

Sorry, not interested spending time debating this with you...I will accept that you have a different opinion, which I don't agree with.

AnalysIR, 2017-10-03:
Don't be concerned about your background...as everyone must start somewhere & nobody knows everything...so you are in a good place to get some help learning. 

I suggest you first start by picking the circuit you want to use. Then source the components. However, almost any common/general purpose NPN should do(2N2222 is very common/cheap). By combining resistors in parallel or series you can get the values required or close to them._(You may be able to salvage all the parts some from old electronics)_. Once you have fixed on a circuit...then you simple get the parts or equivalent parts and build it. 

I think you should do some research on this, as there are many examples & tutorials online.

Tip: Always check the data sheet for pinouts, before powering the circuit.

closed 2017-10-11 by SelaO