Arduino-IRremote/Arduino-IRremote #531: Electra A/C getting 99 array and not working :(

alonikomax, 2017-10-06:
**Board:** ARDUINO UNO
**Library Version:** 2.1.0
**Protocol:** It detects Samsung....

Using IRrecvDumpV2.

**checklist:**
- [x] I have **read** the README.md file thoroughly
- [x] I have searched existing issues to see if there is anything I have missed.
- [x] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [x] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [x] The title of the issue is helpful and relevant 

Encoding  : SAMSUNG
Code      : B24D7B84 (32 bits)
Timing[99]: 
     +4300, -4400     + 550, -1600     + 500, - 600     + 500, -1550
     + 550, -1650     + 500, - 500     + 550, - 600     + 500, -1600
     + 550, - 550     + 500, - 550     + 500, -1600     + 550, - 550
     + 500, - 600     + 500, -1550     + 550, -1650     + 500, - 550
     + 500, -1600     + 550, - 500     + 550, -1600     + 550, -1600
     + 550, -1550     + 600, -1550     + 550, - 550     + 550, -1600
     + 500, -1600     + 550, -1600     + 550, - 550     + 500, - 500
     + 550, - 550     + 550, - 550     + 500, -1600     + 550, - 500
     + 550, - 500     + 600, -1550     + 550, -1600     + 550, -1600
     + 550, - 500     + 550, - 500     + 550, - 550     + 550, - 500
     + 550, - 550     + 500, - 550     + 550, - 550     + 500, - 550
     + 500, -1600     + 550, -1600     + 550, -1600     + 550, -1550
     + 550, -1600     + 550
unsigned int  rawData[99] = {4300,4400, 550,1600, 500,600, 500,1550, 550,1650, 500,500, 550,600, 500,1600, 550,550, 500,550, 500,1600, 550,550, 500,600, 500,1550, 550,1650, 500,550, 500,1600, 550,500, 550,1600, 550,1600, 550,1550, 600,1550, 550,550, 550,1600, 500,1600, 550,1600, 550,550, 500,500, 550,550, 550,550, 500,1600, 550,500, 550,500, 600,1550, 550,1600, 550,1600, 550,500, 550,500, 550,550, 550,500, 550,550, 500,550, 550,550, 500,550, 500,1600, 550,1600, 550,1600, 550,1550, 550,1600, 550};  // SAMSUNG B24D7B84
unsigned int  data = 0xB24D7B84;

This is what I am getting for turning off

Encoding  : SAMSUNG
Code      : B24DBF40 (32 bits)
Timing[99]: 
     +4300, -4350     + 550, -1600     + 550, - 550     + 500, -1600
     + 550, -1600     + 500, - 550     + 550, - 550     + 500, -1600
     + 550, - 550     + 500, - 550     + 550, -1600     + 500, - 550
     + 550, - 550     + 500, -1600     + 550, -1600     + 500, - 550
     + 550, -1600     + 500, -1650     + 500, - 550     + 500, -1650
     + 500, -1600     + 550, -1600     + 550, -1600     + 500, -1600
     + 550, -1600     + 550, - 550     + 500, -1600     + 550, - 550
     + 500, - 550     + 500, - 550     + 550, - 550     + 500, - 600
     + 450, - 550     + 550, - 550     + 550, -1550     + 550, - 550
     + 500, -1600     + 550, - 550     + 500, - 550     + 550, - 550
     + 500, - 550     + 500, -1650     + 500, - 550     + 500, -1650
     + 500, - 550     + 500, -1650     + 500, -1600     + 550, -1600
     + 550, -1600     + 550
unsigned int  rawData[99] = {4300,4350, 550,1600, 550,550, 500,1600, 550,1600, 500,550, 550,550, 500,1600, 550,550, 500,550, 550,1600, 500,550, 550,550, 500,1600, 550,1600, 500,550, 550,1600, 500,1650, 500,550, 500,1650, 500,1600, 550,1600, 550,1600, 500,1600, 550,1600, 550,550, 500,1600, 550,550, 500,550, 500,550, 550,550, 500,600, 450,550, 550,550, 550,1550, 550,550, 500,1600, 550,550, 500,550, 550,550, 500,550, 500,1650, 500,550, 500,1650, 500,550, 500,1650, 500,1600, 550,1600, 550,1600, 550};  // SAMSUNG B24DBF40
unsigned int  data = 0xB24DBF40;

This is for turning on with 23* cold.
But the code is the same for every time im turning on, also with 25* or anything.

It looks incomplete, and I did expend the buffer to 512 and also 254.

Thanks for any help :)

bengtmartensson, 2017-10-06:
Increase RAWBUF (in IRremoteInt.h), for example to 201.

But the signals as given above look somewhat sane: Pasting them into IrScrutinizer makes them decode as 48-NEC. FWIW...

alonikomax, 2017-10-06:
@bengtmartensson as I mentioned, I tried 512 and 254.
Is this the most raw output I can get from an IR receiver?
Or maybe I can record better raw output without even a single touch of any program.
I tried all night to make this work, but it did not.

bengtmartensson, 2017-10-06:
> as I mentioned, I tried 512 and 254.

Sorry I missed that. As I wrote before, the signals do not look quite unplausible.

You can try with my [Infrared4Arduino](https://github.com/bengtmartensson/Infrared4Arduino) as an alternative, available through the Arduino library manager with the name "Infrared"; use the example "IrReceiverSampler".

AnalysIR, 2017-10-06:
Another option would be to try the following sketch:

[Air Conditioners: Recording long Infrared Remote control signals with Arduino](https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/)

alonikomax, 2017-10-06:
@AnalysIR thanks, what pin the reciver should be in?

AnalysIR, 2017-10-06:
it uses INT0, which is on pin 2 on standard arduinos (with ATmega328)....leoanardos & Megas have different pins for INT0....all covered via the link above.

alonikomax, 2017-10-06:
Yes, thank you, I got back 199 raw length this time, now testing it.

AnalysIR, 2017-10-06:
Great, please post some signals, for future reference.

alonikomax, 2017-10-06:
YESSS it works!!!!
Im going to record some codes and maybe do some reverse with a friend of mine.

AnalysIR, 2017-10-06:
Super!

I suspect your problem earlier, may have been that you did not restart the Arduino IDE...after changing the size of RAWBUF!

alonikomax, 2017-10-06:
OFF
Raw: (199) 4360, -4424, 548, -1608, 548, -536, 544, -1612, 544, -1612, 548, -532, 544, -532, 548, -1608, 548, -528, 548, -532, 548, -1608, 548, -532, 548, -528, 552, -1608, 544, -1612, 548, -528, 552, -1604, 548, -536, 544, -1612, 548, -1608, 544, -1608, 552, -1608, 548, -528, 548, -1612, 548, -1608, 544, -1612, 548, -532, 540, -536, 544, -536, 540, -536, 548, -1608, 548, -532, 544, -532, 544, -1608, 552, -1604, 552, -1608, 548, -532, 544, -536, 544, -532, 544, -532, 548, -528, 548, -532, 548, -532, 548, -532, 544, -1612, 540, -1616, 544, -1612, 544, -1612, 544, -1608, 552, -5200, 4360, -4424, 544, -1616, 540, -536, 544, -1612, 548, -1608, 544, -536, 544, -572, 504, -1612, 548, -532, 544, -536, 544, -1608, 548, -528, 552, -532, 548, -1608, 548, -1604, 552, -528, 548, -1608, 552, -528, 548, -1608, 544, -1612, 548, -1608, 548, -1608, 548, -528, 544, -1612, 548, -1608, 552, -1604, 552, -528, 544, -536, 548, -532, 548, -528, 548, -1608, 548, -532, 544, -532, 548, -1608, 548, -1608, 544, -1612, 544, -536, 544, -536, 544, -532, 544, -532, 552, -528, 548, -532, 544, -532, 548, -532, 544, -1612, 544, -1608, 548, -1608, 548, -1612, 548, -1608, 544, 

ON 20:
Raw: (199) 4356, -4428, 544, -1612, 544, -536, 548, -1608, 548, -1604, 548, -532, 548, -532, 548, -1608, 548, -532, 544, -532, 548, -1608, 544, -532, 548, -528, 552, -1604, 552, -1608, 552, -528, 544, -1612, 544, -1612, 548, -528, 548, -1608, 548, -1612, 548, -1604, 548, -1612, 544, -1608, 552, -1604, 552, -528, 548, -1608, 548, -532, 548, -528, 548, -536, 544, -532, 552, -524, 548, -532, 548, -532, 544, -532, 548, -1608, 544, -536, 548, -532, 544, -528, 548, -532, 548, -528, 552, -1604, 552, -1612, 540, -532, 548, -1608, 548, -1608, 548, -1612, 544, -1612, 544, -1608, 548, -5200, 4356, -4432, 548, -1608, 544, -532, 552, -1604, 548, -1608, 548, -532, 548, -528, 552, -1604, 552, -528, 548, -528, 548, -1612, 552, -528, 540, -536, 548, -1604, 552, -1608, 552, -524, 548, -1608, 552, -1604, 548, -532, 548, -1608, 544, -1612, 548, -1604, 552, -1604, 548, -1612, 548, -1608, 552, -528, 548, -1604, 552, -532, 544, -532, 548, -528, 552, -528, 548, -532, 544, -532, 544, -536, 552, -524, 548, -1608, 548, -536, 544, -528, 552, -528, 548, -532, 544, -532, 552, -1604, 544, -1608, 548, -536, 544, -1608, 548, -1608, 552, -1604, 552, -1604, 552, -1608, 548, 

ON 21:
Raw: (199) 4360, -4424, 544, -1608, 552, -528, 548, -1604, 552, -1608, 548, -532, 548, -528, 544, -1608, 552, -528, 544, -584, 496, -1608, 548, -532, 548, -532, 544, -1612, 544, -1608, 548, -532, 544, -1608, 548, -1612, 548, -528, 548, -1604, 556, -1604, 548, -1604, 548, -1608, 548, -1608, 552, -1604, 552, -528, 544, -1608, 552, -528, 548, -532, 544, -532, 552, -528, 548, -528, 552, -524, 548, -532, 548, -1604, 552, -1604, 552, -528, 544, -536, 548, -528, 544, -532, 552, -524, 548, -1612, 544, -532, 548, -532, 548, -1604, 552, -1604, 548, -1608, 552, -1604, 544, -1616, 544, -5200, 4360, -4424, 548, -1604, 548, -532, 548, -1608, 544, -1608, 552, -528, 548, -528, 544, -1612, 548, -532, 548, -532, 540, -1612, 544, -536, 544, -536, 544, -1608, 548, -1608, 544, -536, 548, -1608, 548, -1608, 544, -536, 544, -1608, 548, -1608, 544, -1608, 544, -1612, 548, -1612, 544, -1608, 544, -536, 544, -1612, 544, -536, 540, -532, 548, -532, 544, -572, 504, -536, 544, -536, 544, -532, 544, -1612, 548, -1608, 544, -584, 496, -532, 544, -536, 544, -532, 544, -532, 544, -1612, 544, -532, 552, -524, 548, -1612, 548, -1604, 552, -1604, 544, -1612, 548, -1608, 548, 

ON 22:
Raw: (199) 4360, -4428, 552, -1604, 548, -532, 548, -1604, 552, -1604, 552, -532, 540, -536, 548, -1608, 548, -528, 548, -528, 548, -1608, 552, -528, 548, -532, 548, -1604, 552, -1604, 548, -532, 548, -1608, 552, -1604, 552, -528, 544, -1612, 552, -1608, 544, -1612, 548, -1608, 544, -1612, 540, -1616, 548, -532, 544, -1612, 544, -536, 544, -532, 544, -532, 548, -532, 544, -536, 540, -536, 544, -536, 540, -1612, 544, -1616, 540, -1616, 540, -536, 548, -532, 544, -532, 544, -536, 536, -1616, 548, -532, 548, -532, 544, -532, 548, -1608, 544, -1612, 544, -1612, 544, -1612, 540, -5208, 4360, -4424, 544, -1616, 544, -536, 540, -1616, 540, -1612, 540, -540, 544, -536, 536, -1616, 544, -536, 544, -536, 544, -1608, 548, -532, 548, -532, 544, -1612, 544, -1612, 544, -536, 544, -1612, 544, -1612, 544, -532, 548, -1608, 548, -1612, 540, -1612, 544, -1612, 544, -1616, 540, -1612, 544, -540, 540, -1616, 536, -536, 548, -532, 548, -528, 552, -528, 544, -536, 544, -532, 544, -536, 544, -1612, 544, -1612, 544, -1612, 540, -536, 544, -536, 544, -532, 544, -532, 552, -1608, 544, -532, 548, -532, 540, -536, 544, -1612, 548, -1608, 548, -1612, 544, -1608, 548, 

ON 23:
Raw: (199) 4384, -4404, 552, -1604, 552, -524, 556, -1600, 556, -1600, 556, -524, 556, -524, 552, -1600, 556, -524, 552, -524, 556, -1600, 552, -528, 552, -524, 556, -1600, 556, -1604, 552, -520, 560, -1600, 552, -1604, 556, -524, 556, -1596, 556, -1600, 560, -1600, 552, -1600, 560, -1600, 552, -1604, 556, -520, 556, -1596, 560, -524, 556, -520, 556, -524, 556, -524, 556, -520, 556, -572, 504, -524, 552, -1600, 556, -528, 552, -1604, 552, -524, 556, -524, 552, -524, 556, -524, 548, -1604, 560, -520, 556, -1600, 556, -524, 552, -1604, 552, -1600, 556, -1604, 552, -1604, 556, -5192, 4384, -4404, 552, -1600, 556, -524, 552, -1600, 552, -1604, 556, -524, 552, -528, 552, -1600, 560, -520, 556, -524, 548, -1608, 556, -520, 552, -528, 556, -1600, 552, -1600, 560, -520, 556, -1600, 556, -1600, 556, -524, 556, -1596, 552, -16 [... cut ...] 52, -1608, 552, -524, 552, -576, 508, -1600, 552, -1600, 556, -524, 556, -1600, 552, -1604, 556, -524, 556, -1600, 552, -1604, 552, -1604, 556, -1600, 552, -1600, 556, -1604, 552, -528, 552, -1600, 556, -524, 556, -524, 552, -524, 556, -524, 548, -528, 556, -520, 556, -1600, 556, -524, 552, -1604, 552, -1600, 556, -524, 556, -524, 552, -528, 552, -524, 552, -524, 556, -1600, 556, -524, 556, -520, 556, -1600, 556, -1600, 552, -1604, 552, -1604, 556, 

Some codes. Thanks @bengtmartensson @AnalysIR

closed 2017-10-06 by alonikomax

ilfuriano, 2018-05-07:
@bengtmartensson Infrared4Arduino works like a charm.
It dumps clean signal for use it as raw data. Thanks!