Arduino-IRremote/Arduino-IRremote #532: UNKNOWN Command, each time different raw data

laserir, 2017-10-08:
Dear community, first of all THANKS A LOT for this great library!! It helped me a lot with various projects. For the last weeks I tried to figure out how to decode an unknown IR signal, hope you can help. (I don't need to resend it, just recognize different signals.) Every time the data seems to be different although it is the same button/device.

Attached you find the raw data of a few attempts. Thanks in advance!!!

`Encoding  : UNKNOWN
Code      : 74074BF7 (32 bits)
Timing[23]: 
     + 350, - 200     + 450, - 400     + 300, - 250     + 400, - 200
     + 500, - 300     + 300, - 350     + 400, - 150     + 450, - 400
     + 250, - 350     + 400, - 150     + 450, - 650     + 750
unsigned int  rawData[23] = {350,200, 450,400, 300,250, 400,200, 500,300, 300,350, 400,150, 450,400, 250,350, 400,150, 450,650, 750};  // UNKNOWN 74074BF7

Encoding  : UNKNOWN
Code      : 7FE2F6FE (32 bits)
Timing[23]: 
     + 250, - 400     + 400, - 200     + 400, - 350     + 350, - 250
     + 400, - 200     + 450, - 400     + 250, - 350     + 350, - 200
     + 400, - 450     + 250, - 350     + 400, - 600     + 650
unsigned int  rawData[23] = {250,400, 400,200, 400,350, 350,250, 400,200, 450,400, 250,350, 350,200, 400,450, 250,350, 400,600, 650};  // UNKNOWN 7FE2F6FE

Encoding  : UNKNOWN
Code      : DAC7D8ED (32 bits)
Timing[23]: 
     + 400, - 200     + 350, - 450     + 300, - 250     + 400, - 250
     + 400, - 400     + 300, - 300     + 400, - 200     + 400, - 400
     + 300, - 300     + 400, - 200     + 350, - 750     + 600
unsigned int  rawData[23] = {400,200, 350,450, 300,250, 400,250, 400,400, 300,300, 400,200, 400,400, 300,300, 400,200, 350,750, 600};  // UNKNOWN DAC7D8ED

Encoding  : UNKNOWN
Code      : 12C2E8A1 (32 bits)
Timing[23]: 
     + 400, - 250     + 350, - 450     + 300, - 200     + 450, - 250
     + 400, - 350     + 300, - 300     + 400, - 200     + 400, - 450
     + 300, - 250     + 400, - 200     + 350, - 750     + 700
unsigned int  rawData[23] = {400,250, 350,450, 300,200, 450,250, 400,350, 300,300, 400,200, 400,450, 300,250, 400,200, 350,750, 700};  // UNKNOWN 12C2E8A1

Encoding  : UNKNOWN
Code      : CC6DE2E9 (32 bits)
Timing[23]: 
     + 400, - 200     + 350, - 500     + 250, - 300     + 400, - 200
     + 350, - 450     + 250, - 350     + 300, - 350     + 350, - 400
     + 250, - 350     + 400, - 200     + 400, - 700     + 750
unsigned int  rawData[23] = {400,200, 350,500, 250,300, 400,200, 350,450, 250,350, 300,350, 350,400, 250,350, 400,200, 400,700, 750};  // UNKNOWN CC6DE2E9

Encoding  : UNKNOWN
Code      : BD36B682 (32 bits)
Timing[23]: 
     + 300, - 300     + 400, - 300     + 350, - 400     + 300, - 200
     + 450, - 300     + 350, - 400     + 250, - 350     + 350, - 300
     + 350, - 400     + 250, - 350     + 400, - 650     + 650
unsigned int  rawData[23] = {300,300, 400,300, 350,400, 300,200, 450,300, 350,400, 250,350, 350,300, 350,400, 250,350, 400,650, 650};  // UNKNOWN BD36B682`

AnalysIR, 2017-10-09:
A couple of points:

1. The captures above look inconsistent, which could be due to your IR recevier or your positioning of the remote too close/too far to/from the receiver (1-2m/3-6ft is usually OK). This may explain some of the issues you are seeing.
2. If all you need is to identify different signals from each other, why not try using the 'hashing' function from the original author of IRremote. [Using arbitrary remotes with the Arduino IRremote library ](http://www.righto.com/2010/01/using-arbitrary-remotes-with-arduino.html) However, you will need to get more consistent captures first.

**Tip:** In gerneral, it is a good idea to post the receiver being used, the model of remote and target device. That way you may get more meaningful assistance.

laserir, 2017-10-09:
The "remote" is a LaserTag-phaser gun that emits an ir beam. I suspect every unit has a unique ID, so that the scoring works. I may be that some altering data (like "remaining shots" etc.) are transmitted as well and therefore the raw data differs each time... But I think 12 pulse is not enough for that amount of data? I only need the (not changing) unit ID to determine which unit hit the receiver.

I'm using an "ST1087" arduino ir receiver module. Positioning shouldn't be a problem as the system was fixed when I captured the raw data.

Already tried your second approach (Using arbitrary...), but no luck there, mainly because the signals are to different each time.

AnalysIR, 2017-10-09:
Ok..tnx

That ST1087 seems to be an IR emitter. However, the ST1089 is related to it _(if not post a link)_. If that is the one I think it is, it is a very poor IR Receiver & **could** be the source of your issues. Also, the docs I found for the ST1089, do not even mention carrier frequency, which could be another issue for you.

So my suggestion is to get a better IR receiver...something like a TSOP34438 from Vishay, would be a good general purpose choice.

You could also try out either of the following:
[Poor maker’s Infrared receiver](https://www.analysir.com/blog/2014/05/28/poor-makers-infrared-receiver-2/)
or if you have an oscilloscope, you could easily make one of these & capture the timings & carrier frequency manually from the scope:
[‘Silver bullet’ – the Oscilloscope Infrared Receiver](https://www.analysir.com/blog/2014/05/04/silver-bullet-oscilloscope-infrared-receiver/)

PS: If by **fixed** you meant not moving...the real point is no closer than 1m/3ft!

laserir, 2017-10-09:
Sorry... I mixed up order no. and component. I have a VS1883 connected to my UNO.

Another thing... I can capture and compare the data transmitted by "arena mines" of the same lasertag-system. These don't identify with an ID, but only tell the units to loose their lives if they are of a certain team. Capturing those works with the receiver. So maybe it's just about filtering the information above for the non-changing unit ID... but I'm quite a noob and don't know to start with that -.-

By fixed I meant not moving, yes. I tried distances between 1-4 meters, all with the same result.

AnalysIR, 2017-10-09:
VS1838 vs VS1883???

try recording a set of signals using the following script, it will give better resolution on the timings:
[](https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/)

With more consistent timings, you may be able to figure out the individual timings & thus the coding pattern.

The problem above, is that some of the times are as low as 150/200 uSecs, which is within the error margin (200uSecs) of most standard IR receivers.

laserir, 2017-10-10:
The manufacturer of the system told me they use "TSOP4136" receivers to catch the signals properly.

Do you think the VS 1838 has problems with this carrier frequency resulting in the pattern variations?

AnalysIR, 2017-10-10:
your receiver is 38kHz vs 36kHz of the TSOP4136.

with a typical receiver & most IR protocols that wouldn't be an issue and would just result in slightly reduced range. I cannot say ihow f your receiver would behave.

It looks like the link in my previous post did not come thru....so here it is to try.https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/

It should give a better view of the actual timings & hopefully more consistent results.

laserir, 2017-10-11:
Here are 5 samples of the Units with ID 1 and 2... Don't see a pattern here, maybe you do?

**Unit 1:**
Raw: (25) 304, -392, 304, -360, 392, -276, 364, -356, 280, -356, 336, -332, 336, -384, 304, -308, 392, -276, 364, -356, 276, -364, 392, -604, 416, 

Raw: (25) 360, -364, 336, -304, 364, -300, 364, -356, 280, -360, 392, -276, 360, -356, 316, -324, 396, -272, 360, -332, 340, -300, 420, -580, 416, 

Raw: (25) 280, -360, 420, -276, 360, -332, 332, -308, 420, -276, 336, -356, 304, -332, 392, -276, 360, -360, 308, -332, 388, -276, 360, -640, 364, 

Raw: (25) 332, -308, 416, -276, 360, -336, 332, -328, 392, -276, 304, -388, 300, -340, 360, -336, 336, -356, 280, -360, 360, -304, 364, -664, 356, 

Raw: (25) 312, -328, 360, -308, 364, -356, 276, -364, 360, -308, 364, -356, 308, -332, 384, -280, 356, -368, 300, -308, 360, -332, 336, -664, 360, 

**Unit 2:**
Raw: (23) 304, -312, 416, -280, 300, -420, 304, -304, 388, -280, 360, -388, 276, -336, 388, -280, 356, -364, 304, -332, 392, -612, 688, 

Raw: (23) 308, -336, 416, -248, 360, -364, 272, -368, 412, -252, 356, -368, 300, -308, 416, -276, 360, -336, 332, -332, 388, -584, 664, 

Raw: (23) 336, -364, 332, -308, 412, -276, 304, -392, 332, -304, 392, -276, 360, -364, 304, -336, 384, -280, 360, -360, 308, -692, 696, 

Raw: (23) 388, -308, 384, -364, 332, -308, 412, -284, 360, -448, 216, -336, 388, -280, 352, -368, 300, -340, 328, -332, 360, -644, 708, 

Raw: (23) 388, -312, 308, -388, 328, -308, 412, -256, 360, -360, 276, -364, 380, -288, 356, -364, 300, -336, 412, -256, 300, -704, 744,

AnalysIR, 2017-10-11:
I just checked the first signals from both sets & the reception/timing is much better than before, but still not very consistent.

The main differences seem to be in the last 2 or 3 marks/spaces.

Options: 
1. could be a count of the number of pulse are different between both
2. could be using mancester encoding (like RC5/RC6). One clue could be that @36kHz is the same frequency as RC5/RC6 signals
3. or it could just be pre-define 'arbitrary' timings for each command. (probably not)

However, the consistency of timings is not great & to be honest it is not really worth spending any more  time on this - using that particular receiver. A good one will cost less than $1 from a main distributor. Look for TSOPxx36, TSOP34436 or the one you mentioned. The one you are using now usually costs <$0.10 for a reason!

**UPDATE**: I can now decode some of the first set using AnalysIR as RCMM...using a very large tolerance factor - this could be a coincidence. The second set isn't playing as nice. Again, likely a receiver issue.

laserir, 2017-10-11:
This is exactly the same receiver as used in the units... for the capture above I used an **TSOP4136** !
So they somehow can work with the tolerance.
Had the same idea as mentioned in your Option 1 (count of pulses), but sadly some units send 23, some 25 pulses, also without a pattern I could see.

AnalysIR, 2017-10-11:
> This is exactly the same receiver as used in the units... for the capture above I used an TSOP4136 ! (Bought today!)
OK, looks like I missed that bit in your previous post...sorry.

I will look more closely at the timings to see if I can find something else.

Some other points:
1. Check the batteries in the remote are good.
2. make sure the remote is 1m away, as mentioned above
3. What type of Arduino/MCU are you using. does it have a crystal? Theoretically bad drift could be an issue here, but that would not be common.

AnalysIR, 2017-10-11:
OK ...looked at it in more detail. The overall timings are unusually inconsistent....suggesting there could be a physical issue somewhere???

The RCMM connection seems to be a coincidence.

Please capture some more different signals as above... / different commands & post (max 2 copies of each signal). I will try to look at them tomorrow.

AnalysIR, 2017-10-11:
Here is a shot of the 2 signals for unit 1 & 2. The main difference is the sequence at the end...could be as simple as that.
You can also see the unusually big variance in the timings.
[image]

laserir, 2017-10-12:
Tomorrow I can take some more signals from units 1-10.
I'm using an UNO made by Elegoo, it has a crystal on board.

Just for a check I captured a normal tv remote "power" button with your sketch; as you see these signals seem quite consistent, so I suppose the hardware works fine?!

**Raw-Data of TV remote (not the units I am looking to decode, just for testing):**
Raw: (71) 9088, -4464, 612, -544, 612, -548, 612, -544, 612, -544, 616, -548, 612, -544, 612, -544, 612, -548, 612, -1620, 616, -1620, 616, -1616, 620, -1616, 616, -1620, 616, -1620, 616, -1620, 616, -1620, 616, -544, 616, -544, 640, -1596, 640, -516, 616, -544, 640, -520, 616, -1620, 616, -544, 640, -1596, 612, -1620, 616, -544, 612, -1624, 612, -1624, 640, -1596, 612, -544, 620, -1616, 620, -39360, 9112, -2196, 620, 

Raw: (71) 9072, -4460, 616, -544, 612, -544, 612, -544, 612, -544, 612, -548, 612, -544, 616, -544, 612, -544, 612, -1624, 612, -1620, 612, -1624, 612, -1620, 616, -1620, 616, -1616, 620, -1620, 616, -1620, 616, -544, 612, -544, 616, -1620, 616, -540, 620, -544, 616, -544, 612, -1620, 616, -544, 616, -1620, 616, -1620, 616, -544, 620, -1616, 620, -1616, 620, -1616, 620, -536, 648, -1588, 620, -39364, 9108, -2196, 616,

AnalysIR, 2017-10-12:
Yes the TV signals looks fine, no major issues.

So that would leave the laser-tag device, which sends the IR. Make sure the battery is OK.
You already confirmed that it works with the live system.....so that would suggest the tolerances are high/wide.

The additional signals should help....to figure out the bit pattern/encoding etc.

laserir, 2017-10-13:
So here I captured units 1-15, each 3 times... All batteries full.
When playing back the captured codes (sendraw) all of them work as expected.

**Unit 1**
Raw: (25) 368, -352, 336, -308, 388, -276, 364, -356, 280, -360, 332, -336, 360, -360, 308, -300, 416, -280, 356, -360, 312, -328, 336, -636, 424, 
Raw: (25) 328, -400, 272, -360, 388, -280, 360, -364, 272, -364, 332, -336, 332, -388, 304, -308, 388, -280, 356, -364, 272, -368, 384, -612, 408, 
Raw: (25) 416, -284, 356, -336, 328, -336, 384, -308, 328, -368, 304, -308, 412, -280, 328, -368, 272, -368, 376, -312, 332, -392, 276, -700, 272, 

**Unit 2**
Raw: (23) 308, -340, 408, -284, 332, -388, 276, -336, 356, -340, 332, -360, 300, -364, 360, -304, 280, -444, 276, -336, 384, -640, 696, 
Raw: (23) 332, -308, 388, -280, 328, -420, 308, -304, 384, -308, 332, -364, 272, -368, 384, -276, 364, -360, 308, -332, 384, -612, 696, 
Raw: (23) 392, -332, 340, -272, 448, -276, 364, -328, 340, -272, 396, -296, 340, -380, 312, -296, 400, -272, 364, -352, 284, -720, 672, 

**Unit 3**
Raw: (25) 364, -364, 332, -308, 352, -340, 300, -396, 296, -340, 380, -284, 360, -364, 304, -336, 328, -336, 356, -368, 304, -720, 300, -312, 388, 
Raw: (25) 312, -300, 388, -312, 360, -360, 308, -328, 392, -276, 304, -416, 300, -312, 412, -256, 328, -392, 324, -316, 360, -640, 436, -256, 332, 
Raw: (25) 368, -352, 336, -304, 420, -276, 336, -356, 328, -312, 388, -280, 364, -356, 304, -332, 384, -284, 304, -416, 308, -688, 332, -280, 420, 

**Unit 4**
Raw: (23) 332, -308, 424, -272, 304, -388, 340, -300, 360, -336, 336, -356, 280, -360, 392, -272, 364, -360, 304, -696, 668, -356, 308, 
Raw: (23) 392, -276, 364, -384, 308, -300, 396, -276, 360, -356, 308, -336, 388, -276, 364, -356, 308, -332, 356, -640, 696, -304, 396, 
Raw: (23) 336, -304, 416, -276, 308, -416, 308, -304, 360, -332, 336, -360, 328, -308, 388, -276, 364, -360, 308, -688, 668, -364, 304, 

**Unit 5**
Raw: (23) 392, -280, 328, -388, 332, -308, 380, -284, 364, -360, 308, -332, 384, -280, 364, -360, 304, -328, 388, -616, 696, -688, 308, 
Raw: (23) 396, -276, 328, -420, 308, -300, 392, -300, 308, -388, 300, -336, 392, -276, 364, -360, 308, -328, 392, -608, 692, -696, 308, 
Raw: (23) 308, -332, 360, -304, 364, -360, 304, -332, 392, -276, 300, -420, 308, -332, 384, -280, 308, -416, 308, -688, 640, -664, 308, 

**Unit 6**
Raw: (23) 308, -332, 388, -276, 364, -360, 276, -364, 384, -280, 308, -408, 312, -328, 332, -336, 336, -388, 304, -664, 336, -304, 664, 
Raw: (23) 396, -276, 356, -360, 336, -304, 392, -276, 360, -360, 308, -332, 412, -280, 276, -416, 308, -332, 332, -640, 388, -304, 728, 
Raw: (23) 332, -308, 420, -272, 304, -392, 336, -304, 356, -308, 360, -364, 276, -356, 392, -276, 328, -396, 308, -688, 280, -360, 696, 

**Unit 7**
Raw: (25) 416, -280, 364, -360, 304, -328, 392, -276, 332, -364, 332, -304, 360, -308, 364, -356, 280, -360, 392, -608, 416, -252, 384, -336, 332, 
Raw: (25) 396, -276, 384, -332, 340, -300, 364, -304, 364, -356, 312, -328, 392, -276, 364, -356, 304, -332, 392, -612, 360, -300, 364, -360, 308, 
Raw: (25) 360, -364, 308, -300, 424, -272, 360, -332, 336, -332, 388, -252, 332, -388, 328, -312, 360, -332, 332, -668, 364, -332, 332, -300, 388, 

**Unit 8**
Raw: (23) 420, -276, 304, -388, 340, -300, 360, -336, 304, -388, 304, -332, 392, -276, 364, -356, 280, -720, 612, -416, 304, -304, 420, 
Raw: (23) 336, -308, 356, -340, 300, -420, 304, -308, 356, -340, 328, -392, 296, -312, 384, -284, 360, -664, 720, -308, 332, -388, 280, 
Raw: (23) 360, -368, 304, -328, 392, -276, 360, -364, 304, -308, 384, -280, 328, -396, 304, -332, 388, -612, 684, -344, 380, -284, 364, 

**Unit 9**
Raw: (23) 368, -356, 328, -308, 360, -308, 352, -368, 336, -304, 392, -276, 360, -360, 308, -328, 392, -608, 696, -332, 388, -612, 420, 
Raw: (23) 308, -332, 392, -272, 364, -360, 276, -364, 384, -284, 304, -416, 304, -332, 332, -336, 332, -664, 724, -276, 364, -664, 356, 
Raw: (23) 364, -364, 300, -360, 392, -276, 328, -392, 312, -300, 388, -280, 360, -364, 324, -312, 388, -608, 668, -364, 384, -612, 388, 

**Unit 10**
Raw: (21) 396, -280, 376, -348, 324, -312, 408, -256, 384, -364, 268, -344, 388, -280, 360, -360, 300, -700, 656, -648, 740, 
Raw: (21) 364, -388, 332, -308, 440, -256, 360, -388, 300, -340, 388, -304, 360, -392, 304, -332, 412, -588, 720, -692, 668, 
Raw: (21) 396, -276, 360, -356, 332, -308, 392, -276, 364, -356, 308, -332, 392, -276, 352, -368, 300, -700, 640, -660, 728, 

**Unit 11**
Raw: (23) 312, -328, 416, -252, 364, -360, 276, -360, 416, -252, 360, -364, 304, -336, 384, -276, 360, -644, 744, -612, 392, -276, 360, 
Raw: (23) 360, -304, 388, -336, 300, -340, 412, -252, 364, -360, 304, -328, 392, -276, 364, -356, 312, -692, 664, -640, 332, -388, 276, 
Raw: (23) 420, -272, 360, -364, 308, -304, 416, -276, 360, -336, 332, -304, 412, -284, 332, -360, 276, -752, 636, -640, 388, -332, 276, 

**Unit 12**
Raw: (23) 364, -304, 392, -328, 336, -332, 388, -252, 384, -364, 300, -308, 392, -276, 364, -356, 280, -720, 336, -304, 692, -332, 392, 
Raw: (23) 304, -416, 312, -328, 388, -252, 388, -360, 304, -308, 388, -276, 364, -356, 312, -328, 392, -608, 416, -276, 668, -336, 360, 
Raw: (23) 388, -280, 360, -364, 304, -304, 416, -276, 328, -396, 308, -304, 360, -332, 332, -392, 304, -664, 332, -304, 668, -364, 388, 

**Unit 13**
Raw: (23) 336, -308, 408, -256, 384, -340, 324, -316, 408, -308, 304, -388, 244, -368, 388, -280, 360, -664, 360, -364, 580, -696, 360, 
[... cut ...]
Raw: (23) 392, -280, 328, -392, 336, -304, 356, -340, 300, -392, 300, -336, 388, -280, 356, -364, 272, -732, 328, -308, 384, -280, 744, 

**Unit 15**
Raw: (25) 388, -308, 360, -364, 304, -308, 380, -312, 356, -340, 272, -368, 384, -276, 332, -392, 304, -696, 300, -336, 416, -280, 300, -392, 332, 
Raw: (25) 280, -360, 360, -308, 332, -388, 308, -332, 360, -308, 360, -360, 304, -304, 412, -284, 304, -692, 364, -360, 276, -336, 416, -276, 328, 
Raw: (25) 332, -416, 308, -304, 380, -312, 360, -364, 308, -304, 384, -308, 332, -364, 272, -364, 388, -612, 416, -252, 384, -336, 332, -304, 360,

AnalysIR, 2017-10-13:
It looks like it is based on **_manchester encoding._**

similar to RC5/RC6 but different timings, #bits etc
see [RC5](http://www.sbprojects.com/knowledge/ir/rc5.php) to get an idea.

The bits seem to be changing at the right of the trace I posted above, in line with unit 1,2,3,4 binary values etc

80%+ confident it is this....but not 100% certain. ( I only spent limited time on it using AnalysIR)

It will require a bit more work to verify this. :)

The timings are pretty inconsistent, but as you said the target device can handle it. _Manceshter encloding can be good for removing timing errors, which is why they may have used it - thus allowing them to build a remote/gun with lower BoM costs????_

Let us know how you get on!

laserir, 2017-10-13:
Maybe I found a hint while searching through old documentation of the system. It is about a testing device (which I don't have) and reads:

_"The testing device emits IR code and can be used to simulate any unit. The code is set with a pair of hexadecimal switches. It is necessary to use hexadecimal in order to obtain the range of codes. Hexadecimal uses the base of 16, instead of 8,9,10,11 you have $8,$9,$A,$B,$C,$D,$E,$F,$10,$11, where $11 = 17.

The codes are:
Red Units: $1 - $80
Green Units: $81 - $A0 ($81=Green 1, $A0=Green 80)
(...there follow more codes concerning other arena equipment...)"_

Important to know: **The units from above codes were all set to RED while capturing the codes!**

Maybe this could help? Again, thank you so much for your help!! I just want to keep an old wonderful system alive :)

AnalysIR, 2017-10-13:
Yes that is consitent with what I saw

If you post just 5 **green** signals for unit **1,2,3,4,5**, I can check that using AnalysIR.

rgds

AnalysIR, 2017-10-13:
PS:
put any bulk data inside code tags **<>** above

laserir, 2017-10-14:
**Unit 1 green**
`Raw: (21) 364, -332, 304, -364, 388, -280, 356, -336, 332, -304, 420, -608, 696, -692, 636, -364, 328, -312, 412, -584, 416, 
Raw: (21) 356, -364, 332, -308, 384, -284, 356, -364, 328, -308, 384, -616, 692, -692, 636, -396, 328, -312, 412, -612, 440, 
Raw: (21) 364, -364, 332, -336, 384, -276, 332, -364, 332, -304, 416, -584, 692, -724, 640, -360, 304, -332, 392, -612, 380, `

**Unit 2 green**
`Raw: (19) 364, -360, 328, -308, 392, -276, 360, -360, 304, -336, 388, -612, 692, -696, 664, -392, 304, -720, 664, 
Raw: (19) 364, -300, 360, -364, 304, -308, 412, -280, 328, -396, 304, -664, 668, -668, 716, -280, 360, -640, 748, 
Raw: (19) 308, -332, 388, -280, 356, -364, 276, -364, 384, -284, 356, -640, 752, -608, 692, -308, 412, -612, 640, `

**Unit 3 green**
`Raw: (21) 360, -364, 272, -360, 420, -248, 360, -360, 332, -308, 388, -612, 692, -696, 636, -360, 332, -700, 300, -308, 416, 
Raw: (21) 392, -276, 384, -336, 332, -308, 412, -256, 356, -364, 300, -696, 668, -640, 748, -276, 360, -644, 352, -368, 300, 
Raw: (21) 336, -304, 416, -276, 392, -332, 364, -300, 448, -248, 360, -696, 716, -612, 696, -308, 380, -620, 412, -280, 360, `

**Unit 4 green**
`Raw: (19) 304, -308, 416, -280, 360, -360, 304, -304, 420, -276, 324, -672, 748, -584, 696, -692, 668, -360, 304, 
Raw: (19) 364, -364, 332, -308, 412, -248, 364, -388, 300, -308, 392, -612, 688, -700, 664, -636, 724, -308, 356, 
Raw: (19) 336, -304, 412, -280, 360, -360, 304, -308, 420, -276, 332, -636, 744, -620, 688, -724, 640, -360, 304, `

**Unit 5 green**
`Raw: (19) 332, -308, 412, -280, 360, -336, 332, -304, 444, -248, 332, -644, 744, -612, 696, -692, 668, -636, 360,
Raw: (19) 388, -280, 356, -364, 332, -308, 380, -288, 356, -364, 300, -700, 660, -640, 748, -612, 696, -692, 304, 
Raw: (19) 412, -252, 360, -360, 332, -308, 416, -252, 360, -360, 300, -700, 660, -644, 744, -612, 696, -692, 308, `

AnalysIR, 2017-10-14:
Yes that is still consistent with what I posted above.

However the numbering for Green appear to be consistent with $5n rather than $8n....so either its not green or your colour to HEX numbering is off a bit...etc...(note 0x5 is the inverse of 0xA)

Just manually checked with the help of AnalysIR.... Unit3 Green and its 0x53,,,Unit1 Green is 0x51,,,Unit4 Green is 0x54,,,Unit5 Green is 0x55....from your last post)

The good nows is that you should be able to make some progress, using manchester encoding, as the pattern is there.
You should be able to use the library code for RC5 or RC6 as a basis, using the timings below. There are no headers or toggle bits evident...just a straight binary bitstream.

The number of data bits seems to be 13 (unless the first one serves as a header)

**Timing**
As I mentioned before the timing is all over the place. I suggest using a base timing between 600 & 675 uSecs
So a binary 1 bit will be a space of 666 followed by a mark of 666 uSecs
A binary 0 will be a space of 666 followed by a mark of 666 uSecs

Vary the timing used until you get the best results.

AnalysIR, 2017-10-14:
Out of curiosity, I went back and compared your origninal signals from the VS1838 and it would not have been possible to figure it out using the signals captured from that one. Yet another reason to avoid that particular IR receiver.

laserir, 2017-10-14:
I was thinking of buying your "A.IR Shield Nano, advanced infrared module". Maybe then I have a good piece of hardware for both capturing and sending these signals?

In what part of the signal could you find the hex codes? (RC5 address or command...?)
Can I adapt the irremote library for this? I'm not quite sure which variables in the ir_RC5_RC6.cpp file need to be changed to successfully decode the signal.

AnalysIR, 2017-10-14:
If you read up on [RC5/RC6 ](http://www.sbprojects.com/knowledge/ir/rc5.php) you will then understand how it fits together.

The signal is all data bits, from start to end. Where you see the wider marks or spaces that is where you get a 1&0 or a 0&1 joined together....which is the trick about manchester encoding..

The screen grab below, shows Green 4 (on top) & Green 5 (Below)
[image]

Looking at RC5 it seems to be the closest match...so try changing

```
#define MIN_RC5_SAMPLES     11
#define RC5_T1 889
```
to
```
#define MIN_RC5_SAMPLES     13  //or leave at 11 and try 12 also if it doesn't work
#define RC5_T1 666   //approximate timing basedo n your signals.....if it doesn't work try from 600 to 750
```

You could just do a quick test by changing the RC5_T1 only and try it out. **Note:** Sometimes you need to restart the Arduino IDE for changes to take.

For sending:
`sendRC5 (0x1, 13); // could work. with just changing RC5_T1  to 666`

I suggest you initially just change/edit  the RC5 code in the library as is (pretending it is an RC5 signal), with the changes above. If you get it working...then you can go about creating your own files and issue a pull request for inclusion in the library proper. Don't forget to restore your original RC5 file when finished.

With luck...it could be as simple as that :)

laserir, 2017-10-17:
What I have tried now by changing the **ir_RC5_RC6.cpp** file:
`//#define MIN_RC5_SAMPLES     11
#define MIN_RC5_SAMPLES     13
//#define RC5_T1             889
#define RC5_T1             666
#define RC5_RPT_LENGTH   46000`

Also I commented out this part (as there will be no fixed start bit, as you explained):
`	// Get start bits
	//if (getRClevel(results, &offset, &used, RC5_T1) != MARK)   return false ;
	//if (getRClevel(results, &offset, &used, RC5_T1) != SPACE)  return false ;
	//if (getRClevel(results, &offset, &used, RC5_T1) != MARK)   return false ;`

Tried with the irremote test sketch (using digital pin 11 for input). No luck with that... Example:
`Encoding  : UNKNOWN
Code      : DCECD3FB (32 bits)
Timing[23]: 
     + 350, - 300     + 300, - 400     + 300, - 400     + 200, - 400
     + 350, - 300     + 300, - 400     + 250, - 350     + 300, - 400
     + 300, - 400     + 200, - 350     + 400, - 550     + 650
unsigned int  rawData[23] = {350,300, 300,400, 300,400, 200,400, 350,300, 300,400, 250,350, 300,400, 300,400, 200,350, 400,550, 650};  // UNKNOWN DCECD3FB`

Shall I use another sketch? :-(

AnalysIR, 2017-10-17:
Sorry...big mistake....

use 
`#define RC5_T1 333 //Also try 300-375`

instead of
`#define RC5_T1 666 //Also tried 600-700`

each bit is 666uSecs long, but each space/mark is half of this => 333 (approx.)

Sorry for the confusion. At 333 uSecs its no wonder its difficult....as the error margin on IR receivers can be up to 200uSecs.

laserir, 2017-10-19:
Over the last days I tried it with all demo sketches, but no luck. Still can't figure out how to simply retrieve the hex value or even raw binary of the signals. Do you have a demo sketch/function for manchester decoding? I also tried to adapt the RC5 library from guyc and arduino-libs-manchester from mchr3k - no luck.

PS: I just bought two A.IR shield nano to get better results. I'd even consider buying the "big" version of AnalysIR, if we could only get this to work :(

laserir, 2017-10-27:
So here comes the **first part** of the solution: **Sending the signal**

First of all... Everyone facing a similar problem take my advice: I tried to use different hardware components to capture the original signals. First a cheap VS1838, later TSOP4136 (as we found out the signal is @ 36kHz). Even though the second one was better, I ordered the "A.IR Shield Nano" from AnalysIR, which works perfectly with irremote. **Please use quality hardware for unknown signals, otherwise you waste your time!**

First we couldn't figure out which format the signals have. This was due to the fact that it was manchester encoded, but does not use start bits. So it works kind of RC5, but not completely. The main problem is that the timing of the original signals in very very inconsistent. Anyways with (lots and lots) of help from AnalysIR (thank you so much!!!) and the use of his software he managed to find out that the data bursts just send a plain manchester encoded bitstream. With some pieces of old system-documentation we could translate the signals to hex values. It was important to find out the right timing. RC5 is about 890 uSec per bit, this specific code uses about 333 uSec. Also it is important to send the data as unsigned long.

With the following sketch it is possible to send back the signals. (Signal mode is switchable with a remote) Sorry for the formatting, the code-block doesn't seem to work for this -.-

#include <IRremote.h>
IRsend irsend;
IRrecv irrecv(2);
decode_results results;

int mode = 0; //Start without emitting code

#define RC5_T1 333 // Set timing

void setup()
{
  Serial.begin(9600);
  irrecv.enableIRIn();
}

void loop() {
    checkIR(); //Chance mode?
    switch(mode){
      case 1:mySendRC5Laser(0x1UL, 13);break; // Red 1
      case 2:mySendRC5Laser(0x2UL, 13);break; // Red 2
      case 3:mySendRC5Laser(0x3UL, 13);break; // Red 3
      case 4:mySendRC5Laser(0x51UL, 13);break; // Green 1
      case 5:mySendRC5Laser(0x56UL, 13);break; // Green 6
      }
    delay(200); //Needed to successfully re-enable IRIn
    irrecv.enableIRIn(); //irsend automatically disables irrecv
    delay(2000); //Wait for incoming remote signal
}

void  mySendRC5Laser (unsigned long data,  int nbits)
{
    // Set IR carrier frequency
        irsend.enableIROut(36); //36 kHz

    // Data
    for (unsigned long  mask = 1UL << (nbits - 1);  mask;  mask >>= 1) {
        if (data & mask) {
            irsend.space(RC5_T1); // 1 is space, then mark
            irsend.mark(RC5_T1);
        } else {
            irsend.mark(RC5_T1);
            irsend.space(RC5_T1);
        }
    }

    irsend.space(0);  // Always end with the LED off

    Serial.println("Mode " + String(mode) + " - SEND OK");
}

void checkIR(){

  if (irrecv.decode(&results)) {

  switch(results.value)
   {
   case 0xFF30CF:mode=1;Serial.println("New Mode: " + String(mode));break; //Button 1 on remote
   case 0xFF18E7:mode=2;Serial.println("New Mode: " + String(mode));break; //etc.
   case 0xFF7A85:mode=3;Serial.println("New Mode: " + String(mode));break;
   case 0xFF10EF:mode=4;Serial.println("New Mode: " + String(mode));break;
   case 0xFF38C7:mode=5;Serial.println("New Mode: " + String(mode));break;
   default:Serial.print("Unknown Signal: ");Serial.println(results.value, HEX);break;
   }
   
  }
}

**I will update and close the threat when I can tell you how to decode the original units signals back to hex!**

laserir, 2017-10-31:
Alright... **second part** of the solution: **Decoding the signal**
As you can see above the protocol does not provide very strict timing. This was the main problem. After manipulating a lot of values and trying various combinations I found the following working 90% of the time: (irremoteint.h)
`#define MARK_EXCESS    0 //instead we increase the tolerance`
`#define TOLERANCE       50 //we doubled the tolerance here`
and of course in ir_rc5_rc6.ccp (thanks again to AnalysIR for giving me this hint!):
`#define RC5_T1             333 //value found by AnalysIR software` 

The signals differ in length, so we took out the strict length check: (ir_rc5_rc6.ccp)
`	//if (irparams.rawlen < MIN_RC5_SAMPLES + 2)  return false ;`

Also the signal does not use a header, therefore we must comment out the following lines in decodeRC5: (ir_rc5_rc6.ccp)
```
// Get start bits
	//if (getRClevel(results, &offset, &used, RC5_T1) != MARK)   return false ;
	//if (getRClevel(results, &offset, &used, RC5_T1) != SPACE)  return false ;
	//if (getRClevel(results, &offset, &used, RC5_T1) != MARK)   return false ;
```

As we only want to decode these type of signals it is better to disable all other types of supported protocols by setting the values to 0 in irremote.h.

closed 2017-10-31 by laserir

Aakash3, 2019-02-04:
> Alright... **second part** of the solution: **Decoding the signal**
> As you can see above the protocol does not provide very strict timing. This was the main problem. After manipulating a lot of [...]

I TRIED 3 DAYS, FOR MY IR REMOTE PROJECT.