Arduino-IRremote/Arduino-IRremote #535: Commands aren't being recognized

SelaO, 2017-10-11:
**Board:** ARDUINO UNO
**Library Version:** 2.4.0
**Protocol:** NEC

I'm trying to control my AC, but even when I bring the led very close to the AC's IR receiver, it still isn't being recognized. It happens with both `sendNEC` and `sendRaw`.

I tried to debug with a second Arduino, these are the readings from the AC's remote and the led transmitting the same command. 

```Readings from the remote:
Encoding : NEC
Code : 9EC0030A (32 bits)
Timing[73]:
+8850, -4350 + 750, -1550 + 700, - 450 + 700, - 500
+ 650, -1600 + 650, -1600 + 700, -1600 + 700, -1550
+ 700, - 450 + 650, -1650 + 650, -1600 + 650, - 500
+ 650, - 550 + 650, - 550 + 650, - 500 + 650, - 550
+ 650, - 550 + 650, - 500 + 650, - 550 + 650, - 550
+ 650, - 500 + 650, - 550 + 650, - 550 + 650, -1600
+ 650, -1600 + 650, - 550 + 600, - 550 + 650, - 550
+ 650, - 550 + 600, -1650 + 650, - 500 + 650, -1650
+ 650, - 500 + 650, - 550 + 650, -1600 + 650, - 550
+ 650
unsigned int rawData[73] = {8850,4350, 750,1550, 700,450, 700,500, 650,1600, 650,1600, 700,1600, 700,1550, 700,450, 650,1650, 650,1600, 650,500, 650,550, 650,550, 650,500, 650,550, 650,550, 650,500, 650,550, 650,550, 650,500, 650,550, 650,550, 650,1600, 650,1600, 650,550, 600,550, 650,550, 650,550, 600,1650, 650,500, 650,1650, 650,500, 650,550, 650,1600, 650,550, 650}; // NEC 9EC0030A
unsigned int data = 0x9EC0030A;

Encoding  : NEC
Code      : 9EC0030A (32 bits)
Timing[73]: 
     +8850, -4400     + 700, -1550     + 700, - 500     + 650, - 500
     + 650, -1600     + 650, -1650     + 700, -1550     + 700, -1550
     + 700, - 500     + 650, -1600     + 650, -1600     + 650, - 550
     + 650, - 500     + 650, - 550     + 650, - 550     + 600, - 550
     + 650, - 550     + 650, - 500     + 650, - 550     + 650, - 550
     + 650, - 500     + 650, - 550     + 650, - 550     + 650, -1600
     + 650, -1600     + 650, - 550     + 650, - 500     + 650, - 550
     + 650, - 550     + 600, -1650     + 650, - 500     + 650, -1650
     + 650, - 500     + 650, - 550     + 650, -1600     + 650, - 550
     + 650
unsigned int  rawData[73] = {8850,4400, 700,1550, 700,500, 650,500, 650,1600, 650,1650, 700,1550, 700,1550, 700,500, 650,1600, 650,1600, 650,550, 650,500, 650,550, 650,550, 600,550, 650,550, 650,500, 650,550, 650,550, 650,500, 650,550, 650,550, 650,1600, 650,1600, 650,550, 650,500, 650,550, 650,550, 600,1650, 650,500, 650,1650, 650,500, 650,550, 650,1600, 650,550, 650};  // NEC 9EC0030A
unsigned int  data = 0x9EC0030A;
```

```Readings from the led:
Encoding  : NEC
Code      : 9EC0030A (32 bits)
Timing[73]: 
     +8700, -4350     + 700, -1600     + 650, - 500     + 650, - 500
     + 700, -1600     + 650, -1550     + 700, -1650     + 600, -1600
     + 650, - 500     + 600, -1650     + 650, -1600     + 600, - 550
     + 600, - 600     + 600, - 600     + 600, - 550     + 600, - 600
     + 600, - 550     + 650, - 550     + 650, - 550     + 600, - 600
     + 600, - 500     + 650, - 550     + 650, - 550     + 650, -1600
     + 650, -1600     + 650, - 550     + 600, - 550     + 650, - 550
     + 650, - 550     + 550, -1700     + 600, - 550     + 600, -1700
     + 600, - 550     + 650, - 500     + 650, -1600     + 650, - 550
     + 650
unsigned int  rawData[73] = {8700,4350, 700,1600, 650,500, 650,500, 700,1600, 650,1550, 700,1650, 600,1600, 650,500, 600,1650, 650,1600, 600,550, 600,600, 600,600, 600,550, 600,600, 600,550, 650,550, 650,550, 600,600, 600,500, 650,550, 650,550, 650,1600, 650,1600, 650,550, 600,550, 650,550, 650,550, 550,1700, 600,550, 600,1700, 600,550, 650,500, 650,1600, 650,550, 650};  // NEC 9EC0030A
unsigned int  data = 0x9EC0030A;

Encoding  : NEC
Code      : 9EC0030A (32 bits)
Timing[73]: 
     +8800, -4300     + 800, -1500     + 750, - 400     + 750, - 400
     + 750, -1500     + 700, -1550     + 750, -1550     + 750, -1450
     + 750, - 400     + 700, -1600     + 700, -1550     + 650, - 500
     + 650, - 550     + 700, - 500     + 700, - 450     + 700, - 500
     + 700, - 500     + 700, - 450     + 700, - 450     + 750, - 500
     + 700, - 450     + 700, - 500     + 700, - 500     + 700, -1500
     + 700, -1550     + 700, - 500     + 650, - 500     + 700, - 500
     + 700, - 500     + 650, -1600     + 650, - 500     + 700, -1550
     + 750, - 400     + 750, - 500     + 700, -1500     + 700, - 500
     + 700
unsigned int  rawData[73] = {8800,4300, 800,1500, 750,400, 750,400, 750,1500, 700,1550, 750,1550, 750,1450, 750,400, 700,1600, 700,1550, 650,500, 650,550, 700,500, 700,450, 700,500, 700,500, 700,450, 700,450, 750,500, 700,450, 700,500, 700,500, 700,1500, 700,1550, 700,500, 650,500, 700,500, 700,500, 650,1600, 650,500, 700,1550, 750,400, 750,500, 700,1500, 700,500, 700};  // NEC 9EC0030A
unsigned int  data = 0x9EC0030A;

Encoding  : NEC
Code      : 9EC0030A (32 bits)
Timing[73]: 
     +8800, -4300     + 800, -1500     + 750, - 400     + 750, - 400
     + 750, -1500     + 700, -1550     + 750, -1550     + 750, -1450
     + 750, - 400     + 700, -1600     + 700, -1550     + 700, - 450
     + 700, - 500     + 700, - 500     + 650, - 500     + 700, - 500
     + 650, - 550     + 700, - 450     + 700, - 500     + 650, - 550
     + 650, - 500     + 650, - 550     + 700, - 500     + 700, -1500
     + 700, -1550     + 700, - 500     + 650, - 500     + 700, - 550
     + 650, - 500     + 650, -1600     + 700, - 450     + 650, -1650
     + 650, - 500     + 650, - 550     + 650, -1600     + 700, - 450
     + 700
unsigned int  rawData[73] = {8800,4300, 800,1500, 750,400, 750,400, 750,1500, 700,1550, 750,1550, 750,1450, 750,400, 700,1600, 700,1550, 700,450, 700,500, 700,500, 650,500, 700,500, 650,550, 700,450, 700,500, 650,550, 650,500, 650,550, 700,500, 700,1500, 700,1550, 700,500, 650,500, 700,550, 650,500, 650,1600, 700,450, 650,1650, 650,500, 650,550, 650,1600, 700,450, 700};  // NEC 9EC0030A
unsigned int  data = 0x9EC0030A;
```

They look pretty much the same. What else can I do to debug it? 
Could it be not working in 38Khz?
Could it still be a problem with the led being too weak even though the led was touching the AC's sensor?

**checklist:**
- [x] I have **read** the README.md file thoroughly
- [x] I have searched existing issues to see if there is anything I have missed.
- [x] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [x] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [x] The title of the issue is helpful and relevant 

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

The above is a short template allowing you to make detailed issues!

AnalysIR, 2017-10-11:
Forget sendNEC...they are not NEC signals.

I tried the first signal with AnalsyIR & it decodes to a GREE 32 bit AC protocol.

So using sendRAW, should be fine.

It could be your IR emitter circuit is wrong....but there is no info about that.

If all else fails, search the issues here (or our blog) for our script about recording long AC signals, which should improve the timings for you.

My notes say 38kHz is fine for this protocol.

Did you test your setup with your TV, to make sure everything is working OK?

SelaO, 2017-10-11:
It works with my TV, but my fan now doesn't work although it used to, that's strange. 

I found this: https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/

These are the readings I get with it from on/off:

```
Raw: (139) 8964, -4412, 660, -1624, 656, -536, 656, -540, 656, -540, 648, -1632, 660, -1624, 660, -532, 660, -536, 688, -1592, 688, -1596, 652, -540, 684, -512, 680, -516, 688, -504, 688, -508, 684, -508, 684, -512, 680, -512, 684, -512, 688, -508, 688, -504, 688, -1596, 684, -1596, 684, -512, 680, -512, 692, -504, 688, -504, 688, -508, 684, -1596, 688, -508, 684, -1600, 680, -512, 680, -516, 688, -1592, 688, -508, 684, -19800, 688, -508, 684, -512, 680, -512, 680, -516, 688, -504, 688, -508, 684, -508, 684, -512, 680, -512, 680, -516, 676, -516, 688, -508, 684, -512, 680, -1600, 684, -512, 680, -512, 688, -508, 684, -508, 688, -508, 684, -508, 684, -512, 680, -516, 688, -504, 688, -508, 684, -508, 684, -512, 680, -516, 688, -504, 688, -508, 684, -508, 684, -512, 680, -512, 680, 

Raw: (139) 8968, -4408, 652, -1628, 664, -532, 660, -536, 656, -1624, 656, -1628, 652, -1628, 652, -544, 680, -512, 692, -1592, 656, -1624, 688, -508, 688, -508, 684, -508, 684, -512, 692, -500, 688, -508, 688, -504, 688, -508, 684, -512, 680, -512, 688, -508, 688, -1592, 688, -1596, 684, -508, 684, -512, 680, -512, 692, -504, 688, -508, 684, -1596, 684, -512, 680, -1600, 684, -512, 680, -512, 688, -1596, 688, -508, 684, -19800, 684, -512, 680, -512, 680, -516, 676, -516, 688, -508, 684, -508, 684, -512, 680, -512, 684, -512, 680, -512, 680, -516, 684, -508, 684, -512, 684, -1600, 680, -512, 688, -508, 688, -504, 688, -508, 684, -512, 680, -512, 680, -516, 688, -504, 688, -508, 684, -508, 684, -512, 680, -512, 692, -504, 688, -508, 684, -508, 684, -512, 680, -512, 680, -1604, 688, 

Raw: (139) 8964, -4412, 656, -1628, 656, -536, 656, -540, 652, -540, 660, -1624, 660, -1624, 656, -536, 688, -508, 684, -1596, 684, -1600, 680, -512, 692, -504, 688, -504, 688, -508, 684, -512, 680, -512, 680, -516, 688, -504, 688, -508, 684, -508, 684, -512, 680, -1600, 692, -1592, 688, -508, 684, -508, 684, -512, 680, -512, 684, -512, 680, -1600, 688, -508, 684, -1596, 688, -508, 684, -512, 680, -1600, 680, -516, 676, -19808, 688, -508, 684, -508, 684, -512, 680, -516, 676, -516, 676, -516, 688, -508, 684, -512, 680, -512, 680, -516, 676, -516, 680, -516, 684, -508, 684, -1600, 684, -512, 680, -512, 688, -508, 684, -508, 684, -512, 680, -512, 684, -512, 680, -512, 688, -508, 684, -508, 688, -508, 680, -516, 676, -516, 688, -508, 684, -508, 684, -512, 680, -516, 676, -516, 688, 

Raw: (139) 8964, -4412, 660, -1620, 660, -536, 656, -540, 652, -1628, 652, -1632, 648, -1632, 660, -536, 688, -504, 688, -1596, 684, -1596, 688, -508, 684, -512, 680, -512, 692, -504, 688, -504, 688, -508, 684, -508, 684, -512, 692, -500, 692, -504, 688, -508, 684, -1596, 684, -1600, 680, -512, 680, -516, 688, -504, 688, -508, 684, -512, 680, -1600, 684, -512, 676, -1604, 688, -508, 684, -508, 688, -1596, 684, -508, 684, -19804, 680, -512, 680, -516, 688, -508, 684, -508, 684, -512, 680, -512, 680, -516, 676, -516, 680, -516, 684, -508, 684, -512, 680, -512, 680, -516, 688, -1596, 688, -504, 684, -512, 684, -508, 684, -512, 680, -512, 688, -508, 684, -512, 684, -508, 684, -512, 680, -512, 688, -508, 684, -508, 684, -512, 684, -508, 684, -512, 680, -516, 684, -508, 684, -1600, 684, 
```

I tried one of those in sendRaw but nothing happened. 

Do I need to turn those negative values to positive for `sendRaw`?

AnalysIR, 2017-10-11:
Great!

closed 2017-10-11 by SelaO