Arduino-IRremote/Arduino-IRremote #536: Automatic generation of code for protocols

bengtmartensson, 2017-10-23:
My project [IrpTransmogrifier](https://github.com/bengtmartensson/IrpTransmogrifier) can now generate IRremote code for most of the IR protocols it knows. Presently only sending (not decoding) is implemented. There is a reasonably detailed description [contained therein](https://github.com/bengtmartensson/IrpTransmogrifier/blob/master/src/main/doc/IRremote-protocols.md).

To use, the [development version](https://github.com/bengtmartensson/IrpTransmogrifier/releases/tag/ci-build) needs to be installed.

There are a some problems and "quirks". (For example: The protocol parametrization is in general completely different. "My" protocols end with a space of protocol specific length > 0.)  Most of these (IMHO) comes from design flaws in IRremote. I have tried here not to fix these problems. 

I do not want to scare anyone away, but some software development experience outside of Arduino is highly desirable...

How can this be "integrated"? Should it at all? Possibly generate a few protocols "on demand"? Feedback and suggestions of all sort are welcome! (Right now, I do not want to write a PR.)

z3t0, 2018-01-03:
I think this is a really interesting idea and should be integrated into this project, at least on some level.

bengtmartensson, 2020-10-20:
Closing this, since it does not make sense without more work. Maybe I will take up the idea later.

It someone cares, [this project](https://github.com/bengtmartensson/IrProtocolCodeGeneration) contains some unfinished code.

closed 2020-10-20 by bengtmartensson