Arduino-IRremote/Arduino-IRremote #538: I want KENWOOD AC remote code

Najab, 2017-11-22:
If any one have KENWOOD AC remote code for ON/OFF and Change TEMPERATURES in the form of RAW Data please provide for me.
Thanks in advance.

AnalysIR, 2017-11-22:
I am not sure you will get that info readily, with the information provided.

It might help to specify which model# of AC unit (not just the make) and also the model# of remote control.

FYI: You can get AC remotes online from China, for a few $ delivered. Then you could use an Arduino to capture the RAW signals. (Also available via Amazon for a couple $ more)

Najab, 2017-11-22:
I already have AC remote but i can't read accurate RAW signals for KENWOOD AC.

AnalysIR, 2017-11-22:
OK...in that case a search of the issues here would have provided a solution....

Try using the following sketch, to record the RAW timings:
[Air Conditioners: Recording long Infrared Remote control signals with Arduino](https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/)

Najab, 2017-11-23:
I already try this

closed 2020-08-17 by ArminJo