Arduino-IRremote/Arduino-IRremote #539: Power code for Optoma HD600X-LV

vanlooverenkoen, 2017-11-26:
**Board:** ARDUINO UNO
**Library Version:** 2.1.0
**Protocol:** //no idea

**Code Block:**
```
#include <IRremote.h>
IRsend irsend;

void setup()
{
  Serial.begin(9600);
}

void loop() {
  if (Serial.read() != -1) {
    for (int i = 0; i < 3; i++) {
     //I dont know what to send to the Optoma HD600X-LV

      delay(100);
    }
  }
}

```

I don't know how I could intercept the IR codes because nothing is shown if i use the receiving code sample: https://github.com/z3t0/Arduino-IRremote/wiki/Receiving-with-the-IRremote-library
The remote is a IR2804 from Optoma

AnalysIR, 2017-11-26:
It would be a good idea if you worked thru, the examples provided with the library. First try playing with your TV & remote.

Then move on to the Optoma.

the irecvdump examples are probably of most use to you.

vanlooverenkoen, 2017-11-26:
We only have a Optima Beamer

AnalysIR, 2017-11-26:
OK, then use that with the examples in the library....**under the examples folder**

closed 2020-10-17 by ArminJo