Arduino-IRremote/Arduino-IRremote #544: Sony home theatre problem

FETup, 2017-12-18:
**Board:** ARDUINO UNO
**Library Version:** 2.1.0
**Protocol:Sony

I'am trying to build a basic two button remote to turn on/off my tv & home theatre (both Sony).I successfully implemented the tv part but i'am facing problems for the home theatre. Using IRrecvDump i captured the signal for the home theatre using TSOP1738 receiver.The received code is 540a,but after sending this the home theatre is turning on/off.Any idea what's the problem?

Code:

#include <IRremote.h>
int but = 7; //Button pin 1           
int val=0;
int val1=0;
int led=13;  //Led for indication
int but1=6; //Button pin  2                  
IRsend irsend;
void setup()
{
  Serial.begin(9600);
  pinMode(but,INPUT);
  pinMode(but1,INPUT);
  
  val=digitalRead(but);
  val1=digitalRead(but1);
  //Serial.print(val);
}

void loop()
{
  if(digitalRead(but)!= val)
  {
    digitalWrite(led,HIGH);
    for(int i;i < 3;i++)
    {
      irsend.sendSony(0xa90,12);   //Turn on/off tv
      delay(40);
      Serial.println("Done");
    }
  }
  else
  {
    digitalWrite(led,LOW); 
  }
  if(digitalRead(but1)!= val1)
  {
    digitalWrite(led,HIGH);
    for(int i;i < 3;i++)
    {
      irsend.sendSony(0x540a,12);   //Turn on/off blu-ray home theatre
      delay(40);
      Serial.println("Done");
    }
  }
  else
  {
    digitalWrite(led,LOW); 
  }
}

AnalysIR, 2017-12-18:
First edit your previous post & use code tags  <>, rom the menu.

then try 
`irsend.sendSony(0x540a,15); `

or

`irsend.sendSony(0x540a,20); `

0x540a does not fit into 12 bits!

FETup, 2017-12-18:
It worked with 15.

Thanks

AnalysIR, 2017-12-18:
Great, thanks for the update...

However, you still have not tidied up your first post, by placing the code inside code blocks... <>

Also, please mark the issue as closed....tnx

closed 2020-10-17 by ArminJo