Arduino-IRremote/Arduino-IRremote #546: IR decode for Elgin A/C

hbadotti, 2017-12-27:
I need to decode of this A/C, sending raw commands is consuming a lot of my global variables, I extracted some raw decodes, take a look into their size.

```c
unsigned int  temp26[197] = {6050,7300, 550,1650, 550,1600, 550,1600, 550,1650, 500,1600, 550,1650, 500,1650, 550,1600, 500,600, 500,600, 550,550, 550,550, 550,550, 500,550, 550,550, 550,550, 550,1600, 550,1650, 550,1600, 550,1600, 500,1650, 550,1600, 550,1600, 550,1600, 550,550, 550,550, 550,550, 550,500, 550,600, 500,550, 550,550, 500,600, 550,1600, 550,550, 500,1700, 500,1600, 550,1650, 550,1600, 500,1650, 500,1650, 550,550, 550,1650, 550,550, 550,550, 550,550, 550,550, 500,600, 500,550, 550,1650, 500,600, 550,500, 600,1550, 550,1600, 550,1650, 500,1600, 550,1650, 500,600, 500,1650, 550,1600, 550,550, 550,550, 550,550, 550,550, 500,600, 500,1650, 550,550, 550,1600, 550,500, 550,1650, 500,550, 550,1650, 500,1650, 550,550, 550,1600, 550,550, 550,1600, 550,550, 550,1600, 550,600, 500,550, 550,550, 550,1600, 550,550, 550,1600, 550,550, 550,1600, 550,550, 550,550, 550,1600, 550,600, 550,1600, 500,550, 550,1600, 550,550, 550,1600, 500,1650, 550,7300, 500};  // UNKNOWN 4CB9DC41
unsigned int  temp25[197] = {6050,7350, 500,1650, 550,1650, 500,1600, 550,1650, 500,1650, 550,1600, 500,1600, 550,1650, 500,600, 550,550, 500,550, 550,550, 550,550, 500,600, 550,500, 550,550, 550,1650, 550,1600, 550,1550, 550,1650, 500,1650, 500,1650, 550,1600, 550,1600, 500,600, 550,550, 550,500, 550,600, 500,550, 550,550, 500,600, 550,500, 550,1650, 500,600, 550,1600, 550,1600, 550,1650, 550,1550, 550,1650, 500,1650, 550,600, 500,1650, 550,500, 550,550, 550,600, 500,600, 500,550, 550,550, 550,1650, 500,550, 550,550, 550,1650, 500,1600, 550,1650, 550,1600, 550,1650, 500,600, 550,1600, 550,1600, 550,550, 550,600, 500,550, 550,550, 500,600, 550,550, 550,1650, 500,1650, 550,550, 500,1650, 550,600, 500,1650, 500,1650, 500,1650, 500,600, 550,500, 550,1600, 550,550, 550,1650, 550,550, 500,550, 550,550, 550,1650, 500,550, 550,1650, 500,600, 500,1650, 550,550, 550,550, 500,1650, 550,550, 550,1600, 500,550, 550,1650, 550,550, 550,1600, 550,1600, 550,7300, 550};  // UNKNOWN 520818E1
unsigned int  temp24[197] = {6050,7350, 550,1600, 550,1650, 500,1650, 500,1650, 550,1600, 550,1600, 550,1600, 550,1600, 550,550, 550,550, 550,550, 550,550, 500,600, 500,550, 550,550, 550,550, 550,1650, 550,1550, 600,1600, 550,1600, 550,1650, 500,1650, 500,1600, 550,1650, 550,600, 500,550, 500,600, 550,550, 500,550, 550,550, 550,550, 500,550, 550,1700, 500,550, 500,1650, 500,1650, 500,1650, 550,1600, 550,1600, 550,1600, 550,600, 550,1600, 500,600, 500,550, 550,550, 550,550, 500,600, 500,600, 500,1700, 500,550, 550,550, 500,1650, 550,1600, 550,1650, 500,1650, 550,1600, 500,650, 500,1600, 550,1600, 550,550, 500,600, 550,500, 550,600, 500,550, 500,1650, 550,1600, 550,1600, 550,600, 500,1600, 550,550, 550,1650, 500,1600, 600,550, 550,550, 500,600, 500,1650, 550,550, 500,1650, 500,550, 550,600, 500,650, 500,1600, 600,600, 500,1600, 550,500, 600,1600, 500,600, 500,550, 550,1650, 500,600, 500,1650, 500,550, 550,1600, 550,550, 550,1600, 550,1600, 550,7300, 550};  // UNKNOWN D3D1F385
unsigned int  temp23[197] = {6100,7300, 550,1650, 550,1600, 500,1600, 550,1600, 550,1600, 550,1600, 550,1650, 500,1650, 550,550, 550,550, 550,550, 500,600, 500,600, 500,550, 550,550, 550,550, 550,1650, 500,1600, 550,1650, 550,1600, 550,1650, 500,1600, 550,1600, 550,1650, 500,600, 550,550, 550,500, 550,600, 500,550, 550,550, 500,600, 550,500, 550,600, 550,500, 550,1600, 550,1600, 550,1650, 500,1650, 550,1600, 550,1600, 550,1650, 500,1650, 550,500, 550,550, 550,550, 550,600, 500,550, 550,550, 500,1650, 550,550, 550,1600, 550,550, 500,1650, 500,1650, 500,1650, 550,1600, 550,550, 550,1650, 550,550, 500,1650, 500,550, 550,550, 550,550, 550,550, 500,600, 550,550, 500,600, 500,1600, 550,1650, 550,550, 550,1550, 550,1650, 500,1650, 550,1600, 550,1600, 550,550, 600,500, 550,1650, 500,550, 550,550, 550,550, 550,1600, 550,550, 550,1600, 550,550, 550,1600, 500,600, 550,550, 500,1650, 550,550, 550,1600, 500,600, 550,1600, 500,600, 500,1600, 550,1600, 550,7300, 550};  // UNKNOWN C5EC5641
unsigned int  temp22[197] = {6050,7350, 550,1600, 500,1700, 500,1650, 500,1600, 550,1650, 500,1650, 500,1600, 500,1700, 500,550, 550,550, 550,550, 600,500, 550,500, 550,600, 550,550, 500,550, 500,1700, 500,1650, 500,1650, 550,1600, 550,1600, 550,1650, 550,1600, 500,1650, 550,550, 550,550, 500,600, 500,600, 500,550, 550,550, 550,550, 500,600, 550,1650, 500,600, 550,1600, 550,1600, 500,1700, 500,1600, 550,1600, 500,1700, 500,600, 500,1700, 550,550, 500,550, 500,600, 550,550, 500,550, 500,650, 500,1650, 500,550, 550,600, 500,1600, 550,1600, 550,1650, 550,1600, 500,1650, 550,550, 550,1650, 500,1600, 550,550, 550,600, 500,550, 550,550, 550,550, 550,1600, 550,600, 500,550, 550,1600, 500,1650, 550,600, 500,1600, 550,1650, 500,550, 500,1650, 600,1600, 550,500, 550,600, 500,1600, 550,550, 500,600, 550,550, 550,1650, 550,550, 500,1650, 550,550, 550,1550, 550,550, 550,550, 550,1600, 550,600, 500,1650, 550,500, 550,1700, 500,550, 550,1650, 550,1550, 550,7350, 550};  // UNKNOWN 2355C6E9
unsigned int  temp21[197] = {6050,7300, 550,1650, 500,1600, 550,1650, 500,1600, 550,1650, 500,1650, 550,1600, 500,1650, 550,600, 500,550, 550,550, 550,550, 550,550, 500,600, 500,550, 550,550, 500,1650, 550,1600, 550,1600, 550,1600, 550,1650, 500,1650, 550,1600, 550,1650, 550,550, 550,550, 500,600, 500,550, 550,550, 550,550, 550,550, 500,550, 550,1650, 550,550, 500,1600, 550,1600, 600,1600, 500,1650, 550,1600, 600,1550, 550,600, 500,1600, 550,600, 500,600, 550,500, 550,550, 550,550, 550,600, 500,1650, 550,500, 550,600, 550,1600, 550,1600, 500,1650, 550,1600, 550,1600, 550,550, 550,1650, 500,1650, 550,550, 500,600, 500,550, 550,550, 550,550, 550,600, 500,1650, 550,550, 550,1600, 500,1650, 550,550, 500,1600, 550,1650, 500,1650, 550,500, 550,1650, 500,550, 600,500, 550,1600, 550,550, 500,550, 600,550, 550,1600, 550,550, 550,1600, 550,550, 500,1650, 550,600, 500,600, 500,1650, 550,550, 550,1600, 500,550, 550,1650, 550,550, 500,1650, 550,1600, 550,7300, 550};  // UNKNOWN 34624241
unsigned int  temp20[197] = {6050,7300, 550,1650, 550,1600, 500,1650, 500,1650, 500,1650, 550,1600, 500,1650, 550,1600, 550,600, 550,550, 500,550, 550,550, 550,550, 550,500, 550,600, 500,550, 550,1600, 550,1650, 550,1600, 550,1650, 500,1600, 550,1650, 550,1600, 550,1600, 500,600, 550,550, 550,550, 500,600, 500,550, 550,600, 500,550, 550,550, 550,1650, 550,500, 600,1600, 500,1650, 550,1600, 550,1650, 550,1600, 500,1700, 500,600, 500,1650, 550,550, 550,550, 550,550, 550,550, 550,500, 550,600, 500,1650, 500,550, 550,550, 550,1600, 550,1600, 550,1650, 550,1600, 500,1650, 550,550, 550,1650, 550,1600, 600,550, 500,550, 550,550, 500,600, 550,550, 550,1650, 500,1650, 550,600, 500,1600, 550,1600, 550,600, 500,1600, 600,1550, 550,600, 550,500, 550,1600, 550,550, 550,550, 550,1650, 500,550, 550,550, 550,550, 600,1600, 500,600, 550,1600, 550,600, 500,1600, 550,550, 550,550, 500,1650, 550,600, 500,1650, 500,550, 550,1650, 500,550, 550,1600, 550,1600, 550,7300, 550};  // UNKNOWN B62C1CE5
unsigned int  temp19[197] = {6100,7300, 550,1650, 500,1600, 550,1600, 550,1650, 500,1650, 500,1650, 550,1600, 550,1600, 550,600, 550,550, 500,600, 550,500, 550,550, 500,600, 550,550, 550,500, 550,1650, 550,1600, 550,1600, 500,1650, 550,1600, 550,1600, 550,1650, 500,1600, 550,600, 550,550, 500,550, 550,550, 550,550, 550,550, 500,600, 500,600, 500,1650, 550,600, 500,1650, 550,1600, 550,1600, 550,1650, 500,1600, 550,1600, 550,600, 550,1600, 500,600, 500,550, 550,550, 600,500, 550,550, 550,550, 500,1650, 550,550, 550,600, 500,1600, 550,1650, 550,1600, 550,1600, 550,1600, 550,600, 500,1600, 600,1600, 550,550, 500,600, 550,500, 550,600, 550,550, 550,550, 500,600, 550,1600, 550,1600, 550,1600, 550,550, 500,1600, 550,1650, 500,1650, 500,1700, 500,550, 550,500, 550,550, 550,1600, 550,550, 550,550, 550,600, 500,1600, 600,500, 600,1600, 500,600, 550,1600, 500,600, 550,550, 550,1600, 550,550, 550,1600, 550,550, 550,1600, 550,550, 550,1650, 500,1600, 550,7350, 500};  // UNKNOWN CF31CAE1
unsigned int  temp18[197] = {6050,7350, 550,1650, 500,1650, 550,1600, 500,1650, 550,1600, 550,1600, 550,1600, 550,1650, 500,600, 550,550, 550,500, 550,550, 550,550, 550,550, 500,600, 550,500, 550,1650, 550,1600, 550,1650, 500,1650, 550,1600, 500,1650, 550,1600, 550,1600, 550,600, 550,500, 550,550, 500,600, 550,500, 600,550, 550,550, 500,550, 550,1650, 550,550, 500,1650, 550,1600, 550,1600, 550,1650, 500,1650, 550,1600, 500,60 [... cut ...]
**Board:** ARDUINO MEGA 2560
**Library Version:** 2.1.0
**Protocol:** None

**checklist:**
- [] I have **read** the README.md file thoroughly - OK
- [] I have searched existing issues to see if there is anything I have missed. - OK
- [] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used - OK
- [] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here - OK
- [] The title of the issue is helpful and relevant - OK

AnalysIR, 2017-12-27:
Try the following:
[Sending long AC Signals from Flash with IRremote](https://www.analysir.com/blog/2016/04/11/sending-long-ac-signals-flash-irremote/)

hbadotti, 2017-12-27:
It worked perfecly!

AnalysIR, 2017-12-27:
Great!  :)

PS: Don't forget to close this issue.

hbadotti, 2017-12-27:
Done.

closed 2017-12-27 by hbadotti