Arduino-IRremote/Arduino-IRremote #548: Problem while Interfacing P10 Display and IR sensor

profmanojkavedida, 2018-01-01:
I m facing problem while interfacing P10 display and Tsop1738 with arduino
The code for TSOP1738 when monitor on serial Monitor works very fine 

These is working code 
#include <IRremote.h>

int RECV_PIN = 5;

IRrecv irrecv(RECV_PIN);

decode_results results;

void setup()
{
  Serial.begin(9600);
  irrecv.enableIRIn(); // Start the receiver
}

void loop() {
  if (irrecv.decode(&results)) {
    Serial.println(results.value, HEX);
    irrecv.resume(); // Receive the next value
  }
}

 But as soon as i add the P10 display routine code
#include <IRremote.h>
#include <TimerOne.h>
#include"SPI.h"
#include <ledP10.h>

LedP10 myled;
 
int RECV_PIN = 5;

IRrecv irrecv(RECV_PIN);

decode_results results;

void setup()
{
  Serial.begin(9600);
  irrecv.enableIRIn(); // Start the receiver
}

void loop() {
  if (irrecv.decode(&results)) {
    Serial.println(results.value, HEX);
myled.showmsg_single_static(results.value,0);
    irrecv.resume(); // Receive the next value
  }
}

Remote is sensed properly time and then it give random key codes for that same remote.
I have test code for types of remote
First time it give NEC type 
and next time it gives unknhown format
Please help me...urgently

bengtmartensson, 2018-01-01:
Possibly the LED's light disturbs the sensor.

Please format the code properly.

profmanojkavedida, 2018-01-01:
LED lyt is not intering becasue i have kept light  and sensor such a way that it wont interfer

profmanojkavedida, 2018-01-01:
LED light is not interfering because i have kept light and sensor such a way that it wont interfere

profmanojkavedida, 2018-01-01:
#include <IRremote.h>
#include <TimerOne.h>
#include"SPI.h"
#include <ledP10.h>

LedP10 myled;
 
int RECV_PIN = 5;

IRrecv irrecv(RECV_PIN);

decode_results results;

void setup()
{
  Serial.begin(9600);
  irrecv.enableIRIn(); // Start the receiver
}

void loop()
 {
  if (irrecv.decode(&results)) 
{
    Serial.println(results.value, HEX);
    myled.showmsg_single_static(results.value,0);
    irrecv.resume();                                                // Receive the next value
  }
}

This code is not giving output

closed 2020-07-04 by ArminJo