Arduino-IRremote/Arduino-IRremote #549: Receive on Tiny85 trouble, possible hang

muelli01, 2018-01-11:
Hi there,
I created a small program to test why my "big one" does not seem to receive anything or even makes my Tiny85 seem to hang.
(Note: On my mega328 everything works fine!)

Library versions tried: 2.4beta, 2.2.3, 2.1.0

The code is simple, I have a LED connected to pin 0, the TSOP is connected to pin 2.
I just want to identify the signals received via number of blinks on the LED.
When connected to power, the Tiny85 blinks 5 times, then waits 5s.
If I then press a key on a remote, the LED just turns on and that is about all that happens.

Code:
#include <IRremote.h>
const int RECV_PIN = 2;
const int LED = 0;
IRrecv irrecv(RECV_PIN);
decode_results results;

void setup()
{
  pinMode(LED, OUTPUT);
  digitalWrite(LED, LOW);
  //blink 5 times, then wait 5s
  blinken(5);
  delay(5000);

  irrecv.enableIRIn(); // Start the receiver
  irrecv.blink13(true);

}

void loop() {
  if (irrecv.decode(&results)) {
    if (results.decode_type == NEC) {
      blinken(1);
    } else if (results.decode_type == SONY) {
      blinken(2);
    } else if (results.decode_type == RC5) {
      blinken(3);
    } else if (results.decode_type == RC6) {
      blinken(4);
    } else if (results.decode_type == UNKNOWN) {
      blinken(5);
    }
    irrecv.resume(); // Receive the next value
  }
}

void blinken(int anzahl) {
  for (int counter=0;counter < anzahl; counter ++) {
    digitalWrite(LED,HIGH);
    delay(250);
    digitalWrite(LED,LOW);
    delay(250);
  }
}

Any help is appreciated!
Thanks

muelli01, 2018-01-11:
After setting the Tiny85 to internal 16Mhz at least the "big" project is working. This test stil fails....

bengtmartensson, 2018-01-12:
> I have a LED connected to pin 0

using a serial resistor? How much?

muelli01, 2018-01-12:
It is a 2mA LED, Vf=2.1V, driven at 3.3V -> 680Ohm resitor

bengtmartensson, 2018-01-12:
Ok, sorry for the silly question. 

I _think_ that the Tiny85 has only one timer, and that is (supposed to be) used both by IRremote _and_ `delay(...)`, which may have "interesting" consequences. So, try to replace the `delay(...)` in `blinken(...)` by a spin-wait loop, since you do not care too much (?) about the precision of the 250ms just

    for (uint32_t i = 0; i < xxxxx; i++)
        ;

for suitable xxxxx should do it.

muelli01, 2018-01-12:
Yes afaik the tiny85 has indeed only one timer! good point.
Will try this asap, thanks!

muelli01, 2018-02-22:
Sorry for the late reply, was just now able to test.
Replacing the delay with a loop does (surprisingly) not fix the problem.
As mentioned before, setting the fuses to internal 16mhz pll makes it run "most" of the time but I still experience some hangs from time to time.
I think it might be caused by the internal clock, it is under some conditions not _very_ accurate. I will try to redesign the project to use an external oscillator and will see if this fixes my problem

bengtmartensson, 2018-02-22:
I think that your problems are, strictly speaking, off-topic in the sense that it does not pertain to the software package that is the subject here. It is about how to get very simplistic hardware to work.

Having said that, if you are looking for things to try, try with a few different ("identical") boards, try fiddling with the power supply (buffering capacitators, possibly slightly higher supply voltage).

closed 2020-08-17 by ArminJo