Arduino-IRremote/Arduino-IRremote #550: Issue sending raw to my TCL tv

lkallryd, 2018-01-23:
**Board:** ARDUINO UNO
**Library Version:** 2.1.0
**Protocol:** raw (if any)

**Code Block:**
```
// LARS automation IR

//#include <DS3231.h>
#include <SPI.h>
#include <Wire.h>
#include "RTClib.h"
#include <IRremote.h>

IRsend irsend;
RTC_DS3231 rtc;

char daysOfTheWeek[7][12] = {"Sunday", "Monday", "Tuesday", "Wednesday", "Thursday", "Friday", "Saturday"};

void setup () {

  Serial.begin(9600);

  delay(30); // wait for console opening

  if (! rtc.begin()) {
    Serial.println("Couldn't find RTC");
    while (1);
  }
//    // following line sets the RTC to the date & time this sketch was compiled
//  rtc.adjust(DateTime(2018, 1, 4, 21, 30, 45));
//    // This line sets the RTC with an explicit date & time, for example to set
//    // January 21, 2014 at 3am you would call:
//  rtc.adjust(DateTime(2017, 12, 10, 18, 45, 00));
}

void loop () {
    DateTime now = rtc.now();

    if (daysOfTheWeek[now.dayOfTheWeek()] == "Monday" || daysOfTheWeek[now.dayOfTheWeek()] == "Tuesday" || daysOfTheWeek[now.dayOfTheWeek()] == "Wednesday"
    || daysOfTheWeek[now.dayOfTheWeek()] == "Thursday" || daysOfTheWeek[now.dayOfTheWeek()] == "Friday" && now.hour()==06 && now.minute()==00) {

      sendRawIR();
      
    }

}

void sendRawIR(void) {
  int khz = 38; // 38kHz carrier frequency for the remote protocol
  unsigned int irPwrOn[] = {3800,4000,400,2050,350,2100,350,2050,400,2050,400,1100,350,1100,350,2100,350,1100,400,2050,350,1100,400,2050,350,1100,400,1100,350,1100,400,1050,400,1100,350,2100,350,2050,400,1100,350,2050,400,1100,350,2100,350,1100,350,2100,350};

  unsigned int ch4[] = {3800,4000,400,2050,350,2100,350,2050,400,2050,400,1100,350,1100,350,2100,350,2050,400,1100,350,2100,350,1100,350,1100,400,1100,350,1100,400,1050,400,1100,350,2100,350,2050,400,1100,350,1100,400,2050,350,1100,400,2050,350,2100,350};
  
  irsend.sendRaw(irPwrOn, sizeof(irPwrOn) / sizeof(irPwrOn[0]), khz); //Note the approach used to automatically calculate the size of the array.

  delay(65*1000); //Wait 65 seconds for tv to boot up

    irsend.sendRaw(ch4, sizeof(ch4) / sizeof(ch4[0]), khz);

}

```

I have read out the power up signal with IRrecvDumpV2 from my TCL signal and it figures its a unknown protocol so I use the send raw but nothing is happening. 
I've tried multiple IR LEDs, I use a 270 Ohm in series with the LED.. Did I miss something?

Regards

AnalysIR, 2018-01-23:
First of I would suggest striiping out everything except 1 sendRaw (not other logic or libs) and get it working before introducing anything else.

I had a look at your signals and they both seem consistent.

The only other thing is to reduce the resistor to about 100R (or 3x270 in parallel)....assuming you are connecting directly to the arduino pins. Also make sure to bring the IR Led closer to the TV and point the round top of the LED directly towards the receiver on the TV.

What is the make & model# of the TV & remote (TCL???)

also try the following for the delay
`delay(65000UL);`
[reason](http://forum.arduino.cc/index.php?topic=46540.0)

lkallryd, 2018-01-23:
Thanks for awnser, I've tried that as well. I could try to remove the resistor completely, unsure if I've done that before or not. I've played around with different distances, everything between 10m to 0.5m with no result. The TV is called: U49C7006 - TCL 49"LED/UHD/PPI1600Hz/T2CS. Not sure of the remote.

If I get it working I would love to be able to update support for this brand too, if I get some guidance in how to do it, I've got most of the hardware I believe.

Once again, thanks.

AnalysIR, 2018-01-24:
Some Comments

- Dont remove the resistor (will damage the UNO)
- Use a red LED with a 1K+ resistor instead to see if it blinks when sending
- Alternatively, just check with a phone camera to see if you can see the IR being transmitted.
- What pin are you using for Tx on the UNO
- are you sure the LED is the right way round (shorter lead on teh GND side...longer lead on the UNO side)
- Post a photo of your setup and/or schematic

lkallryd, 2018-01-24:
Great comments:

- I've tried without resistor when I started troubleshooting with the hammer-style tactic but it didnt brake.. 
- When I connected a normal LED to pin #3 I didnt see it flashing or anything. But when using a normal "blink" sketch it turns on. 
- I've checked with phone-camera on the IR sender, it didnt flash.. I've also tried directing it towards the receiver and it does not pick up the signal. I've used two different types of IR LEDs.

- My final conclusion is that its not sending, I will now use another UNO and do another try. 
- Next go at it will be on friday, I'll upload a photo of my setup.

Regards

AnalysIR, 2018-01-24:
in case you dont have the default config for IRremote on the UNO...also try pin 9.

lkallryd, 2018-01-24:
Great tip, I'll try that!

closed 2020-10-17 by ArminJo