Arduino-IRremote/Arduino-IRremote #551: Trouble handling interrupts with IRRemote

lauraagouvea, 2018-01-30:
Hi there, 

I am working on a project to control some air conditioners remotely. I have a device that can send any kind of pulse remotely to Arduino. 
I have written the code bellow, but it doesnt work - at serial monitor it only shows the beggining of the word "sending" and the IR sender goes high all the time. 

when I remove the line "irsend.sendRaw(mais, 199, 38);" it works just fine.

(I am kind of new to Arduino and gitHub, sorry about poor coding).

any clue on what could be happening?

#include <IRremote.h>

IRsend irsend;

//RAW code for rising the temperature
volatile unsigned int rise[199]={4480, 4496, 552, 1660, 548, 560, 548, 1664, 548, 1660, 552, 556, 548, 556, 552, 1660, 548, 560, 548, 556, 548, 1664, 548, 556, 552, 556, 548, 1664, 548, 1664, 548, 556, 548, 1664, 548, 1664, 548, 556, 552, 1660, 552, 1660, 548, 1664, 548, 1664, 548, 1664, 548, 1664, 548, 560, 548, 1660, 548, 560, 548, 556, 548, 560, 548, 556, 548, 560, 548, 556, 548, 560, 548, 1660, 552, 556, 548, 556, 552, 556, 548, 556, 548, 560, 548, 556, 548, 1664, 548, 560, 548, 1660, 552, 1660, 552, 1660, 552, 1660, 552, 1660, 552, 1660, 548, 5348, 4508, 4496, 552, 1660, 552, 556, 548, 1664, 548, 1664, 548, 556, 548, 560, 548, 1660, 552, 556, 548, 560, 548, 1660, 552, 556, 548, 556, 548, 1664, 548, 1664, 548, 556, 552, 1660, 552, 1660, 552, 556, 548, 1664, 548, 1660, 552, 1660, 552, 1664, 548, 1660, 548, 1664, 552, 556, 548, 1664, 548, 556, 548, 556, 552, 556, 548, 556, 552, 556, 548, 556, 552, 556, 548, 1664, 548, 556, 548, 560, 548, 556, 548, 560, 548, 556, 548, 556, 552, 1660, 552, 556, 548, 1664, 548, 1664, 552, 1656, 552, 1660, 552, 1660, 552, 1660, 548}; 

const byte interruptPin = 2; 
volatile byte state = LOW;

void setup() {
  
  Serial.begin(9600);
  pinMode(interruptPin, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(interruptPin),riseAir,FALLING);
}

void loop() {
}

void riseAir(){
    Serial.println("Sending signal...");
    irsend.sendRaw(mais, 199, 38);
    Serial.println("Executing at 38 Hz");
}

AnalysIR, 2018-01-30:
There are a couple of issues.

1. Your code doesn't compile...so can't actually run (You have the wrong label for the rise/mais
2. You are doing a lot inside the Interrupt routine(or ISR) riseAir, which is probably your other main issue.

Better to only set a flag in the ISR and then call the riseAir function from within the main loop.

**IMPORTANT**: edit your last post and place the code inside "code tags"....<> on the edit menu

_FYI: Your signal decodes as a good AC signal on AnalysIR._

lauraagouvea, 2018-01-30:
@AnalysIR , what is the error message that appears for you? 
It does compile on my PC, but like I said it doesnt behave properly. 

I have tried the RAW code separately and it worked just fine.

AnalysIR, 2018-01-30:
> **IMPORTANT**: edit your last post and place the code inside "code tags"....<> on the edit menu

Why did you ignore this request?

do a diff between the code you posted and the the code you are compiling!

lauraagouvea, 2018-01-31:
I did that, but it doesnt save the changes. 
Anyway, adding a flag solved the problem. thanks!

z3t0, 2018-02-04:
@lauraagouvea I am glad to hear the issue has been resolved! Is there any chance you could either share the solution or even a dump of the fixed code so that others with a similar problem can benefit as well?

lauraagouvea, 2018-02-05:
Sure. Here's the code for 2 commands, turning on and off. 

```
#include <IRremote.h>

IRsend irsend;

//RAW 
unsigned int turn_on[199]={4412, 4344, 576, 1560, 576, 520, 600, 1536, 600, 1564, 600, 468, 600, 496, 572, 1540, 600, 520, 600, 468, 576, 1588, 600, 468, 600, 496, 600, 1536, 576, 1588, 572, 500, 572, 1588, 600, 468, 600, 1564, 600, 1536, 576, 1588, 576, 1588, 600, 468, 576, 1588, 572, 1564, 600, 1564, 600, 468, 600, 496, 600, 468, 600, 496, 572, 1544, 596, 520, 600, 468, 572, 1592, 600, 1536, 600, 1564, 600, 468, 600, 496, 572, 496, 600, 496, 572, 496, 600, 496, 600, 468, 572, 472, 648, 1536, 604, 1564, 600, 1536, 600, 1564, 600, 1536, 600, 5132, 4440, 4320, 600, 1536, 576, 520, 600, 1536, 600, 1564, 576, 492, 604, 492, 600, 1536, 604, 492, 576, 492, 600, 1564, 600, 468, 600, 496, 600, 1536, 576, 1588, 600, 468, 576, 1588, 576, 492, 628, 1536, 600, 1564, 572, 1564, 600, 1564, 600, 468, 576, 1588, 600, 1536, 604, 1560, 604, 492, 600, 468, 576, 492, 600, 496, 576, 1560, 604, 472, 596, 492, 600, 1564, 600, 1564, 600, 1536, 576, 520, 600, 468, 576, 492, 576, 516, 604, 468, 576, 516, 604, 468, 572, 524, 600, 1536, 572, 1588, 576, 1564, 600, 1564, 600, 1560, 604};
unsigned int turn_off[199]={4408, 4324, 596, 1564, 572, 524, 572, 1564, 572, 1592, 572, 496, 572, 524, 548, 1588, 572, 524, 572, 496, 572, 1592, 572, 496, 572, 524, 572, 1564, 572, 1592, 572, 496, 572, 1592, 572, 1564, 572, 524, 572, 1544, 592, 1592, 572, 1592, 572, 1564, 572, 1592, 572, 1564, 576, 520, 572, 1564, 572, 524, 572, 496, 572, 524, 572, 496, 572, 524, 572, 496, 572, 520, 576, 496, 572, 1588, 576, 496, 572, 520, 572, 500, 572, 496, 572, 520, 572, 1544, 596, 1588, 572, 500, 596, 1544, 596, 1588, 572, 1564, 576, 1568, 596, 1564, 572, 5160, 4412, 4344, 576, 1564, 572, 524, 572, 1564, 572, 1592, 572, 496, 572, 524, 572, 1564, 572, 524, 572, 496, 572, 1592, 568, 500, 572, 520, 576, 1564, 572, 1588, 576, 496, 572, 1588, 572, 1592, 576, 496, 572, 1588, 576, 1564, 572, 1588, 572, 1568, 572, 1592, 572, 1588, 576, 496, 572, 1592, 572, 496, 572, 520, 572, 500, 568, 500, 568, 524, 572, 496, 572, 524, 572, 496, 572, 1568, 596, 496, 548, 548, 572, 496, 572, 524, 572, 496, 572, 1592, 572, 1564, 548, 548, 568, 1572, 568, 1592, 548, 1588, 572, 1592, 572, 1592, 572};

//flags
volatile bool turnOn = false;
volatile bool turnOff = false;

const byte interruptPinOn = 3;
const byte interruptPinOff = 2;

void setup() {
  
  Serial.begin(9600);
  pinMode(interruptPinOn, INPUT_PULLUP);
  pinMode(interruptPinOff, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(interruptPinOn),sendOn,FALLING);
  attachInterrupt(digitalPinToInterrupt(interruptPinOff),sendOff,FALLING);
}

void loop() {

    if (turnOn) {
 
    Serial.println("Sending command to turn on..");
    sendOn();
    // clear the flag
    turnOn = false;
 }

    if (turnOff) {
 
    Serial.println("Sending command to turn off...");
    sendOff();
    // clear the flag
    turnOff = false;
 }

  
}

void turnOn(){
  
    turnOn = true;
  
}

void turnOff(){
  
    turnOff = true;
  
}

void ligaAr(){
    Serial.println("ligando...");
    irsend.sendRaw(liga, 199, 38);
    Serial.println("Executando a 38 Hz");
}

void desligaAr(){
    Serial.println("desligando...");
    irsend.sendRaw(desliga, 199, 38);
    Serial.println("Executando a 38 Hz");
}
```

closed 2020-07-04 by ArminJo