Arduino-IRremote/Arduino-IRremote #553: Replicating a unknown protocol RAW signal, for controlling EUFY ROBOVAC 11 vacuum cleaner

nakulbende, 2018-03-09:
Hello all,

I am pretty new to IRremote, and wanted some help with controlling my Eufy RoboVac 11 vacuum cleaner. The unit comes with a IR remote which can be used to program/ control the robot. I wanted to use a esp8266 mounted on unit, to make it 'smart'.

For reading the codes and testing this plan, I have a IR receiver (pin 3) and IR LED connected to an Arduino Mega. 

**Board:** Arduino MEGA
**Library Version:** 2.1.0
**Protocol:** Unknown

**Serial output from  IRrecvDump.ino:**
```c

Encoding  : UNKNOWN
Code      : E8BEB62D (32 bits)
Timing[83]:
     +2900, -3000     + 500, - 500     + 550, -1400     + 550, -1450
     + 500, - 450     + 550, -1450     + 500, - 450     + 550, - 450
     + 550, - 400     + 550, - 450     + 450, - 550     + 450, - 550
     + 450, - 500     + 550, -1450     + 500, -1450     + 550, -1400
     + 600, -1400     + 550, - 450     + 500, - 450     + 600, - 400
     + 550, - 450     + 500, - 450     + 550, - 450     + 500, - 500
     + 500, - 450     + 450, - 550     + 550, - 450     + 500, - 450
     + 550, - 450     + 550, - 450     + 550, - 450     + 500, - 450
     + 550, - 450     + 550, - 450     + 550, -1400     + 550, -1450
     + 550, -1400     + 450, - 500     + 550, -1450     + 450, -1500
     + 550, -1450     + 550
unsigned int  rawData[83] = {2900,3000, 500,500, 550,1400, 550,1450, 500,450, 550,1450, 500,450, 550,450, 550,400, 550,450, 450,550, 450,550, 450,500, 550,1450, 500,1450, 550,1400, 600,1400, 550,450, 500,450, 600,400, 550,450, 500,450, 550,450, 500,500, 500,450, 450,550, 550,450, 500,450, 550,450, 550,450, 550,450, 500,450, 550,450, 550,450, 550,1400, 550,1450, 550,1400, 450,500, 550,1450, 450,1500, 550,1450, 550};  // UNKNOWN E8BEB62D

```
When I use the IRrecvSend.ino to send the recorded IR code to the robot, nothing happens. I know some ACs need to 'listen' for longer, and would love to hear some references on this. Is this the case? I have read in some issues for this repo that this issue was solved in IRremote.

AnalysIR, 2018-03-10:
> Board: Arduino MEGA

Assuming this means a Mega2560...
..the default Tx pin for the library is pin 9 (not 3).

This information is readily available as part of the Library's documention (See README)

Also, you will get better performance if you change the timing values above to
3000
500
1500

nakulbende, 2018-03-10:
Thanks! 

I read that in README after I ran the code, but my setup indeed ran with receiver at pin 3. Should'nt this just be changed with line 9 of IRrecDUMPV@.ino (`int recvPin = 3;`)??

Anyways, I changed it to pin 9, and here is what I got: 

```
Encoding  : UNKNOWN
Code      : E8BEB62D (32 bits)
Timing[83]: 
     +3050, -2900     + 500, - 500     + 500, -1450     + 450, -1500
     + 500, - 500     + 500, -1450     + 550, - 450     + 500, - 500
     + 500, - 450     + 550, - 450     + 550, - 450     + 550, - 450
     + 450, - 500     + 500, -1500     + 500, -1450     + 500, -1500
     + 500, -1450     + 500, - 500     + 500, - 500     + 450, - 500
     + 500, - 500     + 500, - 450     + 500, - 500     + 500, - 500
     + 500, - 500     + 500, - 450     + 500, - 500     + 500, - 500
     + 500, - 450     + 500, - 500     + 500, - 500     + 500, - 500
     + 450, - 500     + 500, - 500     + 450, -1550     + 500, -1450
     + 450, -1500     + 500, - 500     + 500, -1450     + 550, -1450
     + 450, -1500     + 500
unsigned int  rawData[83] = {3050,2900, 500,500, 500,1450, 450,1500, 500,500, 500,1450, 550,450, 500,500, 500,450, 550,450, 550,450, 550,450, 450,500, 500,1500, 500,1450, 500,1500, 500,1450, 500,500, 500,500, 450,500, 500,500, 500,450, 500,500, 500,500, 500,500, 500,450, 500,500, 500,500, 500,450, 500,500, 500,500, 500,500, 450,500, 500,500, 450,1550, 500,1450, 450,1500, 500,500, 500,1450, 550,1450, 450,1500, 500};  // UNKNOWN E8BEB62D
```
Can you please explain this to me? Do you mean that I should change all the 2900s or 3050s to 3000, 450/ 550 to 500 and 1450 to 1500? I can go ahead and try that - is there an automated code which can do that?
> Also, you will get better performance if you change the timing values above to
>3000
>500
>1500

AnalysIR, 2018-03-10:
the receive pin can be any pin..so 3 is fine.

the send pin (tx) defaults to 9 on the mega2560.

> Can you please explain this to me? Do you mean that I should change all the 2900s or 3050s to 3000, 450/ 550 to 500 and 1450 to 1500? I can go ahead and try that - is there an automated code which can do that?

Yes that is correct. Use any find&replace tool (or excel)

nakulbende, 2018-03-22:
Hey, 

That worked - thanks a lot! I will post the project on github. One small questions:

I will be porting this to ESP8266, using the anologous library. I am using a ESP8266-12, is there a specific pin to be used for LED? I think the sketch suggest 4. Is that the case? Also, I know ESP8266 can be finicky and need some special attention. Is there anything I should know for using IRremote on ESP?

nakulbende, 2018-03-22:
Oh - the issue can be closed.

AnalysIR, 2018-03-22:
Unless, I am mistaken you should be able to close the issue yourself. (I cannot do it for you)
There is a good fork of IRremote for ESP8266 here on Github.
I believe you can use most pins to send or receive using the fork above....but need to be careful with the pins used for bootloading etc...

closed 2020-10-17 by ArminJo