Arduino-IRremote/Arduino-IRremote #554: Sparkfun Pro Micro Atmega32u4 does not work out of the box

drelephant, 2018-03-10:
**Board:** Sparkfun Pro Micro clone Atmega32u4
**Library Version:** 2.2.3
**Protocol:** Any

Because the pro micro does not have a pin 13, there is no output when you use the library.

I (finally!) fixed it by editing boarddefs.h to use Timer3 and Pin 5.

```c
// Teensy 2.0
#elif defined(__AVR_ATmega32U4__)
	//#define IR_USE_TIMER1   // tx = pin 14
	#define IR_USE_TIMER3   // tx = pin 9
	//#define IR_USE_TIMER4_HS  // tx = pin 10

```
and

```
//-----------------
//#if defined(CORE_OC3A_PIN)
//#	define SEND_PIN  CORE_OC3A_PIN  // Teensy
//#elif defined(__AVR_ATmega1280__) || defined(__AVR_ATmega2560__)
#	define SEND_PIN  5              // Arduino Mega
//#elif defined(__AVR_ATmega1284__) || defined(__AVR_ATmega1284P__)
//#	define SEND_PIN  6              // MightyCore
//#else
//#	error "Please add OC3A pin number here\n"
//#endif
```

I'm sorry, I'm too inexperienced with Arduino libraries and c to come up with a way to automatically detect if the board is a Sparkfun Pro Micro and change it so it will use that Timer and Pin in that case, but I spent so long on trying to fix it that I wanted to write this so hopefully others don't have to waste the time I did.

It would be nice if there was some documentation on how to change the Pin and Timer used.

Readme.md lists the Sparkfun Pro Micro as supported, but you could hardly call it well supported when you need to manually edit the library to get it to work at all!!

z3t0, 2018-03-12:
> It would be nice if there was some documentation on how to change the Pin and Timer used.

That is a really good point.

qtoneos, 2019-05-06:
I've tried those changed but I get an error:
`Arduino: 1.8.3 (Windows 10), Board: "SparkFun Pro Micro, ATmega32U4 (5V, 16 MHz)"

Build options changed, rebuilding all
In file included from C:\Users\XXX\Documents\Arduino\libraries\IRremote/IRremoteInt.h:111:0,

                 from C:\Users\XXX\Documents\Arduino\libraries\IRremote/IRremote.h:24,

                 from C:\Users\XXX\Documents\Arduino\Sony_Minidisc_Namer\Sony_Minidisc_Namer.ino:11:

C:\Users\XXX\Documents\Arduino\libraries\IRremote/boarddefs.h:349:3: error: #error "Please add OC3A pin number here\n"

 # error "Please add OC3A pin number here\n"

   ^

Multiple libraries were found for "IRremote.h"
 Used: C:\Users\XXX\Documents\Arduino\libraries\IRremote
 Not used: C:\Program Files (x86)\Arduino\libraries\RobotIRremote
exit status 1
Error compiling for board SparkFun Pro Micro.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

smartroad, 2019-05-06:
I've done those changes but got an error:

`Arduino: 1.8.3 (Windows 10), Board: "SparkFun Pro Micro, ATmega32U4 (5V, 16 MHz)"

In file included from C:\Users\XXX\Documents\Arduino\libraries\IRremote/IRremoteInt.h:111:0,

                 from C:\Users\XXX\Documents\Arduino\libraries\IRremote/IRremote.h:24,

                 from C:\Users\XXX\Documents\Arduino\Sony_Minidisc_Namer\Sony_Minidisc_Namer.ino:11:

C:\Users\XXX\Documents\Arduino\libraries\IRremote/boarddefs.h:349:3: error: #error "Please add OC3A pin number here\n"

 # error "Please add OC3A pin number here\n"

   ^

Multiple libraries were found for "IRremote.h"
 Used: C:\Users\XXX\Documents\Arduino\libraries\IRremote
 Not used: C:\Program Files (x86)\Arduino\libraries\RobotIRremote
exit status 1
Error compiling for board SparkFun Pro Micro.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`

smartroad, 2019-05-06:
Never mind, I missed out most of the comments on the second section :$

use-sparingly, 2019-09-29:
Fixed in https://github.com/z3t0/Arduino-IRremote/pull/543 and https://github.com/z3t0/Arduino-IRremote/pull/542.

Also allows use of timer 1 (PIN 9) and timer 4_HS (PIN 5). Useful when you have conflicts with other libraries.

ArminJo, 2020-07-04:
Fixed in current version.

piotr-kubica, 2020-09-10:
Unfortunately cannot get it working. I have tried every pin 9, 5, 10, 14. It does not work with Pro Micro, but works flawless with Uno on Pin 3. IR led is powered externally by same source. Any ideas?
Do you have a **full** working example code?

alastairmccormack, 2020-09-11:
@piotr-kubica - The Arduino-IRremote code base looks to have been overhauled recently and my proposed changes only semi applied.

The following definition has been lost:

```
#elif defined(ARDUINO_AVR_PROMICRO)
    //#define IR_USE_TIMER1     // tx = pin 9
    #define IR_USE_TIMER3       // tx = pin 5
    //#define IR_USE_TIMER4_HS  // tx = pin 5
```

which represents the unorthodox pin mapping of the Sparkfun Pro Micro. 

Ensure that you're using the latest master branch although I haven't tested this. 

If you're still having a problem, try using: https://github.com/alastairmccormack/Arduino-IRremote/

ArminJo, 2020-09-11:
Hi,
I commited a few changes relevant for ARDUINO_AVR_PROMICRO. I tested IRrecvDemo + IRsendDemo with my board and it runs like a charm.

@piotr-kubica I updated the examples and the feedback LED definitions, please use the version from the repo and give it another try.
@alastairmccormack The definitions were not lost, but in IRremoteBoardDefs.h line 241.

piotr-kubica, 2020-09-11:
@alastairmccormack and @ArminJo thank you for your fast response and help. I will give it a try.

alastairmccormack, 2020-09-11:
@ArminJo - It's the comments on my commit that are probably the most important here. The only supported pins are 9 and 5.

Also, my changes to README.MD didn't make it through, which reflected these comments:

```
| [Sparkfun Pro Micro](https://www.sparkfun.com/products/12640)            | 9, **5**, 5         | 1, **3**, 4_HS    |
```

Many thanks for reacting so quickly (and for a great project!)

ArminJo, 2020-09-11:
You are the boss 😀 
I corrected the readme.
THANKS!

piotr-kubica, 2020-09-13:
@ArminJo tested my sendRaw payload with pin 5. It works. Thanks again.

closed 2020-09-14 by ArminJo