Arduino-IRremote/Arduino-IRremote #556: Can we Get repeat Counts and frequency as well in IRDUMP v2 example. for unknown protocols

arihantdaga, 2018-03-13:
**Board:** ARDUINO UNO
**Library Version:** 2.1.0

**Code Block:**

Use [a gist](gist.github.com) if the code exceeds 30 lines

**checklist:**
- [X] I have **read** the README.md file thoroughly
- [X] I have searched existing issues to see if there is anything I have missed.
- [X] The latest [release](https://github.com/z3t0/Arduino-IRremote/releases/latest) is used
- [] Any code referenced is provided and if over 30 lines a gist is linked INSTEAD of it being pasted in here
- [] The title of the issue is helpful and relevant 

** We will start to close issues that do not follow these guidelines as it doesn't help the contributors who spend time trying to solve issues if the community ignores guidelines!**

I am using Arduino UNO for making a universal remote database generator. 
I have few IR Codes already which works fine and i am able to control the AC. So i tested if this library works properly. 

I fired a rawcode of length 200. 

`4507, 4507, 573, 1641, 573, 547, 573, 1641, 573, 1641, 573, 547, 573, 547, 573, 1641, 573, 547, 573, 547, 573, 1641, 573, 547, 573, 547, 573, 1641, 573, 1641, 573, 547, 573, 1641, 573, 1641, 573, 547, 573, 1641, 573, 1641, 573, 1641, 573, 1641, 573, 1641, 573, 1641, 573, 547, 573, 1641, 573, 547, 573, 547, 573, 547, 573, 547, 573, 547, 573, 547, 573, 547, 573, 1641, 573, 547, 573, 547, 573, 547, 573, 547, 573, 547, 573, 547, 573, 1641, 573, 547, 573, 1641, 573, 1641, 573, 1641, 573, 1641, 573, 1641, 573, 1641, 573, 5211, 4507, 4507, 573, 1641, 573, 547, 573, 1641, 573, 1641, 573, 547, 573, 547, 573, 1641, 573, 547, 573, 547, 573, 1641, 573, 547, 573, 547, 573, 1641, 573, 1641, 573, 547, 573, 1641, 573, 1641, 573, 547, 573, 1641, 573, 1641, 573, 1641, 573, 1641, 573, 1641, 573, 1641, 573, 547, 573, 1641, 573, 547, 573, 547, 573, 547, 573, 547, 573, 547, 573, 547, 573, 547, 573, 1641, 573, 547, 573, 547, 573, 547, 573, 547, 573, 547, 573, 547, 573, 1641, 573, 547, 573, 1641, 573, 1641, 573, 1641, 573, 1641, 573, 1641, 573, 1641, 573, 5211`

But arduino captures only first 100 and dump gives me this result - 

```

Encoding  : SAMSUNG
Code      : B24DBF40 (32 bits)
Timing[99]:
    +4600, -4400     + 650, -1600     + 650, - 500     + 650, -1550
    + 700, -1600     + 650, - 500     + 650, - 500     + 650, -1600
    + 650, - 500     + 650, - 500     + 650, -1600     + 650, - 500
    + 650, - 500     + 650, -1600     + 650, -1600     + 650, - 500
    + 650, -1600     + 650, -1550     + 650, - 500     + 650, -1600
    + 650, -1600     + 650, -1600     + 650, -1550     + 700, -1500
    + 700, -1600     + 650, - 500     + 650, -1600     + 650, - 500
    + 650, - 500     + 650, - 500     + 650, - 500     + 650, - 500
    + 700, - 500     + 650, - 500     + 650, -1600     + 650, - 500
    + 650, - 500     + 650, - 500     + 650, - 500     + 650, - 500
    + 650, - 500     + 650, -1650     + 650, - 500     + 650, -1600
    + 600, -1600     + 650, -1600     + 650, -1600     + 650, -1600
    + 650, -1550     + 650
unsigned int  rawData[99] = {4600,4400, 650,1600, 650,500, 650,1550, 700,1600, 650,500, 650,500, 650,1600, 650,500, 650,500, 650,1600, 650,500, 650,500, 650,1600, 650,1600, 650,500, 650,1600, 650,1550, 650,500, 650,1600, 650,1600, 650,1600, 650,1550, 700,1500, 700,1600, 650,500, 650,1600, 650,500, 650,500, 650,500, 650,500, 650,500, 700,500, 650,500, 650,1600, 650,500, 650,500, 650,500, 650,500, 650,500, 650,500, 650,1650, 650,500, 650,1600, 600,1600, 650,1600, 650,1600, 650,1600, 650,1550, 650};  // SAMSUNG B24DBF40
unsigned int  data = 0xB24DBF40;

```

The code which my tsop reciever could decode is of length 100(sometimes it's 99) only  and sometimes it also does not detect any protocol and still print rawCode of length 99 only, and when i fire this rawcode again my ac doesn't repond unless i repeat the code 2 times. 

Is there any limitation on code size in this library ? or is there a way to get repeat counts in the library ?

AnalysIR, 2018-03-13:
try restarting the IDE, after changing RAWBUF size

arihantdaga, 2018-03-13:
@AnalysIR  Thank you. That helped. 
What about frequency and repeat counts ? Can we get those ?

AnalysIR, 2018-03-13:
> Can we get those ?

Yes, usually when users contribute the code to the library. It is unlikely, but not impossible that someone will do it for you :)

AFAIK frequency and repeat counts are not a feature of this library.

IRLIB/IRLIB2 has a feature for frequency (originally based on my own code) _and possibly repeats_ . 

This post on our blog, shows how to get frequency/period using an IR LED.
https://www.analysir.com/blog/2014/05/28/poor-makers-infrared-receiver-2/
You could use that directly or as a basis for your own starting point.

arihantdaga, 2018-03-13:
@AnalysIR  Thank you for the libraries and explaining. It'll help. And also I'll try to add features for the frequency and repeat code if possible and try to contribute to the library. 
Once again, Thank you for this library, its a great work.

bengtmartensson, 2018-03-18:
## Frequency: 
Note that you need different hardware; the TSOP-* removes the modulation frequency, so you need a different sensor, suggestion TSMP58000. My library [Infrared4Arduino](https://github.com/bengtmartensson/Infrared4Arduino) does the job.

## Repeats: 
Keep in mind that (most?) IR libraries are small (for obvious reasons), and it is a good idea to instead do the analysis of the received signals on a more powerful platform, like a PC. My program [IrScrutinizer](https://github.com/bengtmartensson/harctoolboxbundle) can identify repeats, as well as the more powerful but (presently) command-line only [IrpTransmogrifier](https://github.com/bengtmartensson/IrpTransmogrifier). Possibly also the program AnalysIR can be used.

closed 2020-07-04 by ArminJo