Arduino-IRremote/Arduino-IRremote #558: IRrecvDump compile error for ESP32: SEND_PIN was not declared

kimeckert, 2018-03-17:
**Board:** Adafruit ESP32 Feather
**Library Version:** 2.2.3

I am trying to use my ESP32 Feather board to read IR codes from my remote.

Both the original IRrecvDump code and my minor mod give the same compile error. I am not sure why the compiler is complaining about SEND_PIN because IRremote does not support sending on ESP32. Thanks for your help. Other sketches are compiling and running, just not IRremote sketches.

**Error message:**
<pre>
In file included from /home/kim/Arduino/IRrecvDump/IRrecvDump.ino:11:0:
/home/kim/Arduino/libraries/IRremote/IRremote.h:365:23: error: 'SEND_PIN' was not declared in this scope
   const int sendPin = SEND_PIN;
                       ^
Using library IRremote at version 2.2.3 in folder: /home/kim/Arduino/libraries/IRremote 
exit status 1
Error compiling for board Adafruit ESP32 Feather.
</pre>

bengtmartensson, 2018-03-18:
Try branch "beta". There has been a lot of sane stuff checked in there, that is not in the master branch. (Caveat: It may or may not solve the problem.)

kimeckert, 2018-03-18:
I get the same error with the beta release: 2.4.0b.  If anyone ever got receive to work on an ESP32 board, then there must be some Feather board issue that makes my compile look for a SEND_PIN definition.

**Note:** The warning below about multiple libraries is because I renamed the stable version of IRremote instead of deleting it.  The compile correctly chose the beta version, located at /home/kim/Arduino/libraries/IRremote

**Error message:**
<pre>
In file included from /home/kim/Arduino/IRrecvDump/IRrecvDump.ino:11:0:
/home/kim/Arduino/libraries/IRremote/IRremote.h:365:23: error: 'SEND_PIN' was not declared in this scope
   const int sendPin = SEND_PIN;
                       ^
Multiple libraries were found for "IRremote.h"
 Used: /home/kim/Arduino/libraries/IRremote
 Not used: /home/kim/Arduino/libraries/IRremote_stable
exit status 1
Error compiling for board Adafruit ESP32 Feather.
</pre>

bengtmartensson, 2018-03-18:
[branch named "beta"](https://github.com/z3t0/Arduino-IRremote/tree/beta), not to be confused with "beta-release"!.

kimeckert, 2018-03-19:
I still get the same error.  Below is an edited version of the verbose error message.  Is it possible that there is a conflict between the esp32 in my Arduino hardware folder and the IRremote src/esp32.cpp ?

I moved the other IRremote libraries completely out of the libraries folder so there are no warnings about multiple libraries.  Only the beta branch remains (2.5.0b).
<pre>
Arduino: 1.8.5 (Linux), Board: "Adafruit ESP32 Feather, 80MHz, 921600, None"

/usr/local/arduino-1.8.5/arduino-builder -dump-prefs ...
/usr/local/arduino-1.8.5/arduino-builder -compile ...
Using board 'featheresp32' from platform in folder: /home/kim/Arduino/hardware/espressif/esp32
Using core 'esp32' from platform in folder: /home/kim/Arduino/hardware/espressif/esp32
Detecting libraries used...
... many lines here ...
In file included from /home/kim/Arduino/IRrecvDump/IRrecvDump.ino:11:0:
/home/kim/Arduino/libraries/Arduino-IRremote-beta/src/IRremote.h:415:23: error: 'SEND_PIN' was not declared in this scope
   const int sendPin = SEND_PIN;
                       ^
exit status 1
Error compiling for board Adafruit ESP32 Feather.
</pre>

bengtmartensson, 2018-03-20:
Sigh. IIRC, we had "ESP32" running a year ago (see commits around April 1, 2017).

Anyhow, the fix
```
diff --git a/src/IRremote.h b/src/IRremote.h
index 268204b..cb2a76c 100644
--- a/src/IRremote.h
+++ b/src/IRremote.h
@@ -411,7 +411,7 @@ class IRsend
                void sleepMicros(unsigned long us);
                void sleepUntilMicros(unsigned long targetTime);
 
-#else
+#elif defined(SEND_PIN)
                const int sendPin = SEND_PIN;
 #endif
 } ;
```
certainly would not hurt. Does it solve the problem?

kimeckert, 2018-03-20:
This fixed my problem.  Thanks.

I had previously replaced the 'SEND_PIN' with an integer.  That eliminated the error also, but was just a stop-gap fix.

I am currently working through some other board upload issues, unrelated to this.  I will report more if it is related to this issue.  Closing for now, with my thanks.

closed 2018-03-20 by kimeckert

mrmonteith, 2018-05-19:
I tried to follow this but not sure exactly what to change and where.  I pulled up the IRremote.h but looks like the diff output and the contents now of the file might different.   What exactly do I put in and where for this to work?  Using the 2.4.0 beta release.