Arduino-IRremote/Arduino-IRremote #560: Tucson TV RC5 does(nt respond to send

Neolex-Security, 2018-03-31:
**Board:** ARDUINO MEGA2560 R3
Hi ! 
I'm trying to control my TUCSON TV with you library but when I send from `irsend.sendRC5(0x84c,12);`
or even in raw , my tv doesnt respond ... 
when i dump the on off button with your library i get :
`Decoded RC5: 84C (12 bits)
Raw (22): 950 -850 950 -850 1850 -850 950 -850 950 -850 900 -1800 1850 -850 950 -1700 950 -850 1850 -900 900 v

Decoded RC5: 4C (12 bits)
Raw (22): 950 -850 1850 -850 950 -850 950 -850 950 -850 950 -1750 1800 -900 950 -1750 900 -850 1850 -900 900 `

I have no idea what's the problem...
I hope you can help me, thanks !

AnalysIR, 2018-04-01:
Try this using sendRAW

```
/*
Automatically Generated by AnalysIR - Batch Export Utility
Registered to: xxxxxxxx
Session History
Type : Key : Value : Bits : Carrier Frequency (kHz)
0 : RAW :  :  :  : 0
1 : RAW :  :  :  : 0
Note: Be sure to use the correct Carrier frequency, for each individual signal, as(or if) indicated above
*/

// NB: Not all protocols are supported by IRremote or IRLib. You may need to edit the code below manually
// Automatically Generated by AnalysIR for xxxxxxxx, visit https://www.AnalysIR.com/ or email info@....... for further details
int khz=38; //NB Change this default value as neccessary to the correct carrier frequency

unsigned int Signal_0_0[] = {889,889,1778,889,889,889,889,889,889,889,889,1778,1778,889,889,1778,889,889,1778,889,889}; //AnalysIR Batch Export (IRremote) - RAW

irsend.sendRaw(Signal_0_0, sizeof(Signal_0_0)/sizeof(int), khz); //AnalysIR Batch Export (IRremote) - RAW
 // AnalysIR IR Protocol: RAW, Key:  
```
The timings have been corrected above.

Below, the toggle bit is removed which may help(???)
or this `irsend.sendRC5(0x04C,12);` and report back.

Neolex-Security, 2018-04-01:
Hi ! Thank for your help but it still doesn't work...
For the arduino mega 2560 R3 the pin of the IR Transmitter must be 9, right ?

AnalysIR, 2018-04-01:
> For the arduino mega 2560 R3 the pin of the IR Transmitter must be 9, right ?
Almost, the default for the mega 2560 is pin 9 (from memory). You can find the other options in the readme on the home page of this library.

It sounds like it may be your emitter circuit. Can you post a circuit diagram and a photo of your wiring.

Neolex-Security, 2018-04-01:
Thank you very much for the help ! 
It's like that but I didn't find the right composent on fritzing instead of the black thing it's : http://www.emartee.com/Images/websites/emartee.com/T27W8xXolXXXXXXXXX_!!94851079.jpg
the - + and S go for left to right like in the picture
[image]

AnalysIR, 2018-04-01:
OK, that looks correct, Provided S is connected to pin 9, + is connected to 5V and - is connected to GND.

Some things to try:

- Check with the camera on a mobile phone to see if the IR Led is blinking, when you send
- Make sure you are pointing the top(round) bit of the IR LED towards the target device.
- bring it much closer to the device for initial testing.
- If you have a DMM, use the diode function to check the IR LED, & post the voltage reported.
- Check there is no bad connection in your wiring
- If you are using a breadboard like the one in the image above, make sure the power rails are continuous, or just connect the power/GND directly to the emitter module.

PS: If you don't understand any of the above, look it up on google or ignore.

Neolex-Security, 2018-04-01:
Thank you !
It works fine when I connected the power/GND directly without passing by the breadboard maybe my breadboard is damaged i'll try to look at that when i get back my multemeter.
Thank you very much, and i'm sorry for the inconveniance !

Neolex-Security, 2018-04-01:
I have one last question, how did you get the right raw numbers ? 
Because it only works with raw, you have to replace all <1000 by 889 and the >1000 by 1778 ?

AnalysIR, 2018-04-01:
> I have one last question, how did you get the right raw numbers ?
> Because it only works with raw, you have to replace all <1000 by 889 and the >1000 by 1778 ?

Yes, I did it with our AnalysIR application, which has an auto-clean feature. All IR signals will have the timing off by up to +/- 200Usecs. Most times this is OK, but if you get the timing more accurate, you will have better performance (range/reliability)

> 
> It works fine when I connected the power/GND directly without passing by the breadboard maybe my breadboard is damaged i'll try to look at that when i get back my multemeter.
> Thank you very much, and i'm sorry for the inconveniance !

As I mentioned, some breadboards do not have a continuous power rail for + and - (usually red and blue). It should be indicated by a break in the red & blue lines (but not always).

Finally, its not an inconvenience :)

Neolex-Security, 2018-04-01:
Ok thanks i'll try your application !
My breadboard got that blue and red lines so it's weird ^^ 
But Thank you so much !

closed 2018-04-01 by Neolex-Security