Arduino-IRremote/Arduino-IRremote #562: Need help to work with Panasonic AC remote

jcl06, 2018-04-14:
Hi,

I'm trying to work out to use ARDUINO Mega2560 to remote control Panasonic A/C.

I got the below data received from Serial Monitor when I run IRrecvDumpV2 and copied the raawData to IRsendRawDemo but not working. Appreciate if can help me what's wrong.

Encoding  : UNKNOWN
Code      : D03A82D3 (32 bits)
Timing[179]: 
     +  50, - 250     + 150, - 250     + 150, - 250     + 150, - 200
     + 150, - 250     + 150, - 250     + 150, - 200     + 200, - 200
     + 150, - 250     +  50, -1800     + 100, - 650     +  50, - 300
     + 100, -1300     + 150, - 950     + 100, - 550     +  50, - 300
     +  50, - 600     + 150, - 550     + 100, - 950     + 150, - 500
     + 100, - 300     +  50, - 600     + 150, - 550     + 100, - 950
     + 150, - 500     + 100, - 250     + 100, - 600     + 150, -1700
     + 100, - 550     + 150, - 900     + 100, - 550     + 100, - 950
     + 150, - 550     + 100, - 950     +  50, - 600     + 150, -1700
     + 100, -1400     +  50, - 300     + 100, -1300     +  50, - 300
     +  50, - 600     + 150, - 600     + 100, -1700     + 100, - 950
     +  50, - 600     + 100, - 650     +  50, - 250     + 150, - 500
     + 150, - 900     + 100, - 550     + 150, - 650     +  50, - 250
     + 100, - 550     + 150, - 850     + 150, - 550     + 100, - 700
     +  50, - 200     + 150, - 550     + 100, - 900     + 150, - 600
     +  50, - 950     + 100, - 600     + 100, - 900     + 100, - 600
     + 100, - 950     + 100, - 550     + 100, - 950     + 100, - 600
     +  50, - 300     +  50, - 600     + 150, - 550     + 100, - 900
     + 150, - 550     + 100, - 300     +  50, - 600     + 100, - 600
     + 100, - 250     +  50, - 600     + 100, - 650     +  50, - 300
     +  50, -1400     + 100, -1700     + 150, - 500     + 100, - 300
     +  50, - 600     + 100, - 600     +  50, - 300     +  50, - 600
     + 150, - 600     + 100
unsigned int  rawData[179] = {50,250, 150,250, 150,250, 150,200, 150,250, 150,250, 150,200, 200,200, 150,250, 50,1800, 100,650, 50,300, 100,1300, 150,950, 100,550, 50,300, 50,600, 150,550, 100,950, 150,500, 100,300, 50,600, 150,550, 100,950, 150,500, 100,250, 100,600, 150,1700, 100,550, 150,900, 100,550, 100,950, 150,550, 100,950, 50,600, 150,1700, 100,1400, 50,300, 100,1300, 50,300, 50,600, 150,600, 100,1700, 100,950, 50,600, 100,650, 50,250, 150,500, 150,900, 100,550, 150,650, 50,250, 100,550, 150,850, 150,550, 100,700, 50,200, 150,550, 100,900, 150,600, 50,950, 100,600, 100,900, 100,600, 100,950, 100,550, 100,950, 100,600, 50,300, 50,600, 150,550, 100,900, 150,550, 100,300, 50,600, 100,600, 100,250, 50,600, 100,650, 50,300, 50,1400, 100,1700, 150,500, 100,300, 50,600, 100,600, 50,300, 50,600, 150,600, 100};  // UNKNOWN D03A82D3

AnalysIR, 2018-04-14:
Yes there are some issues with this signal:
- you are getting some values of 50 uSecs which is bad
- could be your IR receiver or the way you are using it
- Make sure to hold the remote about 1m away from the IR receiver
- Post the make/model/link for your IR receiver
- let us know how you have wired it all up & more details about what your environment is (MCU platform etc...just like you are asked when you open a new issue!)

jcl06, 2018-04-14:
Hi,

The above data/output is came from IR receiver (20 - 60khz) VISHAY TSMP5800 the data/signal pin is connected to PIN 11 of arduino mega2560.

I tried to capture using different IR receiver 38khz (KY-022 TL1838 VS1838B 1838) and connect to pin 11 of arduino mega2560 and below is the output from serial monitor but still not able to work it out when I copied an pasted it on IRsendRawDemo.

Encoding  : UNKNOWN
Code      : 68D13F41 (32 bits)
Timing[100]: 
     +3450, -1700     + 400, - 500     + 400, -1300     + 400, - 450
     + 400, - 450     + 450, - 450     + 400, - 450     + 400, - 450
     + 400, - 500     + 400, - 450     + 400, - 450     + 400, - 450
     + 400, - 450     + 450, - 450     + 400, -1300     + 400, - 450
     + 400, - 500     + 400, - 450     + 400, - 450     + 400, - 450
     + 400, - 500     + 400, - 450     + 400, -1300     + 400, -1300
     + 450, -1300     + 400, - 450     + 400, - 450     + 400, -1350
     + 400, - 450     + 400, - 450     + 400, - 450     + 450, - 450
     + 400, - 450     + 400, - 450     + 400, - 500     + 400, - 450
     + 400, - 450     + 400, - 450     + 400, - 500     + 400, - 450
     + 400, - 450     + 400, - 450     + 400, - 500     + 400, - 450
     + 400, - 450     + 400, - 450     + 400, - 500     + 400, - 450
     + 400, - 450     + 400, - 450
unsigned int  rawData[100] = {3450,1700, 400,500, 400,1300, 400,450, 400,450, 450,450, 400,450, 400,450, 400,500, 400,450, 400,450, 400,450, 400,450, 450,450, 400,1300, 400,450, 400,500, 400,450, 400,450, 400,450, 400,500, 400,450, 400,1300, 400,1300, 450,1300, 400,450, 400,450, 400,1350, 400,450, 400,450, 400,450, 450,450, 400,450, 400,450, 400,500, 400,450, 400,450, 400,450, 400,500, 400,450, 400,450, 400,450, 400,500, 400,450, 400,450, 400,450, 400,500, 400,450, 400,450, 400,450 };  // UNKNOWN 68D13F41

jcl06, 2018-04-14:
Below is the output of the sketch from 
https://www.analysir.com/blog/wp-content/uploads/2014/03/Arduino_Record_Long_AirConditioner_Infrared_Signals_10.txt
using  IR receiver 38khz (KY-022 TL1838 VS1838B 1838) connected to PIN 2 of arduino mega2560.

not sure whats wrong. appreciate your help and thanks in advance.

Raw: (439) 3456, -1756, 408, -460, 412, -1324, 412, -460, 408, -468, 408, -460, 408, -464, 408, -460, 412, -464, 408, -460, 412, -460, 408, -464, 408, -464, 412, -456, 412, -1324, 412, -460, 408, -468, 408, -460, 408, -460, 412, -460, 412, -464, 408, -460, 408, -1328, 408, -1324, 408, -1332, 408, -464, 408, -460, 408, -1328, 408, -464, 412, -460, 408, -464, 408, -460, 408, -468, 408, -460, 412, -460, 408, -464, 408, -464, 408, -464, 408, -460, 408, -464, 408, -464, 412, -460, 408, -464, 408, -460, 412, -464, 408, -460, 412, -460, 408, -464, 408, -464, 408, -464, 408, -460, 412, -460, 408, -464, 412, -460, 408, -464, 408, -460, 408, -468, 408, -460, 412, -1320, 412, -1328, 408, -464, 408, -464, 408, -460, 412, -460, 408, -460, 408, -9960, 3436, -1756, 408, -460, 408, -1328, 412, -460, 408, -464, 412, -460, 408, -460, 412, -460, 408, -464, 412, -460, 408, -464, 408, -460, 412, -464, 408, -460, 408, -1328, 408, -464, 408, -464, 408, -464, 408, -460, 412, -460, 408, -464, 412, -460, 408, -1324, 412, -1324, 408, -1332, 408, -460, 412, -460, 408, -1324, 412, -464, 408, -460, 412, -460, 412, -460, 408, -464, 412, -460, 408, -464, 408, -460, 412, -464, 408, -460, 412, -460, 408, -464, 408, -464, 408, -460, 412, -460, 412, -460, 408, -1328, 408, -1328, 408, -1324, 412, -460, 408, -468, 408, -460, 412, -456, 412, -1324, 408, -1332, 408, -460, 408, -1328, 408, -464, 408, -464, 412, -460, 408, -460, 412, -460, 408, -464, 412, -460, 408, -464, 408, -460, 408, -1332, 412, -1324, 408, -1324, 412, -1324, 408, -1332, 408, -460, 408, -1328, 408, -460, 412, -1328, 408, -1324, 412, -460, 408, -1324, 412, -1328, 412, -460, 408, -460, 412, -460, 408, -464, 412, -460, 412, -460, 408, -460, 412, -464, 408, -464, 408, -460, 412, -460, 408, -464, 412, -456, 412, -1324, 412, -1320, 412, -1328, 412, -460, 408, -464, 408, -460, 412, -464, 408, -460, 412, -460, 408, -464, 408, -464, 412, -456, 412, -1324, 408, -1324, 412, -1328, 412, -460, 408, -460, 412, -460, 412, -464, 408, -460, 412, -460, 408, -464, 408, -464, 408, -464, 408, -460, 412, -460, 408, -464, 412, -460, 408, -464, 408, -460, 412, -460, 412, -1324, 412, -460, 408, -464, 408, -464, 408, -464, 408, -460, 412, -460, 408, -1328, 412, -460, 412, -460, 408, -460, 412, -464, 412, -460, 408, -464, 408, -460, 412, -464, 412, -460, 408, -464, 408, -460, 412, -464, 412, -460, 408, -464, 408, -460, 412, -464, 408, -1328, 408, -460, 408, -1328, 408, -468, 408, -1328, 408, -460, 412, -460, 408, -464, 408, 

Press the button on the remote now - once only

Raw: (439) 3432, -1756, 408, -464, 408, -1328, 408, -460, 408, -468, 408, -460, 408, -464, 408, -464, 404, -468, 408, -464, 408, -460, 408, -464, 408, -468, 408, -460, 408, -1328, 408, -464, 404, -468, 408, -464, 408, -460, 408, -464, 408, -464, 408, -460, 408, -1328, 408, -1324, 408, -1332, 408, -464, 408, -460, 408, -1328, 408, -468, 408, -460, 408, -464, 408, -464, 408, -464, 408, -464, 408, -460, 408, -464, 408, -468, 408, -460, 408, -464, 408, -460, 408, -468, 408, -464, 404, -464, 408, -464, 408, -464, 408, -464, 408, -460, 408, -464, 408, -468, 404, -464, 408, -464, 408, -460, 412, -464, 408, -464, 404, -464, 408, -464, 408, -464, 408, -464, 408, -1324, 408, -1328, 408, -468, 408, -460, 412, -460, 408, -460, 408, -460, 412, -9960, 3432, -1756, 408, -464, 408, -1324, 408, -464, 408, -464, 412, -460, 408, -464, 408, -464, 404, -468, 408, -464, 408, -460, 408, -464, 408, -464, 408, -464, 404, -1328, 408, -464, 408, -468, 408, -460, 408, -464, 408, -464, 408, -464, 408, -460, 408, -1328, 408, -1324, 408, -1332, 408, -464, 408, -460, 408, -1328, 408, -468, 404, -464, 408, -464, 408, -460, 408, -468, 408, -460, 408, -464, 408, -464, 404, -468, 408, -464, 408, -460, 408, -464, 408, -464, 408, -1328, 408, -464, 408, -460, 408, -1328, 408, -1328, 408, -1328, 408, -464, 404, -468, 408, -464, 404, -464, 408, -1324, 408, -1332, 408, -464, 404, -1328, 408, -464, 408, -468, 404, -464, 408, -464, 408, -460, 408, -468, 408, -464, 404, -464, 408, -460, 408, -1336, 408, -1324, 408, -1328, 408, -1324, 408, -1332, 408, -464, 404, -1328, 408, -464, 408, -1328, 408, -1328, 408, -460, 408, -1328, 408, -1332, 408, -460, 408, -464, 408, -460, 408, -468, 408, -464, 404, -464, 408, -464, 408, -464, 408, -464, 408, -464, 404, -464, 408, -468, 408, -460, 408, -1328, 404, -1328, 408, -1332, 408, -464, 404, -464, 408, -464, 408, -468, 404, -464, 408, -460, 408, -464, 408, -468, 408, -460, 408, -1328, 404, -1328, 408, -1332, 408, -464, 404, -464, 408, -464, 408, -464, 408, -464, 408, -460, 408, -464, 408, -468, 404, -464, 408, -464, 408, -460, 408, -468, 408, -464, 404, -464, 408, -464, 404, -468, 408, -1328, 408, -460, 408, -464, 408, -468, 408, -460, 408, -464, 408, -460, 408, -1332, 408, -464, 404, -464, 408, -464, 408, -468, 408, -464, 404, -464, 408, -464, 408, -468, 408, -464, 404, -464, 408, -464, 404, -472, 408, -460, 408, -464, 408, -464, 404, -472, 408, -460, 408, -1328, 408, -1328, 404, -468, 408, -1328, 408, -464, 404, -468, 404, -464, 408,

AnalysIR, 2018-04-14:
The TSMP58000 is na IR Learner and output the modulated carrier, which explains your original issue.

The next problem, is that you have hit the RABUF limit. (You can search these issues for a workaround)

Alternatively, try using this sketch instead: https://www.analysir.com/blog/2014/03/19/air-conditioners-problems-recording-long-infrared-remote-control-signals-arduino/

PS: The IR receiver you are now using works sometimes, but in general is a very poor IR receiver (**_in my opinion_**)

OK it looks like you found the sketch by yourself :)

I will look at your updated post shortly.

AnalysIR, 2018-04-14:
OK
It comes up as a panasonic 216 bit AC protocol on AnalysIR. so try using sendRAW now.

jcl06, 2018-04-14:
i tried to send it but nothing happened.

i used the the below sketch IRsendRawDemo and connect IR LED to pin 9 of arduino mega2560.

#include <IRremote.h>

IRsend irsend;

void setup()
{

}

void loop() {
  int khz = 38; // 38kHz carrier frequency for the NEC protocol
  //unsigned int irSignal[] = {9000, 4500, 560, 560, 560, 560, 560, 1690, 560, 560, 560, 560, 560, 560, 560, 560, 560, 560, 560, 1690, 560, 1690, 560, 560, 560, 1690, 560, 1690, 560, 1690, 560, 1690, 560, 1690, 560, 560, 560, 560, 560, 560, 560, 1690, 560, 560, 560, 560, 560, 560, 560, 560, 560, 1690, 560, 1690, 560, 1690, 560, 560, 560, 1690, 560, 1690, 560, 1690, 560, 1690, 560, 39416, 9000, 2210, 560}; //AnalysIR Batch Export (IRremote) - RAW
  unsigned int  irSignal[] = {3432, 1756, 408, 464, 408, 1328, 408, 460, 408, 468, 408, 460, 408, 464, 408, 464, 404, 468, 408, 464, 408, 460, 408, 464, 408, 468, 408, 460, 408, 1328, 408, 464, 404, 468, 408, 464, 408, 460, 408, 464, 408, 464, 408, 460, 408, 1328, 408, 1324, 408, 1332, 408, 464, 408, 460, 408, 1328, 408, 468, 408, 460, 408, 464, 408, 464, 408, 464, 408, 464, 408, 460, 408, 464, 408, 468, 408, 460, 408, 464, 408, 460, 408, 468, 408, 464, 404, 464, 408, 464, 408, 464, 408, 464, 408, 460, 408, 464, 408, 468, 404, 464, 408, 464, 408, 460, 412, 464, 408, 464, 404, 464, 408, 464, 408, 464, 408, 464, 408, 1324, 408, 1328, 408, 468, 408, 460, 412, 460, 408, 460, 408, 460, 412, 9960, 3432, 1756, 408, 464, 408, 1324, 408, 464, 408, 464, 412, 460, 408, 464, 408, 464, 404, 468, 408, 464, 408, 460, 408, 464, 408, 464, 408, 464, 404, 1328, 408, 464, 408, 468, 408, 460, 408, 464, 408, 464, 408, 464, 408, 460, 408, 1328, 408, 1324, 408, 1332, 408, 464, 408, 460, 408, 1328, 408, 468, 404, 464, 408, 464, 408, 460, 408, 468, 408, 460, 408, 464, 408, 464, 404, 468, 408, 464, 408, 460, 408, 464, 408, 464, 408, 1328, 408, 464, 408, 460, 408, 1328, 408, 1328, 408, 1328, 408, 464, 404, 468, 408, 464, 404, 464, 408, 1324, 408, 1332, 408, 464, 404, 1328, 408, 464, 408, 468, 404, 464, 408, 464, 408, 460, 408, 468, 408, 464, 404, 464, 408, 460, 408, 1336, 408, 1324, 408, 1328, 408, 1324, 408, 1332, 408, 464, 404, 1328, 408, 464, 408, 1328, 408, 1328, 408, 460, 408, 1328, 408, 1332, 408, 460, 408, 464, 408, 460, 408, 468, 408, 464, 404, 464, 408, 464, 408, 464, 408, 464, 408, 464, 404, 464, 408, 468, 408, 460, 408, 1328, 404, 1328, 408, 1332, 408, 464, 404, 464, 408, 464, 408, 468, 404, 464, 408, 460, 408, 464, 408, 468, 408, 460, 408, 1328, 404, 1328, 408, 1332, 408, 464, 404, 464, 408, 464, 408, 464, 408, 464, 408, 460, 408, 464, 408, 468, 404, 464, 408, 464, 408, 460, 408, 468, 408, 464, 404, 464, 408, 464, 404, 468, 408, 1328, 408, 460, 408, 464, 408, 468, 408, 460, 408, 464, 408, 460, 408, 1332, 408, 464, 404, 464, 408, 464, 408, 468, 408, 464, 404, 464, 408, 464, 408, 468, 408, 464, 404, 464, 408, 464, 404, 472, 408, 460, 408, 464, 408, 464, 404, 472, 408, 460, 408, 1328, 408, 1328, 404, 468, 408, 1328, 408, 464, 404, 468, 404, 464, 408};  

  irsend.sendRaw(irSignal, sizeof(irSignal) / sizeof(irSignal[0]), khz); //Note the approach used to automatically calculate the size of the array.

  delay(5000); //In this example, the signal will be repeated every 5 seconds, approximately.
}

AnalysIR, 2018-04-14:
please re-edit your post and use code tags from the menu ( i.e. <>)

AnalysIR, 2018-04-14:
So...
- did you check with a camera if the IR LED is blinking?
- point the top of teh LEd towards the AC unit & move it closer
- please post details of your IR emitter circuit

jcl06, 2018-04-14:
I dont have the IR emitter circuit, i just bought IR module online from https://www.aliexpress.com/item/1pcs-Infrared-Emission-Module-IR-Transmitter-For-Arduino-Electronic-Blocks-new-Free-shipping/32655173182.html?spm=a2g0s.9042311.0.0.rK735B

but the IR LED is working fine. it work if I use to my TV but not with the AC.

AnalysIR, 2018-04-14:
If you are just connecting the transmitter directly to the Arduino, you won't get much power from that module (based on the photo). I suspect very little IR is being generated, because the resistor values on the PCB are all wrong.

If you can, change the resistor connected to the IR Led to about 68R or 100R (instead of the 220R that is installed. 

You are probably, just lucky that it works for the TV.

Ask the supplier for the circuit diagram of the module to double check.

My earlier comments still apply.

PS: I will have no more replies until you instert 'code tags' in all of your earlier posts.!!

jcl06, 2018-04-14:
I'll buy resistor and change the circuit.

```
#include <IRremote.h>

IRsend irsend;

void setup()
{

}

void loop() {
  int khz = 38; // 38kHz carrier frequency for the NEC protocol
  //unsigned int irSignal[] = {9000, 4500, 560, 560, 560, 560, 560, 1690, 560, 560, 560, 560, 560, 560, 560, 560, 560, 560, 560, 1690, 560, 1690, 560, 560, 560, 1690, 560, 1690, 560, 1690, 560, 1690, 560, 1690, 560, 560, 560, 560, 560, 560, 560, 1690, 560, 560, 560, 560, 560, 560, 560, 560, 560, 1690, 560, 1690, 560, 1690, 560, 560, 560, 1690, 560, 1690, 560, 1690, 560, 1690, 560, 39416, 9000, 2210, 560}; //AnalysIR Batch Export (IRremote) - RAW
  unsigned int  irSignal[] = {3432, 1756, 408, 464, 408, 1328, 408, 460, 408, 468, 408, 460, 408, 464, 408, 464, 404, 468, 408, 464, 408, 460, 408, 464, 408, 468, 408, 460, 408, 1328, 408, 464, 404, 468, 408, 464, 408, 460, 408, 464, 408, 464, 408, 460, 408, 1328, 408, 1324, 408, 1332, 408, 464, 408, 460, 408, 1328, 408, 468, 408, 460, 408, 464, 408, 464, 408, 464, 408, 464, 408, 460, 408, 464, 408, 468, 408, 460, 408, 464, 408, 460, 408, 468, 408, 464, 404, 464, 408, 464, 408, 464, 408, 464, 408, 460, 408, 464, 408, 468, 404, 464, 408, 464, 408, 460, 412, 464, 408, 464, 404, 464, 408, 464, 408, 464, 408, 464, 408, 1324, 408, 1328, 408, 468, 408, 460, 412, 460, 408, 460, 408, 460, 412, 9960, 3432, 1756, 408, 464, 408, 1324, 408, 464, 408, 464, 412, 460, 408, 464, 408, 464, 404, 468, 408, 464, 408, 460, 408, 464, 408, 464, 408, 464, 404, 1328, 408, 464, 408, 468, 408, 460, 408, 464, 408, 464, 408, 464, 408, 460, 408, 1328, 408, 1324, 408, 1332, 408, 464, 408, 460, 408, 1328, 408, 468, 404, 464, 408, 464, 408, 460, 408, 468, 408, 460, 408, 464, 408, 464, 404, 468, 408, 464, 408, 460, 408, 464, 408, 464, 408, 1328, 408, 464, 408, 460, 408, 1328, 408, 1328, 408, 1328, 408, 464, 404, 468, 408, 464, 404, 464, 408, 1324, 408, 1332, 408, 464, 404, 1328, 408, 464, 408, 468, 404, 464, 408, 464, 408, 460, 408, 468, 408, 464, 404, 464, 408, 460, 408, 1336, 408, 1324, 408, 1328, 408, 1324, 408, 1332, 408, 464, 404, 1328, 408, 464, 408, 1328, 408, 1328, 408, 460, 408, 1328, 408, 1332, 408, 460, 408, 464, 408, 460, 408, 468, 408, 464, 404, 464, 408, 464, 408, 464, 408, 464, 408, 464, 404, 464, 408, 468, 408, 460, 408, 1328, 404, 1328, 408, 1332, 408, 464, 404, 464, 408, 464, 408, 468, 404, 464, 408, 460, 408, 464, 408, 468, 408, 460, 408, 1328, 404, 1328, 408, 1332, 408, 464, 404, 464, 408, 464, 408, 464, 408, 464, 408, 460, 408, 464, 408, 468, 404, 464, 408, 464, 408, 460, 408, 468, 408, 464, 404, 464, 408, 464, 404, 468, 408, 1328, 408, 460, 408, 464, 408, 468, 408, 460, 408, 464, 408, 460, 408, 1332, 408, 464, 404, 464, 408, 464, 408, 468, 408, 464, 404, 464, 408, 464, 408, 468, 408, 464, 404, 464, 408, 464, 404, 472, 408, 460, 408, 464, 408, 464, 404, 472, 408, 460, 408, 1328, 408, 1328, 404, 468, 408, 1328, 408, 464, 404, 468, 404, 464, 408};  

  irsend.sendRaw(irSignal, sizeof(irSignal) / sizeof(irSignal[0]), khz); //Note the approach used to automatically calculate the size of the array.

  delay(5000); //In this example, the signal will be repeated every 5 seconds, approximately.
}
```

cpebrian28, 2019-05-28:
Hai did you solved the problem, i have the same problem. I change rawbuff and success identify the remote  and also the data. Here is the data result in my serial monitor.

Encoding  : PANASONIC
Code      : 4004:7200000 (48 bits)
Timing[131]: 
     +3450, -1700     + 450, - 400     + 450, -1250     + 500, - 400
     + 450, - 400     + 450, - 400     + 450, - 400     + 450, - 400
     + 450, - 400     + 500, - 400     + 450, - 400     + 450, - 400
     + 450, - 400     + 450, - 400     + 500, -1250     + 450, - 400
     + 450, - 400     + 450, - 400     + 450, - 400     + 500, - 350
     + 500, - 400     + 450, - 400     + 450, -1250     + 450, -1300
     + 450, -1250     + 450, - 400     + 450, - 400     + 450, -1300
     + 450, - 400     + 450, - 400     + 450, - 400     + 450, - 400
     + 450, - 400     + 500, - 400     + 450, - 400     + 450, - 400
     + 450, - 400     + 450, - 400     + 500, - 350     + 500, - 400
     + 450, - 400     + 450, - 400     + 450, - 400     + 450, - 400
     + 500, - 350     + 500, - 400     + 450, - 400     + 450, - 400
     + 450, - 400     + 450, - 400     + 500, - 400     + 450, - 400
     + 450, - 400     + 450, - 400     + 450, - 400     + 450, - 400
     + 500, - 400     + 450, - 400     + 450, -1250     + 450, -1250
     + 500, - 400     + 450, - 400     + 450, - 400     + 450, - 400
     + 450, - 400     + 450
unsigned int  rawData[131] = {3450,1700, 450,400, 450,1250, 500,400, 450,400, 450,400, 450,400, 450,400, 450,400, 500,400, 450,400, 450,400, 450,400, 450,400, 500,1250, 450,400, 450,400, 450,400, 450,400, 500,350, 500,400, 450,400, 450,1250, 450,1300, 450,1250, 450,400, 450,400, 450,1300, 450,400, 450,400, 450,400, 450,400, 450,400, 500,400, 450,400, 450,400, 450,400, 450,400, 500,350, 500,400, 450,400, 450,400, 450,400, 450,400, 500,350, 500,400, 450,400, 450,400, 450,400, 450,400, 500,400, 450,400, 450,400, 450,400, 450,400, 450,400, 500,400, 450,400, 450,1250, 450,1250, 500,400, 450,400, 450,400, 450,400, 450,400, 450};  // PANASONIC 4004:7200000
unsigned int  addr = 0x4004;
unsigned int  data = 0x7200000;

Next i send this raw data using below program but nothing happen.

`/*
 * IRremote: IRsendRawDemo - demonstrates sending IR codes with sendRaw
 * An IR LED must be connected to Arduino PWM pin 3.
 * Version 0.1 July, 2009
 * Copyright 2009 Ken Shirriff
 * http://arcfn.com
 *
 * IRsendRawDemo - added by AnalysIR (via www.AnalysIR.com), 24 August 2015
 *
 * This example shows how to send a RAW signal using the IRremote library.
 * The example signal is actually a 32 bit NEC signal.
 * Remote Control button: LGTV Power On/Off. 
 * Hex Value: 0x20DF10EF, 32 bits
 * 
 * It is more efficient to use the sendNEC function to send NEC signals. 
 * Use of sendRaw here, serves only as an example of using the function.
 * 
 */

#include <IRremote.h>

IRsend irsend;

void setup()
{

}

void loop() {
  int khz = 38; // 38kHz carrier frequency for the NEC protocol
  unsigned int irSignal[] = {3450,1700,450,400,450,1250,500,400,450,400,450,400,450,400,450,400,450,400,500,400,450,400,450,400,450,400,450,400,450,1300,450,400,450,400,450,400,450,400,450,400,500,400,450,400,450,1250,450,1250,500,1250,450,400,450,400,450,1250,500,400,450,400,450,400,450,400,450,400,500,400,450,400,450,400,450,400,450,400,450,400,500,400,450,400,450,400,450,400,450,400,450,400,500,400,450,400,450,400,450,400,450,400,450,400,500,400,450,400,450,400,450,400,450,400,500,350,500,400,450,1250,450,1250,500,350,500,400,450,400,450,400,450,400,450};
 
  irsend.sendRaw(irSignal, sizeof(irSignal) / sizeof(irSignal[0]), khz); //Note the approach used to automatically calculate the size of the array.

  delay(5000); //In this example, the signal will be repeated every 5 seconds, approximately.
}`

I using my phone camera to make sure that the LED is blingking.  Any Suggestion are welcome.
Thansk a lot. 

This is transmitter that i use
https://www.bukalapak.com/product_detail_history?product_id=142368911&product_sku_id=135855386&transaction_id=2019667165

and this is the receiver
https://www.bukalapak.com/product_detail_history?product_id=159913451&product_sku_id=147603641&transaction_id=2019667165

AnalysIR, 2019-05-28:
3 things

1. The links you posted require a login/registration (most people wont bother)
2. Please read all of the posts above, particularly about using code tags
3. Probably a good idea to open a new issue and link back to this one if it makes sense.

cpebrian28, 2019-05-29:
Oke I will open e new issue.
And sorry i do a mistake cause i new to use forum like this.

closed 2020-10-17 by ArminJo