Arduino-IRremote/Arduino-IRremote #563: Not working with TSOP1238

sinand99, 2018-04-18:
Board: ARDUINO Mega2560
Library Version: 2.2.3

Does this library support TSOP1238 receiver ? (38 Khz)  Testing with a Humax remote via IRrecvDemo and IRrecvDump, I can't get any signal. I confirmed that the hardware is working by testing the output with a led. Maybe some decode issue? Is there any code to display raw signals without decoding?

AnalysIR, 2018-04-18:
with the little detail provided....

...more likely a wiring issue....have you tried connecting to pin 9 on the Mega?

`Maybe some decode issue? Is there any code to display raw signals without decoding?`

The examples do that for you. Will deoce is possible and provide the raw data as well.

sinand99, 2018-04-18:
Wiring is very basic just like in the wiki: https://github.com/z3t0/Arduino-IRremote/wiki/Hardware-setup

According to the datasheet, GND to IR pin1, 5V to IR pin2. I tried connecting the output to pin 11 and 9. None of them works.  Serial monitor is empty. 

Which example provide raw data? I saw a function for sending, "IRsend::sendRaw()" but nothing like "ReceiveRaw()"?

sinand99, 2018-04-18:
You are right, those pins do not work for receiving. I tried with pin 8 and it works now. Thanks for your help!

closed 2018-04-18 by sinand99

AnalysIR, 2018-04-18:
Actually, if you have no IRsend defined & nothing else is using the related Timer, they should work fine.

....in case anyone else is reading. However, probably a good idea to avoid them on the Mega, as they usually have lots of other things connected, that may use these timers.