Arduino-IRremote/Arduino-IRremote #564: Problem with Asus/Philips remote

tomaszg7, 2018-04-24:
**Board:** ARDUINO UNO
**Library Version:** 2.2.3 (according to arduino ide) / 2.4.0beta / HEAD
**Protocol:** RC6/?

Hi, I'm trying to send signals which I grabbed from Asus remote (included in their old HTPC AT3ION motherboard). It has a USB dongle receiver (identified as `USB HID v1.00 Keyboard [PHILIPS MCE USB IR Receiver- Spinel plusf0r ASUS]`). I was able to use it with LIRC serial receiver long time ago, now I just want to try to emulate it with Arduino. I already managed to clone some of my other remotes (Panasonic TV, Denon amp), so my setup is probably correct.

I managed to dump codes sent by the remote, but receiver doesn't recognize the signals. Here is a sample output of `IRrecvDumpV2` for several buttons:

```
Encoding  : RC6
Code      : 80263D58 (36 bits)
Timing[61]: 
     +2650, - 850     + 450, - 450     + 450, - 450     + 400, - 900
     + 450, - 850     +1300, - 900     + 450, - 450     + 400, - 450
     + 450, - 450     + 450, - 450     + 400, - 450     + 450, - 450
     + 450, - 450     + 400, - 450     + 900, - 850     + 450, - 450
     + 900, - 400     + 450, - 900     + 450, - 400     + 450, - 450
     + 900, - 400     + 450, - 450     + 450, - 450     + 400, - 900
     + 850, - 900     + 900, - 850     + 900, - 450     + 400, - 900
     + 450, - 450     + 400, - 450     + 450
unsigned int  rawData[61] = {2650,850, 450,450, 450,450, 400,900, 450,850, 1300,900, 450,450, 400,450, 450,450, 450,450, 400,450, 450,450, 450,450, 400,450, 900,850, 450,450, 900,400, 450,900, 450,400, 450,450, 900,400, 450,450, 450,450, 400,900, 850,900, 900,850, 900,450, 400,900, 450,450, 400,450, 450};  // RC6 80263D58
unsigned int  data = 0x80263D58;

Encoding  : RC6
Code      : 8026BD59 (36 bits)
Timing[57]: 
     +2650, - 850     + 500, - 400     + 450, - 400     + 500, - 850
     + 450, - 850     +1350, - 850     + 450, - 450     + 450, - 400
     + 450, - 450     + 450, - 450     + 450, - 400     + 450, - 450
     + 450, - 450     + 450, - 400     + 900, - 850     + 500, - 400
     + 900, - 400     + 500, - 850     + 900, - 850     + 900, - 400
     + 500, - 400     + 450, - 400     + 500, - 850     + 900, - 850
     + 900, - 850     + 900, - 400     + 500, - 850     + 450, - 400
     + 900
unsigned int  rawData[57] = {2650,850, 500,400, 450,400, 500,850, 450,850, 1350,850, 450,450, 450,400, 450,450, 450,450, 450,400, 450,450, 450,450, 450,400, 900,850, 500,400, 900,400, 500,850, 900,850, 900,400, 500,400, 450,400, 500,850, 900,850, 900,850, 900,400, 500,850, 450,400, 900};  // RC6 8026BD59
unsigned int  data = 0x8026BD59;

Encoding  : RC6
Code      : 80263D5A (36 bits)
Timing[59]: 
     +2700, - 850     + 450, - 400     + 500, - 400     + 450, - 850
     + 500, - 800     +1350, - 850     + 500, - 400     + 450, - 400
     + 500, - 400     + 500, - 400     + 500, - 350     + 500, - 400
     + 500, - 400     + 500, - 350     + 950, - 850     + 450, - 400
     + 950, - 400     + 450, - 850     + 450, - 450     + 450, - 400
     + 900, - 450     + 450, - 400     + 500, - 400     + 450, - 850
     + 950, - 800     + 950, - 800     + 950, - 400     + 450, - 850
     + 950, - 800     + 500
unsigned int  rawData[59] = {2700,850, 450,400, 500,400, 450,850, 500,800, 1350,850, 500,400, 450,400, 500,400, 500,400, 500,350, 500,400, 500,400, 500,350, 950,850, 450,400, 950,400, 450,850, 450,450, 450,400, 900,450, 450,400, 500,400, 450,850, 950,800, 950,800, 950,400, 450,850, 950,800, 500};  // RC6 80263D5A
unsigned int  data = 0x80263D5A;
```

I tried putting `irsend.sendRC6(0x8026BD59, 36);` into `IRsendDemo`, but no signal was received by my receiver. Sending raw data via `IRsendRawDemo` seems to work, but it's not a very elegant solution ;) I'll be glad for any hint or help.

For reference, here's part of my old `.lircd.conf` file for that remote:

```
begin remote

  name  phil.lirc.conf
  bits            8
  flags RC6|CONST_LENGTH
  eps            30
  aeps          100

  header       2666   867
  one           453   433
  zero          453   433
  pre_data_bits   29
  pre_data       0x37FD942
  gap          106039
  toggle_bit_mask 0x8000
  rc6_mask    0x100000000

      begin codes
          up                       0xA7
          down                     0xA6
          ...............
      end codes
end remote
```

AnalysIR, 2018-04-24:
The sendRC6 in the library supports up to 32 bits.

So your options are:
1. Use sendRAW for this protocol. Take care about toggle bits.
2. Update the code in the library to support this MicrosoftRC6 variant.

Note: if changing the code, the toggle bit may not be in the same position as standard RC6. It may be better to just implement a new protocol as Microsoft_RC6 or MCE_RC6.

tomaszg7, 2018-04-25:
Thanks. For now I'll stick to the raw codes. I'll look into the code in free time, but probably don't know about those protocols to fix it.

tomaszg7, 2018-04-25:
Ok, it seems that necessary changes in the code were fairly minimal (https://github.com/tomaszg7/Arduino-IRremote/commit/39bcd69d032a02bc9bca3097dbd7a1abbd09f106). Of course it depends on whether on all supported platforms `long long` is really 64 bit wide. This code is WIP and obviously would need to be protected by some `#IFDEF` in order not to increase memory footprint.

I didn't look yet into the toggle bit, but the comment in the source suggests that RC6 ignores it and makes it callers job to apply it. However in this form it seemed to work for me just fine, but I didn't test it extensively.

closed 2020-08-17 by ArminJo