Arduino-IRremote/Arduino-IRremote #566: sonySend() format and how to it from ir Codes

ILoveCplusplus, 2018-04-30:
Protocol: Sony 12
Hello, I am currently sending the example code to my Sony tv succesfully. 
"sendSony(0xa90, 12)". it turns on and of my TV. But when looking for ir codes online they are in pronto hex format.
this is supposedly the code to turn on and off sony tv's: "0000 0068 0000 000D 0060 0018 0030 0018 0018 0018 0030 0018 0018 0018 0030 0018 0018 0018 0018 0018 0030 0018 0018 0018 0018 0018 0018 0018 0018 0408", which looks nothing like "0xa90". I searched through here and online and the irdb converter does not produce anything that looks like what im supposed to send. 
Can anyone help me in understanding how to convert these codes?

ILoveCplusplus, 2018-04-30:
this is exactly what ive been looking for, thank you!

closed 2018-04-30 by ILoveCplusplus

LeeFuuChang, 2021-10-30:
> Protocol: Sony 12 Hello, I am currently sending the example code to my Sony tv succesfully. "sendSony(0xa90, 12)". it turns on and of my TV. But when looking for ir codes online they are in pronto hex format. this is supposedly the code to turn on and off sony tv's: "0000 0068 0000 000D 0060 0018 0030 0018 0018 0018 0030 0018 0018 0018 0030 0018 0018 0018 0018 0018 0030 0018 0018 0018 0018 0018 0018 0018 0018 0408",
> 
> which looks nothing like "0xa90". I searched through here and online and the [irdb](http://irdb.tk/convert/) converter does not produce anything that looks like what i'm supposed to send. I have downloaded irScrutinizer as suggested in a different thread and do not see how it can get me any closer to the "0xa90" piece I need Can anyone help me in understanding how to convert these codes?

I've the same problem

> > I have downloaded irScrutinizer as suggested in a different thread and do not see how it can get me any closer to the "0xa90" piece I need
> 
> Use the export format "Arduino/IRremote" and it will generate a complete sketch for you,

but I don't understand the reply above
how IRsender convert those codes?

ArminJo, 2021-10-30:
Do you have the original remote  and a IR receiver module at hand?
And there are different databases using decimal notations for the codes http://www.harctoolbox.org/IR-resources.html and https://github.com/probonopd/irdb/tree/master/codes/Sony/TV.